cleanflight/cleanflight #912: Failsafe = 1000 and motor_stop disabled does not stop motors in failsafe

Spirre, 2015-05-18:
I was just testing my failsafe setup and noticed that if armed and throttle up and the turning off my TX it goes into failsafe but not as I expected, the motors run at the min throttle, same as motor_stop disabled when just armed??
 
I've set failsafe to 1000 and with motor_stop enabled it cuts the motors and dies. In my world I would expect the same thing happen when I have motor_stop enabled, or are I'm missing something?

Min throttle = 1150
Failsafe throttle = 1000
Min command = 1000

hydra, 2015-05-19:
your failsafe throttle should be configured as per the failsafe instructions.

Spirre, 2015-05-19:
But it's ignoring my failsafe throttle, that was the problem.

Spirre, 2015-05-19:
it works with motor_stop enabled

Spirre, 2015-05-19:
And it totally ignores the delays too, it never stop spinning the motors

hydra, 2015-05-19:
Oh ok, I will take a look. Maybe its an edge case that has been missed.  Wouldn't surprise me if its always been like that.  Must admit I don't use motor_stop myself.

hydra, 2015-05-19:
I'm still not sure about this, i've read the title of this issue and the description like 10 times now and still don't fully understand the problem.

Please can you give more details and exact steps to repeat and what your expectation is?

hydra, 2015-05-19:
Please also provide a full configuration dump.  Please also give details of your receiver.  Finally please give details of how your receiver behaves when your TX is off.

hydra, 2015-05-19:
The MOTOR_STOP takes precedence over failsafe currently.

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L704-L713

hydra, 2015-05-19:
I think adding this to the failsafe documentation should help clarify things:

5.  If you are using `MOTOR_STOP` feature ensure that your `failsafe_throttle` is greater or equal to the `min_check` setting. Otherwise the motors will use the `min_command` instead of `min_throttle`.

Now, If you configure your failsafe in accordance with the failsafe documentation *and* 5. above do you still have an issue?

hydra, 2015-05-19:
Documentation is correct.  Behaviour here matches my expectations.  Tested with and without MOTOR_STOP.  Failsafe kicks in in both cases but when MOTOR_STOP is enabled the low value for failsafe_throttle causes the motors to stop.

hydra, 2015-05-19:
I'm wondring if the behaviour should be changed so when MOTOR_STOP is enabled the motors don't stop if the failsafe_throttle is less than the min check value while failsafe is active.

It seems sensible to me otherwise your aircraft will fall out the sky.

hydra, 2015-05-19:
Fixed in https://github.com/cleanflight/cleanflight/commit/cc5c736362793aaa124442275a7b7b6078d75433

closed 2015-05-19 by hydra

Spirre, 2015-05-19:
But I would like it to be that if I set failsafe to 1000 the motors should stop even if motor_stop is disabled as the motors never stops otherwise.

hydra, 2015-05-19:
@Spirre please describe 
a) what settings you want to set (given)
b) what actually happens for you (observed behaviour)
c) and what you want to happen (expected behaviour)

hydra, 2015-05-20:
After discussion in IRC Spire appeared to want this

' Allow mixer to use failsafe throttle value that is less than minthrottle'
https://github.com/cleanflight/cleanflight/commit/22bf890cf045c55960e586ecc7acdbb379b7c9d2

timdonovanuk, 2015-12-21:
I'm experiencing the same thing, probably not understanding - but with MOTOR_STOP disabled, my motors continue to spin at minimum throttle, when failsafe is active (switched off TX). If I enabled MOTOR_STOP they actually stop once failsafe is active. 

I want my motors to come to a complete standstill when failsafe is active with MOTOR_STOP disabled, but I cannot find a way to do this.