cleanflight/cleanflight #915: Request for D term Low Pass filter

borisbstyle, 2015-05-19:
As proved before. The D term is seriously affected by vibrations and even jitter in looptimes.

Especially the vibration sensitivity of it can introduce even more oscillations as the D term will strongly affect the total PID sum.
This can be demonstrated with Blackbox data and is discussed before
See here an example
http://www.multiwii.com/forum/viewtopic.php?f=23&t=2624

harakiri PID controller already has one, but should we try to implement a generic filter for all PID controllers  somehow what can be enabled?

@MJ666 @hydra @ledvinap 
What do you think about this?

ledvinap, 2015-05-19:
It's on my roadmap ;-)

It is possible to implement filter that has same frequency characteristic as differentiation for low frequencies and low gain at high frequencies. And it may be designed to have zero group delay at DC.

Savitzky–Golay (http://en.wikipedia.org/wiki/Savitzky%E2%80%93Golay_filter) is one class of filters with this property ... 
http://www.researchgate.net/profile/Jianwen_Luo/publication/220227041_Savitzky-Golay_smoothing_and_differentiation_filter_for_even_number_data/links/0c960518edf8f39444000000.pdf

The implementation should be quite straightforward and easy, just some googling + FIR filter implementation ... Any volunteers?

borisbstyle, 2015-05-19:
Interresting stuff for reading. 

Well I was also about to start digging into this, but feel free to do it :D. I was going to try the simple apporach like harakiri did for example.

ledvinap, 2015-05-19:
The main trick here is that the system is assumed to be linear. The coefficients are computed offline (there is Matlab code for that) and all that needs to be implemented is FIR filter.

BTW: existing PID D filtering code just subtracts  3 samples old value from actual value, in quite complicated way ..

borisbstyle, 2015-05-19:
Yes I see it uses cycletime in it as well to overcome jitter.

ledvinap, 2015-05-19:
Using actual loop time is problematic - gyro samplerate is relatively low (1kHz), so in value seen in loop may be up to 1ms old. Even with 3.5ms looptime jitter caused by sampling is much bigger than actual looptime difference (maybe on ATMega the situaton is different)

borisbstyle, 2015-05-19:
So what do you suggest as a good approach to test with? Just FIR filter and still 2 old values combined with the actual one?

Also what about syncyng the gyro and the looptime?

So basically each time there is a new sample from the gyro the new PID calculation will be done based on a Interrupt request, but I guess that would change some basic fundementals of the current code.

ledvinap, 2015-05-19:
As for D term - here is Savitzky-Golay filter implementation: http://www.mathworks.com/matlabcentral/fileexchange/30299-savitzky-golay-smooth-differentiation-filters-and-filter-application/content/savitzkyGolay.m . It works in octave.

This is what I get with 15 tap Savitzky-Golay filter (`savitzkyGolay([0:15], 2, 1)`) when applied to @thenickdude armattan (very low noise) data (LOG00568.01):
[image]
[image]

Blue is gyro signal, red Savitzky-Golay output, green is integrated SG

Very important feature is minimum delay - reconstructed signal very closely matches original.

I have very little experience with system control, but graphically the results look promising.

For refference: current algorithm, green trace:
[image]

ledvinap, 2015-05-19:
And gyro sync - in my 'local' branch I have code that uses MPU6050 FIFO to read samples at 1 kHz and than filter them. Main loop is synchronized to MPU6050 frequency, on Naze it is run every fourth sample (4ms looptime) and it is flyable ... 
It is just quick proof of concept, it needs a lot to work, but it may be a way to improve CF.
BTW: 1kHz gyro and acc blackbox logging is hacked into it (really ugly hack ... ), it makes possible to analyze full gyro data ...

borisbstyle, 2015-05-19:
wow....that looks seriously much much smoother.
Is there any other fc firmware or controller using this method of filtering that you know?

If you want make an experimental PR so I can see what you did?
Perhaps try to squeze the filter code in it so we can work from there. Probably a long way to get this 100% right, but in theory this could be a big improvement without doubt.
I am just worried about the potential delay that can be intruduced due to the filtering if it would affect performance by any chance?

ledvinap, 2015-05-19:
The images are generated in Octave from blackbox data. I have no CF implementation of D term filter.

But it is simple FIR filter, so it should be very easy to implement. Then just get the coefficients right, Matlab/Octave is great testing tool for that.

The trick with SG is that it may have zero delay (at DC/as long as polynomial interpolation holds; you can choose point where to estimate derivative; this may be even in future, with negative delay).

borisbstyle, 2015-05-20:
So what do you suggest FIR filter should only apply to D term or entire gyro output?

ledvinap, 2015-05-20:
I would try applying difference-aproximating FIR to gyro output ...

borisbstyle, 2015-05-20:
Wasn't there someone doing simillar thing with kalman filter in baseflight like 2 years ago?

robert-b, 2015-05-20:
i would opt for a simple first order filter.
mathematically stable.

using higher order filter there is a lot of energy inside the tabs ;-)
having higher order filters cascaded may result in interesting effects.

there is a price shield for filters:
 - cpu load
 - delay
 - phase response

borisbstyle, 2015-05-20:
I was testing with smoothing of D term in pidrewrite and quickly tried more averaging and this really smoothed it out.
deltaSum = delta1[axis] + delta2[axis] + delta3[axis] + delta4[axis] + delta5[axis] + delta;

ledvinap, 2015-05-20:
@borisbstyle :
```
delta = gyro[t] - gyro[t-1]
delta1 = gyro[t-1] - gyro[t-2]
delta2 = gyro[t-2] - gyro[t-3]

deltaSum = (gyro[t] - gyro[t-1] + gyro[t-1] - gyro[t-2] + gyro[t-2] - gyro[t-3])/3 =  (gyro[t] - gyro[t-3])/3
```

The filtering works, but this may not be the best way to implement it ...

borisbstyle, 2015-05-20:
of course not....this was just for testing.

I was able to get motors more silent.

borisbstyle, 2015-05-20:
i accidantaly closed :(

borisbstyle, 2015-05-20:
@robert-b 
I see you were already working with FIR filter for baseflight. What's your experience with it?
https://github.com/robert-b/baseflight/tree/dev_filter_fir

borisbstyle, 2015-05-21:
@ledvinap 
So you are still suggesting to smooth the gyro output itself? Any suggestions for this approach? I guess this would have to be done in imu.c during imuUpdate perhaps?
This would mean we would need to add some delay to gyroData to get a window for smoothing it for example with FIR filter?

ledvinap, 2015-05-21:
@borisbstyle : 

https://github.com/robert-b/baseflight/tree/dev_filter_fir is only quite complicated way to implement first order exponential decay filter (except for startup transient) - Kallman gain in this case is dependent only on (constant) noise parameters, so it will quickly converge to constant value.

As for implementation .. SG differentiation can be implemented as FIR and can be designed to have zero DC delay(at the cost of frequency response). It also suppresses signals in higher frequencies (but not too much). 

Using lowpass (no matter how implemented - mechanical/analog/digital) will introduce delay that will degrade performance, mainly in D term, so it should be used very carefully. Minimizing noise sources is best here ... 

Personally I would try SG filter to get D term:

```
savitzkyGolay([0:10], 2, 1) = 
  -0.2590909
  -0.1106061
   0.0037879
   0.0840909
   0.1303030
   0.1424242
   0.1204545
   0.0643939
  -0.0257576
  -0.1500000
```

Adding second order butterworth IIR filter with 40Hz passband does improve results visually, but I have no idea how that will help with flight.

All simulated on @thenickdude's armattan recording, so on extremely low-vibration quad. It may be totally invalid on quads with larder props (>6in).

robert-b, 2015-05-21:
i see it - you need to see the price of your SG filter ;-)
you forgot, that the d-term is able to amplify noise.
thus you need a filter ... doing not so is problematic ...

i agree on the phase delay - which is a killer.
thus only first order filters.

hmm, the d-term tends to grow on changes - right.
so weed no big delay on higher frequencies.

robert-b, 2015-05-21:
http://research.microsoft.com/en-us/um/people/jckrumm/SavGol/SavGol.htm
http://www.statistics4u.info/fundstat_eng/cc_filter_savgolay.html

i have to admit, that i have been looking at SG - but not as solution for the d-term.
http://www.holoborodko.com/pavel/numerical-methods/numerical-derivative/smooth-low-noise-differentiators/
i have implemented the upper proposal to smooth the first integration step of the accelerometer.

ledvinap, 2015-05-21:
It's not about SG filter ... it's about using filter with frequency-characteristic that matches differentiation on low frequencies and has low gain at high frequencies. You can design this as filter or build it as lowpass + differentiator. The filter design approach may be a bit more systematic and may provide optimal results (according to  chosen optimum metric).

SG is nice that it can be designed with zero group delay at DC and tools are readily available, but any filter with this characteristic will do.

here is another approach: http://www.holoborodko.com/pavel/numerical-methods/numerical-derivative/smooth-low-noise-differentiators/ , but these filters have constant group delay of half the filter length (unless I did something wrongly)... 

I don't see why first order filter will have any advantage. For lowpass filtering delay is about the same to for second order filter and first order with half bandwidth.

IIR filter is probably preferable, because it can do filtering with lower (but non-constant) delay than (symmetrical) FIR filter. Higher order can be decomposed to second order sections, so the stability may be maintained.

I am no expert on this topic, I am learning system control on the go. All I state here is just my opinion..

robert-b, 2015-05-21:
:-)

robert-b, 2015-05-21:
ok, i remember having control laws 35 years back ... so i forgot quite a few elements.
and never done anything with it until flying arducopter ...
the phase delay for a pt1 element is 90 degrees and for a pt2 element 180 degrees.
more than 90 degrees is problematic for me.

i am not so happy with the time delay of pavel's approach.
the but is - i have to redo my testing without the pt2 elements.
and then it might be better.

i had to many desktop ideas which where producing nice data sets.
but in most cases it didn't fly very well.

one stupid idea is using a median filter for the mems devices.
a surprise :-)
and the SG filter might be better to tame the accelerometer.

borisbstyle, 2015-05-31:
I see robert already made a PR

https://github.com/cleanflight/cleanflight/pull/950

borisbstyle, 2015-06-09:
https://github.com/cleanflight/cleanflight/pull/1004

closed 2015-09-09 by borisbstyle