cleanflight/cleanflight #918: RX - Signal loss results in violent change in direction before failsafe kicks in.

hydra, 2015-05-19:
A problem occurs when no receiver is powered on, or data is not received or sbus rx is in failsafe mode, etc.

An X8R SBus RX in failsafe mode outputs 880 on all channels.

Basically the current code resets the channel to the middle (midrc) 1500 when no signal is received.  this is *bad* 

For stick channels, it makes sense to do so since that is a neutral position but for aux channels it is not good at all, since modes get actived.  Further more, it uses the values 750 and 2250 to decide what is good or bad, and this value is not configurable

So, when you're flying, and you loose signal it essentially makes the craft yaw to the left, pitch backwards and roll left.  Then the failsafe kicks in after the failsafe delay then levels it and uses `failsafe_throttle`.

The new code uses `rx_min_usec` and `rx_max_usec` and/or the range 750 to 2250 to validate the range (which ever defines the smallest range).

Any AUX channel falling outside that range (instead of the previous fixed range of 750/2250) will be reset to either rx_min_usec when the value is less or rx_max_usec if the value is too high
Any STICK channel falling outside of that range will be reset to the midrc (~1500)

This means in that short period where your sbus rx is in failsafe mode before the failsafe kicks in the aircraft won't suddenly behave erratically.

It also means that AUX channels won't accidentally enable modes that are covered by the midrc value.

hydra, 2015-05-19:
See 7a01bab50ff467c72adf38f4f385873f7a681d4d

ctzsnooze, 2015-07-27:
Although the quad may go level, replacing the bad PPM with midRC could be unsafe.

All controls level plus throttle at 1500 would cause a quad to shoot upwards very fast until failsafe kicked in.

This could cause injury if a person that was leaning over their quad and switched off their Tx without realising the quad was armed.

Wouldn't it be better to repetitively persist with the last valid data point until activating failsafe - that's what most Rx do?

hydra, 2015-08-04:
Fixed with #995.

hydra, 2015-08-20:
Fixed more with #1224

closed 2015-08-20 by hydra