cleanflight/cleanflight #922: CLI - Remove it

hydra, 2015-05-19:


ProDrone, 2015-05-19:
The CLI is what i use to (constantly) backup & restore. How much codespace is used by the CLI. I assume you are looking for places to eliminate code and make room for other things.

hydra, 2015-05-19:
CJMCU with cli = 63020
CJMCU without cli = 50096

hydra, 2015-05-19:
@ProDrone CLI would not be needed given #347

hydra, 2015-05-19:
CJMCU without CLI, with Blackbox = 55884

See #923

ProDrone, 2015-05-19:
I see what you mean and like the idea of removing CLI as it is. But we may end up with another (fat?) client that needs to be maintained and in sync. Atm the CLI is a very practical tool when developing code and support for features not yet in the configurator (also a client that quickly falls behind, like the EzGUI Android app i like so much.

Another thing is that not all contributors to C code may be able to adapt the client tools to test their code. The CLI is also there and easy. It should not stop us from moving forward but may be something to think about.

It would be nice if modules of code could be choosen to fill up the available flash/ram space, like with a configurator i mean. I was thinking of that today and looked at the RapidFlash tool for SimonK.

MJ666, 2015-05-20:
In my experience all CJMCU wave 128KByte Flash. Has someone really a confirmation there will be only 64KByte? The CF configurator will flash without issues the larger version. The STM Demo Loader need to be switched to 128KByte manually in one of the initial dialogs before flashing. I thing the effort to reduce code size is not really need. I know STM is advertising the STM32F103C8T6 with 64KByte only but in reality they all locks to have 128KByte.

ProDrone, 2015-05-20:
Interesting! Maybe we should have an additional entry in the makefile, for a 64 and a 128K version. Then we can ask users to test the 128K version and later remove the 64K version if no longer needed,

fpvrecursion, 2015-05-21:
Any idea on what that protocol might be?

hydra, 2015-06-10:
@ProDrone the conversation around CJMCU is off topic.

@fpvrecursion not yet.

ProDrone, 2015-06-10:
@hydra Sorry for off topic comment. I removed it.

u8mydog, 2015-07-31:
I agree. dump CLI and move to MSP. CLI just gets in the way. The only time I've used a terminal emulator instead of gui is when playing around with bluetooth.

jevring, 2015-08-11:
While I conceptually agree, I'd be wary of making life hard for other tools, especially the android tools. If they already use something else, that's fine, but it would be interesting to check first.

joshuabardwell, 2015-08-11:
@jevring I once used MW GUI Base Station's CLI functionality to re-build a FC config from scratch in the field. It was very useful.

bdoiron74, 2015-08-11:
Has anyone looked at the way MAVLINK deals with configuration variables? I'm not sure it is more code efficient, but it allows developers to add custom variables which the gui can then display / backup / restore. Currently I find cleanfligt a pain to restore with custom firmware since the backup doesn't cover everything in the cli.

Maybe I'm doing it wrong ... I restore via the gui, then cli dump / diff and restore the rest of the diffs individually. Yuck.

waslater, 2015-08-23:
Practically speaking, when the CLI is gone how do you change all the arcane settings that you wouldn't necessarily want included in a friendly GUI?  Like these for example:
set fixedwing_althold_dir = 1
set reboot_character = 82

Would this external CLI client serve that purpose? Would it have to keep pace with the actual firmware in the same way that the GUI/configurator does?

joshuabardwell, 2015-08-23:
@waslater All configurable options would be ported to the new protocol. The same mechanisms that now are used to define command line options would instead define protocol parameters. From a user perspective, there would be no difference, except that you would now have to use the CLI tab in the GUI to access the command line, instead of also being able to use a terminal emulator. The CLI tab would present a command prompt, at which you would enter commands just as before. The CLI tab would translate those commands to MSP (or whatever protocol is used) and communicate with the FC, then print out the results.

readerror67, 2015-09-21:
I think removing the CLI would be a horrible idea, it would require '0day' changes within the configurator to use new functionality and seriously bloat the GUI with more obscure options that some people may require but many wouldn't use.

It would make more sense to drop some stuff like the OLED code, most settings can be done using an OSD, if you aren't using an OSD then shell out the $6 for a BT adapter and use a phone.

hydra, 2015-09-21:
@readerror67 not if the protocol was good, there would be no need to even change the configurator.

example configuration.

configurator: what can i change FC?
fc: here's a list
configurator: ok, change this.
fc: thanks.

given a UI in the configurator that allows all settings to be changed no changes would be required to support a new 'thing that can be changed'

NullVoxPopuli, 2015-09-26:
Ardupilot has that type of interface, where the ardupilot code gives a list of variables to the apm planner.

tracernz, 2015-09-28:
See OP/TL/LP UAVO browser. ;)

digitalentity, 2015-09-28:
@tracernz, @NullVoxPopuli, @hydra 
This issue is probably related to #968 and #1049. Basically, keep MSP, extend it so we are able to receive a list of configurable parameters and change them.

tracernz, 2015-09-28:
Yeah. Implementing something like the UAVO browser in the configurator would be a good replacement for CLI. It would give you a full tree of all the settings groups and fields and allows the writeable settings to be changed. It could be done completely in the configurator with MSP being used to send changed settings to the board. Another benefit is you would no longer need to remember the names of commands and settings since they would all be easy to find in the tree (you could even add a search box).

skaman82, 2016-01-08:
Working on this in here: https://github.com/skaman82/cleanflight-configurator/issues/59

hydra, 2016-01-08:
@skaman82 See #922.

hydra, 2016-01-08:
@skaman82 See #1049.

nveeser, 2016-02-09:
Is the general consensus that UAVO is too heavyweight? 

I only have a limited understanding from using OpenPilot but I assume that it's similar to [protobuf/releases](https://github.com/google/protobuf/releases) and [capnproto](https://capnproto.org/cxx.html)

digitalentity, 2016-02-09:
We already have MSP and we probably don't want another bi-directional protocol only for configuration.

tracernz, 2016-02-09:
UAVOs are not a telemetry layer, they are objects for passing state around both within the firmware and externally (and also storing it in the flash filesystem in the case of settings objects). UAVTalk is the telemetry layer used to access them externally.

hydra, 2016-07-21:
This is more pressing now due to extra code space required for F3 targets that have built-in OSDs.

hydra, 2016-07-21:
#2302 Will really help here.

martinbudden, 2016-07-23:
@hydra , if you are having code space problems with built in OSDs, then one easy way to save some space is the font. It's 16K in size!

It looks like it is a 256 character font, but with lots of blank characters. You could just put all the special characters together and then chop the font array at that point. I don't know how many special characters you have, but I imagine you could probably get away with a 128 character font, saving yourself 8K of ROM.

closed 2017-02-25 by hydra