cleanflight/cleanflight #924: S.BUS range issue in 1.9.0

VirtualEnder, 2015-05-20:
Just loaded up 1.9.0 on a spare board I had laying around to test with a receiver. I run S.BUS out of OpenLRSng and I got some weird behavior. Default range on S.BUS is 890-1200 with the midpoint at 1520. Everything is good around center sticks with midpoint set to 1520, but when I get below 980 or above 1100 the value suddenly jumps to 1500. Min_check and max_check have no effect on this. I tried another board with 1.8.1 on it, and the full range was visible, so this is unique to 1.9.0

VirtualEnder, 2015-05-20:
Related to #918

VirtualEnder, 2015-05-20:
Looks like the default range just needs to get increased when S.BUS is selected.

hydra, 2015-05-20:
Correct. Configure your endpoints on your tx and/or adjust the RX_min/max_usec settings

closed 2015-05-20 by hydra

VirtualEnder, 2015-05-20:
I still think defaults should be changed. If someone were to upgrade and try to fly without checking this could cause serious problems in flight.

thenickdude, 2015-05-21:
That is true of all receivers, they all need to have their ranges calibrated before the first flight.

hydra, 2015-05-21:
@thenickdude i could not agree more!

VirtualEnder, 2015-05-29:
But those values aren't way out of range for S.BUS.  They're the standard values for S.BUS.  Basically what you're saying is that cleanflight is not actually fully compatible with the S.BUS signal protocol (which specifies that range).  I have no desire to adjust my endpoints in my radio as I appreciate the extra resolution. This actually kept me out of the air on a new flash this week because I didn't have a way of adjusting the values in cleanflight at the field.  Also it did this even when I had turned OFF the failsafe feature.  It should not be horribly difficult to have the rx_min_usec and rx_max_usec automatically adjust to the compatible range for S.BUS when S.BUS is selected.  This seems pretty logical to me.

VirtualEnder, 2015-05-29:
OK, after a bit of research, I found that the issue is an incompatibility between the OpenLRS and FrSky implementations of S.BUS.  OpenLRS transmits a larger range by default.  It looks like cleanflight is already scaling the transmitted values from S.BUS, as S.BUS protocol's defined range is 0-2048, transmitted as an unsigned int.  So it looks like the scaling factor was based on FrSky's narrower range of transmissions, as FrSky S.BUS shows up as 1000-2000, which is as you intended it.  Unfortunately for me OpenLRS transmits a larger range, which I wasn't aware of.  So you're right, I should have checked my endpoints.  It was just a surprise for it to have been working for over a year, and to suddenly and catastrophically not be working.  I should have read the release notes more carefully.

slowjet, 2015-07-05:
Virtual Ender,
I am having the same issue with my Brotronics 4ch Remix outputting PPM. Im useing DTF UHF on a taranis and have never seen this issue before. So its not just related to SBUS. Im not sure how to adjust my endpoints so we still land on 1500 center stick. I'll have to look into that. I reduced the weight to 98 and its fine unless I use trim (which i dont in a multirotor) . Any further suggestions>?