cleanflight/cleanflight #925: Servo rethink

MauroMombelli, 2015-05-20:
Hi,
working on tilting quad we need to know the estimated position of servo.

to reach this goal i've modified servo structure like this:

    int16_t min;                            // servo min
    int16_t max;                            // servo max
    int16_t middle;                         // servo middle
    int8_t rate;                            // range [-100;+100] ; can be used to ajust a rate 0-100% and a direction
    int8_t minLimit;                        // range [-90;90] ; can be used to adjust a rate 0-100% and a direction
    int8_t maxLimit;                        // range [-90;90] ; can be used to adjust a rate 0-100% and a direction
    int8_t forwardFromChannel;              // RX channel index, 0 based.  See CHANNEL_FORWARDING_DISABLED

min is the PWM value to set the servo at 0° (on a 180° swipe), normally 1000us, middle for 90° (normally 1500us), and finally max for make it at 180° (2000us, normally)

then i added the field minLimit and maxLimit to soft-limit the rotation; very helpfull if you have some sort of physical obstacle. Also during our test we found out not all servo follow are calibrated, and some can't even do a real full 180°; in that case just set up the min/max Limit to 45° or 60° and use a set square to check the angule and set configure min and max PWM.

What do you think about this? this should also help a lot the gimbal sistem, but requre the user set up.

after launch i'll prepare the pull request, but meanwhile i'm extremely interested  in your opinion/idea/wathever. Actually is already working (GUI + firmware),see the screenshot
[image]

hydra, 2015-05-20:
Yeah, i think i see what you mean, since with just min/max/mid you don't know about the ANGLE of the servo right?

MauroMombelli, 2015-05-20:
exactly, it SHOULD be 1000 = 0°, 2000 = 180° on 18° servo, BUT maybe the user is gonna use a 90° servo (used with plane) and we found out weird value like 800-2700...

i think to know the angle is GREAT for gimbal, plane.. and obviusly the tilt quad ;)

MauroMombelli, 2015-12-17:
please see #1591

closed 2017-02-25 by hydra