cleanflight/cleanflight #926: body tilting compensation

MauroMombelli, 2015-05-20:
i'm working on a system to compensate the thrust based on body tilting from horizon, the only set up required by user is how much thrust stick it have to use to hover (let's call this "thrustHover"

the formula used is very simple:

   thrustStick += max(thrustStick - thrustHover, 0) * sin(angle);

what do you think? is already there something similar?

thenickdude, 2015-05-21:
This is already implemented by the `throttle_correction_value` CLI setting:

https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

MauroMombelli, 2015-05-21:
thank you, missed that, but it also very different; this model use a proportion, while my system use a sine wave, witch is what base physic teach us.

The advantage is that it doesn't care about the "hovering value" and it has a hard limit to prevent overreaction; maybe i can just create a variant and add it to the GUI.

MauroMombelli, 2015-05-29:
as this has became part of milestone, i will take care of it ASAP, so if someone is already doing that please tell me

MauroMombelli, 2015-06-01:
ok, looking at the code, the throttle_correction_value already take care of the sine compensation but it works on all axis together while for tilting quad we want to compensate only one axis and even if in acro mode.
Also i would like some input by you about basing the math on the "thrust to pich"
So i'm gonna implement it as a separate function, but still generic

MauroMombelli, 2015-06-01:
see #974 for initial implementation.
Any input is appreciated

hydra, 2015-06-10:
See #691

leothehuman, 2015-08-07:
I hope this doesn't go into 1.10 as is. It will be hard to get rid of later (as the current correction, which sucks) and it doesn't really do what it is supposed to do. I would prefer to do it the proper way, even if I have to code this myself. I already have an implementation in my test branch, but have not flown it yet.

MauroMombelli, 2015-08-08:
Its OK for me to keep this on hold and replace if better alternative

kgustine, 2016-02-14:
I currently have a mixer setup on my Taranis to add a portion of pitch and roll to my throttle output.  I can't imagine it working any better as it makes maintaining altitude an afterthought.

Obviously mine doesn't work of a Sine function, but it works pretty dang good.  I even have my S2 Pot setup to adjust the weight of the adder, but the only time I've adjusted it is when I changed my max angle from 50deg to 60deg for angle mode.

Is your variant intended for use in rate mode?  thrustStick would be the adder applied to the rcCommand[throttle] right?

MauroMombelli, 2016-02-14:
> Is your variant intended for use in rate mode?

yes, that would work also in rate mode

> thrustStick would be the adder applied to the rcCommand[throttle] right?

yes

@leothehuman  as no alternative has been proposed, should that be implemented?

leothehuman, 2016-02-15:
The correct way would be to divide throttle percentage by the cos of the tilt angle. It will allow to have constant vertical thrust independent of the attitude of the quad. Unfortunately, there needs to be the limit of the angle, we can't keep increasing throttle for any angle.

MauroMombelli, 2016-02-15:
i see that, the tilt code take into account the pitch of the body AND the pitch of the motor as they are independent, and it is limited to +-90°.
Do you think the limiting should done with a constrain as if we go over we don't have a sudden loose all the  power, witch may be seen as counter-intuitive by pilots, or should default to zero (as i think would be logical), or should mirror the +-90° range?

closed 2017-02-25 by hydra