cleanflight/cleanflight #929: FLIGHT - PID cleanup

MauroMombelli, 2015-05-20:
in pid.c and in mw.c there is the declaration of the pid's prototype. (this duplication should disappear, maybe just move the declaration into pid.h?

it is the following:

    typedef void (*pidControllerFuncPtr)(pidProfile_t *pidProfile, controlRateConfig_t *controlRateConfig,
        uint16_t max_angle_inclination, rollAndPitchTrims_t *angleTrim, rxConfig_t *rxConfig);            // pid controller function prototype

and it is "feeded" in mw.c like this:

    pid_controller(
        &currentProfile->pidProfile,
        currentControlRateProfile,
        masterConfig.max_angle_inclination,
        &currentProfile->accelerometerTrims,
        &masterConfig.rxConfig
    );

now i was going to need some data from the master configuration, and this require going to edit all the PID function, get a lot of "unused warning", and make that prototype longer.

what do you think if instead we just use the following?

    typedef void (*pidControllerFuncPtr)(controlRateConfig_t *controlRateConfig, master_t *masterConfig, profile_t *currentProfile);            // pid controller function prototype

and then just use it as

    pid_controller(
        currentControlRateProfile,
        &masterConfig,
        &currentProfile
    );

this seems a lot more clean to me!
if for you is ok i may prepare a pull request for tomorrow evening.

as side note: there is this "currentControlRateProfile" that does not belong neither to masterConfig nether to currentProfile, and this puzzle me a lot, what are the motivation of this exception?

leothehuman, 2015-05-24:
This adds a lot of unnecessary dependencies...

MauroMombelli, 2015-05-24:
Look at the linked issue, a forward reference to master_t and profile_t can
fix this.

Obviously dependence are still here, just hidden. Also I don't get why you
don't want them; OK maybe not all struct data is used, but at least if you
need something new you don't have to edit the other PID code

hydra, 2015-06-05:
Dependencies often create tight coupling and tight coupling reduces code re-use.
Dependencies make things harder to test and work on in isolation.

closed 2015-06-05 by hydra