cleanflight/cleanflight #932: System - Add software I2C

digitalentity, 2015-05-21:
I've ran into an issue with CC3D when I have MainPort occupied by GPS, FlexiPort occupied by I2C mag & baro and need additional serial port for telemetry. I opted for softserial, but it was somewhat unstable and telemetry packets were frequently dropped.
My suggestion is to have an option to have a software I2C bus to connect  additional sensors like baro & mag and leave hardware UART at FlexiPort available for other uses.
Such SoftI2C implementation can use the same pins as SoftSerial and would be rather simple due to synchronous nature of I2C bus and simplicity of protocol.

MauroMombelli, 2015-05-21:
one i2c port can have almost 127 device connected, why you don't want to use the same for all the sensor? especially baro and mag are really slow (60Hz normally) so shouldn't slow down too much the others (but then if you care about the sensor read speed you should use SPI)

digitalentity, 2015-05-21:
@MauroMombelli the problem on CC3D is that the physical port (FlexiPort) is shared between I2C and UART, you can use one the the other, not both. Basically on CC3D you can have UART + I2C or UART + UART. If I need two UARTs, I have to sacrifice I2C with all sensors.

MauroMombelli, 2015-05-21:
I get it, but sensor need high speed of communication and reliability, its a lot better to use a Sofware Serial than a I2C Serial; UART can be run a low baudrate, and this prevent a lot of interrupt overhead while communicating (if event driven) or always (if timer driven)

Also as you seen software implementation tends to be more unstable than hw based, and you don't want to happen with your flight sensor, wouldn't you? :)

digitalentity, 2015-05-21:
I2C reads are synchronous, also current implementation of I2C blocks the main loop anyways. I see no problem in implementing I2C in software it would be more stable that softserial because of every bit on I2C bus is clocked and slight delays do not affect the data transfer. Also STM32 is powerfull enough to do pure software I2C transfer at 400 kHz. 
Speaking of flight sensors, when flying long range I would prefer loosing my magnetometer and barometer but keep my OSD and telemetry working to be able to fly my craft manually.

hydra, 2015-05-21:
There is already a soft i2c implementation in cleanflight though it is unused. Check the drivers directory.

MauroMombelli, 2015-05-21:
@digitalentity  aren't even gyro and acce on the i2c line?

digitalentity, 2015-05-21:
@hydra, thanks. Than I should be able to enable them on some RC pins on my CC3D and leave FlexiPort available for other stuff.

hydra, 2015-05-21:
See

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_i2c_soft.c

you can try it on a naze by uncommenting these two lines.

https://github.com/cleanflight/cleanflight/blob/master/src/main/target/NAZE/target.h#L136-L137

It's not been used for a long time.

I should shed some light on the history of this, IIRC it was used in baseflight for a short time when there was a bug in the handlers of the hardware i2c implementation - under some conditions the bus locked up, though my memory might be wrong; this was a couple of years ago.

hydra, 2015-05-21:
@digitalentity yes in theory.  though you would likely need to make the pins configurable somehow.  I don't think we should entertain looking at this until a solution is found for #357 and #884 .

digitalentity, 2015-05-21:
@hydra thanks alot! Finding solutions for #357 and #884 might take a while and having a temporary solutions is great! It is good that I won't have to design all code from scratch. I'll test this when I have some time to spare on this - nav-rewrite (#197) takes most of my free time at my computer.

LupinIII, 2015-06-03:
I know this is not a support forum, but you can have hardware I2C on PB6/7 (I2C1), 8/9 (I2C1 remapped) or 10/11 (I2C2), even number is always the SCL, odd SDA.
So you could simply switch to I2C1 ("#define I2C_DEVICE (I2CDEV_1)") if the pins 6 and 7 are free.
For remapping I2C1 you have to add a 
"GPIO_PinRemapConfig(GPIO_Remap_I2C1, ENABLE);"
in the i2cInit() function and specify the different pins in "i2cHardwareMap[]", both to be found in "main/drivers/bus_i2c_stm32f10x.c"

digitalentity, 2015-06-03:
@LupinIII thanks for suggesting this, but PB6/PB7 as well as 8/9 are occupied by some motor pins on a CC3D, that leaves us pins 10/11 as it is now and these pins are shared with UART3. Also, CC3D has acc/gyro connected to SPI bus so implementing I2C in software won't mess up the basic flight capabilities.

closed 2017-02-25 by hydra