cleanflight/cleanflight #933: FLIGHT - mixer math improvemet (brainstorm)

MauroMombelli, 2015-05-21:
looking at the mixer configuration, we can see how this is created by a scalar (thrust) that represent the translation in the space provided by the engine, and a vector (roll,pitch,yaw) that represent the rotation provided by the engine.

This has some limitation; the engine, for example should all be provide the same thrust direction, and this should always be up.

This is not always true, for example in tilting quad, but even if you want to extends the mixer to represent aileron for plane; a vector should be used instead of a scalar, and this may open the route tho have a more complex but unified mixer logic for ANY kind of flyer.

what do you think? someone has an idea if the math involved would be too heavy/complex to be feasible?

thenickdude, 2015-05-21:
I don't think unifying planes and quads with a common mixer would bring any concrete gains. They seem to do just fine on their own. What would the advantage be?

MauroMombelli, 2015-05-21:
the advantage would be less code reuse as there is only one mixer, and increase the ability ti implement new strange system, for example tilting motor, but also VTOL, variable asset plane, or even if you loose an engine (and somehow you can detect this) can be take into account (if the array of motor/surface is editable at runtime, otherwise).

Another advantages is moving forward in the usage of more math and physic instead of putting everything on PID magic (if instead of using value from -1 to 1 real measure of distance motor to motor and motor orientation from body horizon you can even create a true model that take into account lever effect and if you add the mass you can even predict the angular/linear momentum.. but now i'm going too far)

Also to implement a new plane/multicopter/whatever, you will just produce a new configuration that describe it. One example is the difference between plane and wing, but also if you create a plane with different lift surface that harcoded one...

taileron, 2015-12-16:
there is only one Logic that covers nearly all kinds of swiveling or tilting systems it doesn´t matter if there are edfs, propellers or other control surfaces.
It needs one mixer combination inclusive PIDs (mmix and smix) covering control and stabilization for 0 degree of tilt, another mixer set for about 45 degree and one for 90 degree tilt of the power/system.

In all cases the yaw function changes to roll and reverses its direction. Roll changes to yaw wthout reversing. In 90 degree a differential tilt is necessary more tilt of the front system and less at the behind is the only way to bring the nose up and down.

Best would be mixers that slowly change its functions along the tilting procedure - but i guess thats to complicated morphing from a 0 degree profile to a 90 degree profile with endless values in between. 

In a tricopter version with three tilt servos, differential tilt of the two front servos would cover yaw at 0 degree and reversed roll at 90 ...

MauroMombelli, 2015-12-16:
not only one, but i agree one universal solution use more resources than divide and conquer approach

closed 2017-02-25 by hydra