cleanflight/cleanflight #934: Graupner Sumh.c calculation

Crashpilot1000, 2015-05-21:
Hi!
This line https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/sumh.c#L117
        sumhChannels[channelIndex] = (((uint32_t)(sumhFrame[(channelIndex << 1) + 3]) << 8)
                + sumhFrame[(channelIndex << 1) + 4]) / 6.4 - 375;
actually compiles to a "double" instead of a "float" variable due to the missing "f" behind 6.4. So it will work and is no real error :).
This is a copy paste error from Harakiri I just found out myself. The code was submitted by Copterwerkstatt http://fpv-treff.de/viewtopic.php?f=18&t=1368&start=3020#p44535 and first seen from Ernieift here: http://fpv-community.de/showthread.php?29033-MultiWii-mit-Graupner-SUMD-SUMH-und-USB-Joystick-auf-ProMicro (inside the zip-file) plz give credit to them.
It is doubtful if even a floatvariable is necessary since we store it as integer and have a lot of bitshifts going on anyway. This ((blabla * 10) >> 6) - 375 will produce the same result shorter and faster.
Note: I have no Graupner gear around.
Cheers Rob

hydra, 2015-05-21:
@Crashpilot1000 hi!

Thanks for letting me know.  I did a quick size code size comparison of the two approaches:
integer - 62940(F1)/108824(F3)
float - 62924(F1)/108808(F3)

Since we don't currently have issues with speed but we do with code size I went with the latter.

hydra, 2015-05-21:
See 11a4622a469bc822ba48dd8359b319b4ef155c6c and d9de29129e12ceb6a7c5e661fdc097f0b6ec7c62

closed 2015-05-21 by hydra

Crashpilot1000, 2015-05-22:
Thank you very much for giving the credits in the code!
Interesting I see exactly those 16Bytes gain (less) on the integer version, maybe some compiler options.
Cheers Rob