cleanflight/cleanflight #939: Failsafe - Instant signal loss detection

leothehuman, 2015-05-22:
Basically, I would suggest to detect radical changes on all channels and enable failsafe mode immediately. Also, should not throttle up while in failsafe mode. Other way is to launch a Wizard when Failsafe feature gets enabled, which would ask the user to turn off the Tx, disconnect the wires, disconnect power from Rx, and so on...

I've tested the failsafe by turning the receiver off (props off) while the craft was safely (armed) on the ground. I've noticed that the motors received half throttle before failsafe kicked in. My failsafe value is 1000, so after the detection motors stopped. I decided to test what this will does to the newbie, who accidentally turns off his Tx after arming a craft on the ground. I knew how long it will be uncontrollable and picked a safe place to test with my expendable 250. It jumped about 5m in the air from completely stopped props and has fallen a couple meters from the original position. No good... My setup is 9x + HK-TR6A V2, CF sets all channels to 1500 instantly after losing signal from Tx.

anwe79, 2015-05-25:
+1
I'm new to Cleanflight (and multicopters in general) and am just setting up my copter.
I have a 9x with the stock module + 9x8cv2 rx.
I saw rumors online about this combination not having failsafe, and when i tried it on CF, like leothehuman says above, the throttle jumps to mid_rc for a split second before going to failsafe_throttle. So for a while there I didn't think CF actually triggered failsafe because of my RX (props off, but that throttle burst freaked me out and had me shut down right away, so didn't notice that it went low afterwards)

After testing some more tonight, it seems it actually does go to failsafe, it just bumps the throttle before setting the actual failsafe value.
So that had me tricked for a while that I needed to buy an FrSky setup just to get FC failsafe, but that doesn't really seem to be true. 

Although throttle going to mid_rc seems to be a bug, no? Shouldn't it go to failsafe_throttle or keep last known signal before the failsafe delay kicks in? I'm using 1.9.0 (I'm not actually flying yet...)

mid_rc on throttle can't be assumed to be safe. On my (slightly overpowered) copter, mid_rc would have it instantly shoot several meters up in the air.

hydra, 2015-06-02:
latest 1.9.0 has fixes for the throttle bump before failsafe.  closing.  If there's still an issue please test @prodrone 's new failsafe PR and feedback on that. See #852 and #918

hydra, 2015-06-02:
Also, the title of this topic doesn't make a lot of sense based on the content of the first two comments.

closed 2015-06-02 by hydra

ProDrone, 2015-06-02:
@hydra This is actually about RX behavior on signal loss while the title of the topic suggest this is about failsafe.