cleanflight/cleanflight #940: System - Optimize configuration storage

digitalentity, 2015-05-23:
We are comming close to a limit of free space in EEPROM configuration storage. Probably we should review what is stored in EEPROM, clean out what is not needed and optimize the rest.

thenickdude, 2015-05-23:
From memory the number of adjustment ranges is crazy high and it's the bulk of the storage space. I bet that can be trimmed without consequence.

digitalentity, 2015-05-23:
Thanks, I'll look into it soon

digitalentity, 2015-05-23:
@ledvinap, you are right, MAX_ADJUSTMENT_RANGE_COUNT is 12, that's way too much. I think this can be safely recuded to 6 (6 pos switch) - this will free 6 * sizeof(adjustmentRange_t) * MAX_PROFILE_COUNT =  108 bytes of EEPROM storage. If we reduce it down to 3 ranges (3 pos switch) we instantly get additional 162 free bytes in EEPROM.

digitalentity, 2015-05-23:
Now `MAX_MODE_ACTIVATION_CONDITION_COUNT` is 40. Isn't it a bit too high? Can this be reduced as well?
`MAX_SUPPORTED_MOTORS` is 12. Is somebody flying dodecacopter with cleanflight? We don't even have a supported board with more than 6 outputs. Probably reduce this to 8, what do you think? Or even better - make this per-target configuration.
Also, across configuration many binary flags occupy whole `uint8_t`, i.e. `mixerConfig_t` contains `pid_at_min_throttle`, `yaw_direction` (a binary by meaning), `tri_unarmed_servo` and `servo_lowpass_enable`. All these can safely be packed as a bit field with minimal changes to other code and thus occupy only 1 byte instead of 4.
@hydra, @ledvinap what do you think about such changes?

digitalentity, 2015-05-23:
Altered mixer code as @Crashpilot1000 suggested in #943 (see 319dc7449f6b93cc64d979536f99ce94c4615fe0). Saves us 96 bytes of EEPROM with MAX_SUPPORTED_MOTORS of 12

digitalentity, 2015-05-24:
Applying 65d1a78da0a662af806cb8c363b45d2268413697 and 319dc7449f6b93cc64d979536f99ce94c4615fe0 actually saved 224 bytes of flash:

master:
   text    data     bss     dec     hex filename
 117268     260   13992  131520   201c0 ./obj/main/cleanflight_NAZE.elf

config-optimisations:
   text    data     bss     dec     hex filename
 117044     260   13772  131076   20004 ./obj/main/cleanflight_NAZE.elf

thenickdude, 2015-05-24:
Here are the sizes of some of the larger EEPROM members as of the current commit (93454d82a7b6f64f2a37dd28dcf7a0de2c22eee2):

```
master_t - 2012 bytes
    customMixer[12] - 192 bytes
    serialConfig - 42 bytes
    ledConfigs[32] - 128 bytes
    colors[16] - 64 bytes
    controlRateProfiles[3] - 36 bytes
    profile[3] - 1392 bytes
        pidProfile - 92 bytes (276 bytes total)
        servoConf[10] - 80 bytes (240 bytes total)
        adjustmentRanges[12] - 72 bytes (216 bytes total)
        modeActivationConditions[40] - 160 bytes (480 bytes total)
```

IMO it would be enough for the moment to reduce the number of modeActivationConditions, I don't think anybody needs 40 of these.

Crashpilot1000, 2015-05-24:
Hi just scooped a little through your code. My "SpecialIntegerRoundUp" is just a substitute for "lrintf" because I had a strange encounter with an older GCC version where lrintf didn't work as expected.
The cliCMix function has a few possibilities to shorten it. First of all you can take advantage of the integer nature of the mixer and omitt the float point operations (shorter) like: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/ef80ee0abdfa6f514a25fca9117c08f722a6fff9/src/cli.c#L643
Text can be shorter ;) https://github.com/Crashpilot1000/HarakiriWebstore1/blob/ef80ee0abdfa6f514a25fca9117c08f722a6fff9/src/cli.c#L700 .
Counting up numbers not needed here: https://github.com/cleanflight/cleanflight/blob/319dc7449f6b93cc64d979536f99ce94c4615fe0/src/main/io/serial_cli.c#L680 because there is already a "i++" so this is shorter: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/ef80ee0abdfa6f514a25fca9117c08f722a6fff9/src/cli.c#L633
The same optimization can be done here as well: https://github.com/cleanflight/cleanflight/blob/319dc7449f6b93cc64d979536f99ce94c4615fe0/src/main/flight/mixer.c#L334
This line is actually an error imho: https://github.com/cleanflight/cleanflight/blob/319dc7449f6b93cc64d979536f99ce94c4615fe0/src/main/io/serial_cli.c#L722 because a negative motornumber input by the user in cli can do an out of bound write. This problem also exists in Baseflight. This is the fix: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/ef80ee0abdfa6f514a25fca9117c08f722a6fff9/src/cli.c#L673 .
12 motors seem to be a waste but will probably not change the EEPROM space. BTW even Dominic Cliftons new flightcontrol is advertised as 8PWM http://seriouslypro.com/spracingf3

 Cheers Rob

hydra, 2015-05-24:
Adjustment ranges get eaten up with 2 pots and a 3pos for rate profile selection.  I need 7 per pot and 1 for the rate profile.

I rarely find myself needing all the mode ranges however.

Thanks for all the comments and suggestions so far, just skimming this morning.

digitalentity, 2015-05-25:
How many mode renages do we really nead? I think this can safely be reduced down to 20. I have used 8-10 mode ranges tops (usually less than that)

hydra, 2015-05-26:
yes, 20 seems like a good number, all of my aircraft builds use less than that and i have built about 25 aircraft now.

A bit of history:  when we migrated from the old 3-position switch code (L/M/H checkboxes) we didn't know how many checkboxes were being checked so it was set generously to 40.  We now know that number is too high.

Be aware we do want to add a switch to allow blackbox to be enabled/disabled during flight which will likely cause users to use more ranges.

hydra, 2015-05-29:
Reduced to 20 mode ranges in 6f3c16fe6c136491d26208b4fcddfbf57c2efb27.

I'd be open to going to 25 if people need more.  Feedback encouraged.

digitalentity, 2015-05-30:
@hydra thanks. What do you think about changing MAX_MOTOR_COUNT to 8 (1f84d590cd8f4567c864764d22e09177aaced810) and modifying mixer code to use int16_t in storage and map that to float upon mixer initialisation (ab9bc8cb551159ccdd01724180529dd91f6ac7f4)? It this worth merging?

Crashpilot1000, 2015-06-06:
I replaced the sin/cos functions with a pimped Taylor series that saved me 4248 Bytes of EEPROM.
Benefits besides this: still very precise: maximal absolute error = 0,00000024 Rad. and a speedgain of 13%. I think that is considerable. Code is here: https://github.com/Crashpilot1000/HarakiriWebstore1/commit/62511c6dd5b0e2579554e1a3623c4c9044268369
Cheers Rob

Crashpilot1000, 2015-06-06:
@ledvinap Thank you for the Chebyshev - Sin link! I have updated the version here: https://github.com/Crashpilot1000/HarakiriWebstore1/commit/6e8e8127f7ec8a0ae07778e7ab360a8a86b2f84e

digitalentity, 2015-06-07:
@Crashpilot1000 Thanks for the faster sin/cos, they will save us some flash as well as CPU cycles. @hydra do you think is this worth doing for CF?

Crashpilot1000, 2015-06-08:
@digitalentity @ledvinap Thank you very much for the nice feedback. In the last commit I made the method selectable https://github.com/Crashpilot1000/HarakiriWebstore1/commit/3fc18b51c18d3973c15fc2e68d1aee048d772b70 . I go with the more accurate variant for now, since it's about saving EEPROM space in my case. The benefit besides still being faster is that it has a constant execution time and not so much time jitter like orig sinf()  (on non FPU STM). I know it's still a taylor series but those numbers: http://lolengine.net/blog/2011/12/21/better-function-approximations may be better for double precision but they are not for single precision float. If someone here could come up with better numbers for sinus (+- 1/2Pi range) by using the "lolengine" Remez c++ program that would be awesome!

ledvinap, 2015-06-08:
Here are sin(x) approximations on `<-PI/2 - PI/2>` generated using http://lolengine.net/wiki/oss/lolremez

order 5, error is ~ +- 1e-4
```
a1=9.998918212558108928638337369325221875413e-1
a3=-1.659601165408789889714685427729180272215e-1;
a5=7.602903343369351160819273949501310306040e-3;
```

order 7, error is ~ +-6e-7, about 10x float precision:
```
a1=9.999966159080027730784793134643817634584e-1;
a3=-1.666482838189505682902228394288919128857e-1;
a5=8.306325227159893961305887257007410988819e-3;
a7=-1.836365397694678520672878653316690150932e-4;
```

```
y = x .* (a1 + x.*x .* (a3 + x.*x .* (a5 + x.*x * a7)));
```

Both approximations minimize relative error.

ledvinap, 2015-06-08:
And coefs with a1 = 1:
```
    a1 = 1
    a3 = -1.666568107299974993500583665040916571532e-1
    a5 = 8.312366210460148540630089477385518834149e-3
    a7 = -1.849218155809176638927483742439855005475e-4

    a1=1
    a3=-1.660786242169313753522239789881311891420e-1
    a5=7.633773374658546665581958822838108771028e-3
```

Accuracy is a worse (~40% ? Just guessing from graph ... ) , but you save one multiply ...

Crashpilot1000, 2015-06-09:
Hi @ledvinap! Thank you very much for running the numbers! However I see my longer taylor series having a smaller absolute error. Probably I am fishing at the resolution margin of the single precision "float" specs but could you do a series with 5 numbers to check out?

ledvinap, 2015-06-09:
```
    a1=1
    a3=-1.666665709650470145824129400050267289858e-1
    a5=8.333017291562218127986291618761571373087e-3
    a7=-1.980661520135080504411629636078917643846e-4
    a9=2.600054767890361277123254766503271638682e-6
```

Crashpilot1000, 2015-06-09:
@ledvinap Thank you very much!!!!!!!!!!!!!
This saves me two taylor steps and has the same accuracy! Measured max absolute error: 0,0000069 Degree. Nice!

ledvinap, 2015-06-09:
I still think that there is seventh order (A1-A7) approximation is more than adequate, but both will do ...

Crashpilot1000, 2015-06-09:
@ledvinap I ran a comparison between the original double and the original normal float sin ("sin() vs sinf()") and also found an max abs error of 0,00000012 Rad (in single precision float point space) so the a1-a9 is definitely working on the brink of the float resolution. If you want to optimize for speed and not for eeprom you can have different precision versions for every task (like most precision for imu, lowest precision for headfree and medium precision for bearing calculations).

ledvinap, 2015-06-09:
IMO functions from libm are accurate to last bit, including correct rounding.

In cleanflight, version with 1e-4 error will be perfectly OK, calculations are based on data with noise or variable resolution is limited anyway.
Only place where accuracy may be important is filter coefficient calculation, ant it is not used in current code.
But using ninth order approximation wont cause any problems, so if you feel better with it ...

thenickdude, 2015-07-07:
Hey guys could you create a new issue / PR for the Taylor approximation for trig functions? It sounds like something that would give a nice performance boost.

Crashpilot1000, 2015-07-07:
Just a few lines of code that are easily implemented.
The working code is here: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/mw.c#L1235
The predefined constants are here: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/board.h#L20
So if this project is in need of saving a few (>4) Kilobytes EEPROM and having an almost equal precision Sin/Cos function that is 28% faster and has a way more constant execution time (less additional timejitter) it is a breeze to implement for the coders that have an Cleanflight oversight. It would take me hours to set up a proper pullrequest here that may be rejected.

digitalentity, 2015-07-07:
@thenickdude, @Crashpilot1000
I'll look into it in a few days. I like the possibility to save a few cpu cycles and some memory.

thenickdude, 2015-07-07:
Understandable, @Crashpilot1000, thanks for the code! @digitalentity that would be great.

digitalentity, 2015-07-07:
Done some testing in my fast-trig branch here: https://github.com/digitalentity/cleanflight/tree/fast-trig

current master:
   text    data     bss     dec     hex filename
 119204     364   12416  131984   20390 ./obj/main/cleanflight_CC3D.elf

Tested with looptime=0, Harakiri PIDC, LEVEL mode
Measured average cycle time=1015ms, sigma=38, min=889, max=1129

#define FAST_TRIGONIMETRY
   text    data     bss     dec     hex filename
 118596     364   12412  131372   2012c ./obj/main/cleanflight_CC3D.elf

Tested with looptime=0, Harakiri PIDC, LEVEL mode
Measured average cycle time=1011ms, sigma=23, min=926, max=1082

Using FAST_TRIGONIMETRY, looptime has much less jitter. Integer-based PID controllers will surely benefit from this. Also I was able to save 608 bytes of flash and 4 bytes of RAM.

digitalentity, 2015-07-07:
@thenickdude please see if this is what you asked for. Looks like performance boost is not that big, but cycle time is more stable now. Probably my navigation code will benefit from this more as it uses alot more sin/cos calculations.

digitalentity, 2015-07-07:
Now maths_unittest is failing. Actual result is seamingly the same, but some tiny error is enough for test to fail:
```
[ RUN      ] MathsUnittest.TestRotateVectorWithNoAngle
unit/maths_unittest.cc:121: Failure
Value of: b->X
  Actual: 1
Expected: a->X
Which is: 1
[  FAILED  ] MathsUnittest.TestRotateVectorWithNoAngle (0 ms)
[ RUN      ] MathsUnittest.TestRotateVectorAroundAxis
unit/maths_unittest.cc:121: Failure
Value of: b->X
  Actual: 1
Expected: a->X
Which is: 1
[  FAILED  ] MathsUnittest.TestRotateVectorAroundAxis (0 ms)
```

giryan, 2015-07-07:
Looking at the unit tests they are comparing floating point numbers with equality, this is generally a bad thing because there can be multiple representations of the same number in a float( e.g. -0.0f & +0.0f)
and some numbers can't be represented exactly(e.g. 1)
In this instance I think that the tests in maths_unittest.cc that are comparing floats should use EXPECT_FLOAT_EQ instead of EXPECT_EQ.

giryan, 2015-07-07:
I can't submit a change while I'm at work, but I'll have a go tonight if it's not done already :)

digitalentity, 2015-07-07:
@giryan I'll submit a patch for math unittests as a part of #1092

akcom, 2015-07-09:
I'm curious,what's more important: saving space or cycles?  Because it seems that using a lookup table would be incredibly fast.  This is a relatively good reference: https://jfdube.wordpress.com/2011/12/06/trigonometric-look-up-tables-revisited/

ledvinap, 2015-07-09:
Did you check the precision table and compare it to polynomial results?

closed 2015-10-31 by digitalentity