cleanflight/cleanflight #943: System - Review and optimize EEPROM configuration storage

digitalentity, 2015-05-23:
We are comming close to a limit of free space in EEPROM configuration storage. Soon we'll hit the limit when trying to add a new cool feature that requires configuration storage. Probably we should review what is stored in EEPROM, clean out what is not needed and optimize the rest.

Crashpilot1000, 2015-05-23:
Yep you can optimize the mixertables to int16_t and that gives plenty of resolution. So you have a storage format https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/mw.h#L233 and the "after-power-up" used format https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/mw.h#L241. Then you need a conversion/bitshift factor: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/mw.h#L28 and take this into account for the mixer and its readout: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/mixer.c
Also the cli custom mixer stuff must unfergo slight changes: (starting here:) https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/cli.c#L607
This will nearly half the mixer storage eeprom usage. The motorvalue might be omittable as well but it's currently used for detecting the end of a custom mix table, so I didn't change that.
Another thing is the cli parameter list. The minvalue can be reduced to int16_t and rearranging the bytesequence in the struct helps the compiler as well (i don't know the gcc commands, maybe there is a better way) https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/cli.c#L15 and https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/cli.c#L154. Changing the valueTable this way saved me 1432 Bytes of EEPROM. Cleanflights table is much shorter but you should be able to shove off *some* bytes as well. Hope anything of my blubber helps.
Cheers Rob

Crashpilot1000, 2015-05-23:
This https://github.com/cleanflight/cleanflight/blob/master/src/main/common/typeconversion.c#L195
can be reduced to this: https://github.com/Crashpilot1000/HarakiriWebstore1/blob/master/src/cli.c#L373
Because we don't deal with user inputs like 2.4E-02 so omitting the exponential part is ok for our cli input parsing.

digitalentity, 2015-05-23:
@Crashpilot1000 great idea, thanks!

digitalentity, 2015-05-23:
This issue is a duplicate of #940 (opened by my mistake due to unstable internet connection)

closed 2015-05-23 by digitalentity