cleanflight/cleanflight #944: Code Clean - move include to header file

MauroMombelli, 2015-05-23:
having the include in the .c file create a lot of mess, as if you have to include a header, then you'll also need to include all of its dependency, and so on. (see #925 )

This can also fix some big problem like the funtion mixerUseConfigs
is a function declarated and initailizated in flight/mixer.c
BUT is is also declared into config/config.c and flight_mixer_unittest.cc
this is really bad, as when i added a filed to the function i was lucky to update config.c, so the program is working, BUT as i miss flight_mixer_unittest.cc i had segfault on the test.

Obviously this situation can be fixed moving the declaration of mixerUseConfigs to the header file, so it's shared and does NOT need to be re-declared, avoiding this problem (if you forget to update one of the caller you have a compile time error instead of a segfault, maybe while flying...
BUT
As it depends on a lot of structure, and their include is in .c, it cannot be moved to .h withut moving all this include into header.

I look forward to have your opinion and the eventual "ok" on this issue, as maybe it break something i don't know.

ledvinap, 2015-05-23:
You can use structure forward declaration in header file:
```
struct servoParam_s;
struct gimbalConfig_s;
...
void mixerUseConfigs(
struct servoParam_s *servoConfToUse,
struct gimbalConfig_s *gimbalConfigToUse,
...);
```

Then just make sure that structure is defined correctly
```
typedef struct servoParam_s {
    int16_t min;                           // servo min
    int16_t max;                            // servo max
    int16_t middle;                         // servo middle
    int8_t rate;                            // range [-100;+100] ; can be used to ajust a rate 0-100% and a direction
    int8_t forwardFromChannel;              // RX channel index, 0 based.  See CHANNEL_FORWARDING_DISABLED
} servoParam_t;
```

Cleanflight is inconsistent in this regard - not all struct are defined with `_s` suffix in struct namespace ... it is probably OK to fix this for structures that  need forward declaration.

Personally I avoid typedef struct for this reason ...

ledvinap, 2015-05-23:
https://www.kernel.org/doc/Documentation/CodingStyle chapter 5 ...

MauroMombelli, 2015-05-23:
@ledvinap i'm not sure if I understood correctly, i mean, then i have to declare the structure pain attention is correct with the declaration around the others file.. i'm just moving the problem from function to struct, right? and this mean more code than just duplicating the function declaration, aka more possibility of error. Or I'm missing something?

personally i avoid typedef as much as possible, if you look at my repo you wont find it. but i think its better follow to the code style of the project, and if i really think something is bad like in this case, that's what i do.

ledvinap, 2015-05-23:
You use forward-declaration to tell compiler there is `struct x`, but this type is opaque. Actual structure declaration can be in other header file or even .C file, you don't need it to declare function (or as long as you are using only pointer to structure).

MauroMombelli, 2015-05-23:
ah, ok, i don't have to re-declare the structure's fields, the compiler do the magic even if no header is directly linked. fixed that function as  #945 i'll fix others as i'll spot them

MauroMombelli, 2015-05-25:
@ledvinap sorry to bother but this technique is new to me. Trying to fix #930 it is very helpful to move things from .c to .h, but .c still have to have all include to the struct and its nested element, otherwise if you try to use forward-declaration even in .c you loose the ability to access nested structure.
we could import directly the nested structure used and forward-declare the others, but seems a more messy solution than just import directly everything.

ledvinap, 2015-05-25:
The idea is to keep header file free of unnecessary dependencies/definitions - just forward-declare structure names that are important to call given function. 
Including this header file is enough if you want to pass pointer to structure to function (and store this pointer in variable ect.). The internals of structure (and it's size) are hidden, the structure is opaque (which is usually good thing). You can use accessor functions (that take structure pointer as argument), that way you can work with structure without need to know it's contents (encapsulation may be implemented this way in C).

Any .c that want to use structure internals has to include .h file that contains full structure declaration. This limits inclusion of 'detailed' header only to C files where it is necessary ...

MauroMombelli, 2015-05-26:
but look at master_t, i want to import directly it to pid.c, so with this logic i should import only "config_master.h" right?
BUT
as inside the master_t there are struct declared using the typedef definition (for example motorMixer_t instead of motorMixer_s), i can't use forward declaration (well i could redefine the typedef, but this comes from the C++ and not from the C, and generate a warning)

and the solution is to import all the header dependency.

ledvinap, 2015-05-26:
Yes, you can't declare structure without knowing size of all it's members ...
So pid.c still needs all structures that comprise master_t.

It may be possible to defined assessor function for every master_t member (and that use forward declarations), but I'am code complexity will be about the same .. At least with  LTO used in compilation this won't create any overhead ...

hydra, 2015-06-02:
i don't think there's anything to do here. closing this issue, feel free to continue to comment and discuss here however.

closed 2015-06-02 by hydra