cleanflight/cleanflight #947: Erratic Rx signal from FrSky L9R PWM

rcflo, 2015-05-23:
Upgraded form 1.8.1 to 1.9, same exact setup except for reflash:

flip32+ w/ gps enabled
oneshot enabled
FrSky L9R PWM Rx

As the title states, the receiver throws in the configurator go crazy, or don't respond at all.  Sometimes you get good behavior for a ten seconds or so upon repowering up, i did a couple times.  One time, i had enough normal response time to arm the board, it then went crazy and the throttle channel jumped to ~1400, good thing i had the quad attached and fingers out of the way, as the 8in props were on.  had to unplug lipo.

reflashed 1.8.1 and dumped same CLI, no issues again.

thenickdude, 2015-05-24:
1.9 defaults to having FAILSAFE engaged. So if your channels are significantly out of the 1000...2000 range that Cleanflight requires you to calibrate your Tx to, failsafe will be triggered. Try disabling the FAILSAFE feature and see if the problem goes away.

rcflo, 2015-05-25:
Thanks for the reply.
Failsafe is checked off. Rx failsafe is on but that shouldn't effect
anything. Throws go to 2010-2015 on the high end

thenickdude, 2015-05-25:
Please post the output from the "dump" command on the CLI tab on the malfunctioning system.

hydra, 2015-05-29:
And also two screenshots of the receiver tab with your sticks all at LOW and all at HIGH.

hydra, 2015-05-29:
@thenickdude and @rcflo - failsafe setting doesn't affect new RX loss detection

rcflo, 2015-05-29:
Hey guys, sorry for the delay I had an esc and motor go on me during tuning
that particular rig back on 1.8.1 and haven't gotten it back in the air
until today.  Now that it works, I will try 1.9 again :)  I'll send a dump
and a short video of the behaviour if i notice it again.

Best,
Flo

hydra, 2015-06-02:
ok, i will close this. please re-open if there's an issue.

closed 2015-06-02 by hydra