cleanflight/cleanflight #949: Code clean - unify motor count definition

digitalentity, 2015-05-25:
Currently we have 3 defines related to maximum motor count:
https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.h#L20
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.h#L20
https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.h#L23
All three of them define the same number.
Is there a reason for keeping these 3 `#define`s or they can be replaced by 1 target-specific?

hydra, 2015-05-26:
Primarily the single responsibility principle applies here, code re-use is also promoted.  Same goes for servos too btw.

The driver code is the limiting factor, not the FC code.  A pwm driver could be written that only supports 4 motors for example.  A pwm driver could be written that can handle more motors or servos than the FC code would ever possibly use.

They are separate concerns, thus separate defines.  IIRC The `BUILD_BUG` macro is used to prevent developer mistakes at compile-time.

Likely some of this will get cleaned up when we look at #357 and #884 

At the moment it's fine unless you have a good reason to change it.

closed 2015-05-26 by hydra

digitalentity, 2015-05-27:
Thanks for explaining this.