cleanflight/cleanflight #954: Sensors - Should call batteryInit() whenever a battery is connected

DarkVegetableMatter, 2015-05-26:
If the FC (Naze in my case) is powered up by USB then the battery is plugged in then the number of cells is incorrect. 

batteryInit() is called once by main and should be called whenever VBatt rises above some 'battery present' threshold

JohnOCFII, 2015-05-26:
I've seen that behavior too.  I didn't really care until I set up telemetry monitoring on my Taranis and now it squawks at me all the time until I re-power the FC with the Battery before I plug in the USB.  This is with a DragonFly32 Pro (Flip+/Naze32 Full clone).  Seen in 1.8.1.

DarkVegetableMatter, 2015-05-26:
It's a problem because it means batteryCellCount, batteryWarningVoltage and batteryCriticalVoltage are calculated for a 1S pack and that's not helpful when flying on 3S ;) 

This should fix it. 
-When there's no battery connected the CLI status command reports 0S
-When a 'battery plug in event' is detected (Vbatt rises to > 1V && battery was unplugged) the number of cells and warning/critical voltages are recalculated
-It takes a while for the filter cap to discharge after a battery is disconnected, until that level drops below 1V the battery is considered to still be present. This means, for example, if you start with a 2S pack, remove it and immediately plug in a 3S pack the number of cells and warning/critical voltages are NOT recalculated. I think I can live with that though, it's a very corner case

JohnOCFII, 2015-05-26:
Nice solution.  Thanks.

hydra, 2015-05-26:
@DarkVegetableMatter Good work. Did you open a PR for this or do you want to update your existing one (#952) with the commit https://github.com/DarkVegetableMatter/cleanflight/commit/77197aceccb1b7096efd83827b37a9fcc23a46d1 ?

If you have a PR with two commits that build on each other i can review and test them.

closed 2017-02-25 by hydra