cleanflight/cleanflight #956: Sensors - Replace averaging filter for VBatt monitoring

DarkVegetableMatter, 2015-05-27:
The current idea is to:
-Blackbox logging of raw VBatt
-10Hz LPF applied to VBatt for other consumers (OSD & VBatt warning) 

How do we feel about some hysteresis on the warning and critical thresholds?

DarkVegetableMatter, 2015-05-27:
What should the VBatt monitoring interval be?

DarkVegetableMatter, 2015-05-27:
-Removed the CLI VBatt interval cmd
-Reverted VBatt interval to 6 * 3.5ms (what it was originally I believe)
-Replaced averaging with LPF at 10Hz
-Blackbox VBatt sampling is as before

Let me know what you think

hydra, 2015-05-27:
will do, thanks for your work @DarkVegetableMatter 

Any chance you could write some new tests and update the old tests if you haven't already?

DarkVegetableMatter, 2015-05-27:
Will do (looks like I broke the test build anyway).

DarkVegetableMatter, 2015-05-27:
Hmm, fixed my failure but seems something else was broken when I cloned the repo

g++ -g -Wall -Wextra -pthread -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/gtest/inc -std=gnu++98 -Iunit -I../main -c unit/telemetry_hott_unittest.cc -o ../../obj/test/telemetry_hott_unittest.o
In file included from unit/telemetry_hott_unittest.cc:37:0:
../main/io/serial.h:49:23: error: duplicate ‘const’
 extern const uint32_t const baudRates[];
                       ^
../main/io/serial.h:64:37: error: duplicate ‘const’
 extern const serialPortIdentifier_e const serialPortIdentifiers[SERIAL_PORT_COUNT];
                                     ^
make: *** [../../obj/test/telemetry_hott_unittest.o] Error 1

Tut tut ;)

thenickdude, 2015-05-27:
That's a bug that only shows up on g++ (tests are normally built using clang), will be fixed in #941

DarkVegetableMatter, 2015-05-27:
Hmm, well I need some help then 

$ CC=clang make
...
In file included from ../main/telemetry/hott.c:70:
../main/io/serial.h:49:23: warning: duplicate 'const' declaration specifier [-Wduplicate-decl-specifier]
extern const uint32_t const baudRates[];
                      ^
../main/io/serial.h:64:37: warning: duplicate 'const' declaration specifier [-Wduplicate-decl-specifier]
extern const serialPortIdentifier_e const serialPortIdentifiers[SERIAL_PORT_COUNT];
                                    ^
2 warnings generated.
g++ -g -Wall -Wextra -pthread -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/gtest/inc -std=gnu++98 -Iunit -I../main -c unit/telemetry_hott_unittest.cc -o ../../obj/test/telemetry_hott_unittest.o
In file included from unit/telemetry_hott_unittest.cc:37:0:
../main/io/serial.h:49:23: error: duplicate ‘const’
 extern const uint32_t const baudRates[];
                       ^
../main/io/serial.h:64:37: error: duplicate ‘const’
 extern const serialPortIdentifier_e const serialPortIdentifiers[SERIAL_PORT_COUNT];
                                     ^
make: *** [../../obj/test/telemetry_hott_unittest.o] Error 1

$ clang --version
clang version 3.6.1 (tags/RELEASE_361/final)
Target: x86_64-unknown-linux-gnu
Thread model: posix

thenickdude, 2015-05-27:
Try `make CC=clang` rather than adding it to the environment like you tried.

DarkVegetableMatter, 2015-05-27:
Tried that originally, same result

thenickdude, 2015-05-27:
Hm, you don't have a shellscript on your path called "clang" that invokes
gcc instead, do you? The reverse is common on macos.

DarkVegetableMatter, 2015-05-27:
no, also clang's bin directory is the first thing in my path. I have updated the tests as far as I am able and will do some tests on hysteresis with a PSU later

DarkVegetableMatter, 2015-05-27:
Tested with PSU and hysteresis is working fine. Beeper not so much...

DarkVegetableMatter, 2015-05-29:
Ok, I flew 15 packs with the new battery monitoring code and it seems to work fine. I still get a low battery  warning on punch out near the end of the flight. More testing tomorrow

ctzsnooze, 2015-06-03:
I think that these changes would be a good improvement.  I have LEDs flashing when battery is at warning or critical. But they flash on short transients when lowish, and do that sometimes for the most brief of events.  So an LPF for VBat sounds good to me.

DarkVegetableMatter, 2015-06-04:
That's what the hysteresis is for ;)

ledvinap, 2015-06-05:
@DarkVegetableMatter : integrating undervoltage with threshold will be event better - alarm will trigger fast if voltage is really low, but it will tolerate transient undervoltage conditions.

Or even better use current measurement (or estimation) and subtract battery internal resistance + wire resistance from measurement. That way unloaded battery voltage may be estimated (at the cost of new 'internal resistance' parameter ), possibly allowing very accurate battery assessment. But I am not absolutely sure that li-ion  chemistry works this way ...

hydra, 2015-08-04:
Merged.

closed 2015-08-04 by hydra