cleanflight/cleanflight #957: Code cleaning

mgirlando, 2015-05-27:
Hi all,
I am new to this project (I'm working to implement lidar avoidance with Lidar-Lite), so I just learning the structure of the software and for that I had to search for the architecture (I wrote for myself a little design doc)
At the moment, I have a concerns about readability in main.c :
* all compiles well because everything is included in main.c and that should not. As all the units, main.c should only include the files it needs for itself.
* main.c declares all the init functions but they should be forward declared in their respective header files.
* main.c is really too big as the only function that is important in main.c is the main proc.

What I propose is : 
* to split main.c to put inits out in a main_initialisation.c for example
* to move the declarations of init funcs in their header files
* to remove unnecessary include files

I already did most of those modifications and look forward for your returns to make the corresponding pull request.

PS : Sorry for my english, if there are any chance of misunderstandings please tell me.

Trussroads, 2015-05-30:
@mygirlando I happily look forward to your results with Lidar-Lite integration, I have been hopeful someone would tackle this, as I have not got the skills myself, but believe it will be an important step for us at hobby level and to begin to move away from the other less reliable measuring devices. (Also Lidar-Lite is only around $80 now which is ok..) 
Thank you for your development work for this system in advance. ;D

mgirlando, 2015-05-31:
I am waiting for the device at the moment but am already working on the hardware and interface. Even if lidar lite is very slow (tipicaly 50hz, 100 with shorter distance) it is possible to have a full front coverage with up and down securty also.
I did all the code cleaning I said and am waiting to fly with it before doing a pull request but the weather is terrible at the moment in France..

hydra, 2015-06-02:
There would be nothing left in main.c if you removed the initialisatiion code, that is the single responsibility of main.c.

The initialisation order of the system is critical and must be as clear as possible, any changes that improve it are welcomed.

updating the header files so that functions don't need to be redeclated in main.c would be useful, but ensure that the dependencies of other source files are not changed if you tackle that.

Closing this as there's nothing to do, but if you work on this please update this ticket with a comment and reference it in any pull requests.

Please use another issue/pr for any lidar work.

closed 2015-06-02 by hydra

mgirlando, 2015-06-03:
Dear Dominic,
I did all the cleaning and did not change anything in the initialisation proc (too dangerous). I corrected almost 60 header files and now the order of inclusion is not important. I did not change anything in fact in functionnalities and  It seems to work as before (1.9.0). I fly with it this afternoon and toomorow. If it's ok I wiil propose a pull request on satturday (6-6-15).
Regards