cleanflight/cleanflight #958: Enhancement - Add MSP_ALARM to MSP

DarkVegetableMatter, 2015-05-27:
At the moment MWOSD et al request VBatt from the FC and then do the maths (again) to alert the user to a low battery condition. If the OSD can fetch VBatt _and_ BatteryState then

-Only have to configure it once (in CF)
-OSD doesn't care if you change number of cells, FC sorts it out

We can't break compatibility by changing the size of a message (i.e. add an 'alarms' field to MSP_STATUS) so how about a new MSP_ALARMS message? Only MSP clients that know about MSP_ALARMS will request it and know how to handle it, everything else doesn't care.

This could be useful for RSSI too. I guess a UINT8 would be enough for 3 flags

DarkVegetableMatter, 2015-05-27:
Maybe GPS lock lost too?

JohnOCFII, 2015-05-27:
Remember that some people use MWOSD without connecting to a flight controller, so changes in CF may not equate with many savings for the MWOSD developers.

hydra, 2015-05-29:
I'm wondering if there should be a more generic system state MSP command... Hmm.

DarkVegetableMatter, 2015-05-30:
I think so too. Current system where FC & OSD have to agree on limits and alarms is broken. Much better if OSD has to know nothing and FC has all the smarts

closed 2017-02-25 by hydra