cleanflight/cleanflight #959: Unable to configure Softserial in Configurator

udvlp, 2015-05-27:
since a370d60 I'm not able to configure softserial ports in configurator. They are not listed on the Ports tab and the feature SOFTSERIAL becomes unchecked after reboot.

My flightcontroller is a Naze32 rev5 and all settings are defaults.

hydra, 2015-05-27:
Enable softserial first. Reopen if you have issues

thenickdude, 2015-05-28:
>feature SOFTSERIAL becomes unchecked after reboot

Are you sure your receiver isn't set to parallel PWM? Parallel PWM takes up the softserial pins so SOFTSERIAL can't be enabled when the receiver is set to parallel PWM.

udvlp, 2015-05-28:
Thank you, @thenickdude & @hydra. I now understand that features can disable other features that are conflicting.
 
So this is a documentation/support issue. Before a370d60, I was able to configure ports on the ports tab and then all features on the Configuration tab.

Now, as a new user, coming from default settings, I have to follow the following steps:

1. On the Configuration tab, enable SOFTSERIAL and SERIALRX (SBUS). Save and Reboot.

2. On the Ports tab, configure Blackbox on Uart1, GPS on softserial etc.

3. On the Configuration tab, enable the rest of the features (Blackbox, GPS...)

All right, problem solved for me, but perhaps this could be documented by someone who speaks English fluently.

Thanks again.

hydra, 2015-05-28:
@udvlp It is documented

https://github.com/cleanflight/cleanflight/blob/master/docs/Serial.md#constraints

closed 2015-05-28 by hydra

thenickdude, 2015-05-28:
As for Blackbox specifically, you can enable that before its port is
configured if you like.

udvlp, 2015-05-28:
@hydra, I don't want to be nit-picky, but the following sentence from Serial.md: "Configure serial ports first, then enable/disable features that use the ports."  is misleading and (since a370d60) not true anymore for the two specific features SOFTSERIAL and RX_SERIAL.

Propitiation, 2016-01-30:
I think this might be the same issue I am having with my quad. Cleanflight configurator 1.1 will not allow me to enable serialRX in ports tab under UART2, it just won't save. I have tried enabling sbus and serialRX in config tab first and second and still nothing. My quad flies with baseflight so I know its not a wiring thing. I'm using a naze32 rev5 from Armattan.

thenickdude, 2016-01-30:
Take a screenshot of the ports tab before you're trying to save it. (You
can attach it here by dragging and dropping)

Propitiation, 2016-01-31:
[image]
 
if i turn on UART2 it saves, but Serial RX never saves. I've tried doing config tab first second still nothing.

Propitiation, 2016-01-31:
[image]

Sorry, this is what it looks like before I hit save, the first image I posted is what it looks like after I hit save.

thenickdude, 2016-01-31:
By "turn on UART2" you mean ticking serial rx for UART2, right? Because
there's no switch to "turn on" that port. You aren't enabling MSP for UART2
by mistake are you?

thenickdude, 2016-01-31:
Yeah untick MSP for UART2, you can't run MSP and SerialRX on the same port,
which is why the configuration gets rejected.

Propitiation, 2016-01-31:
You Rock! yep that was my problem, I figured it would be something stupid... I don't know why cleanflight doesn't do this automatically when you click sbus...

edgeflyer30, 2024-02-18:
Where can you click sbus?