cleanflight/cleanflight #960: CLI does not accept baudrates >115200

udvlp, 2015-05-27:
Example:

# set serial_port_1_blackbox_baudrate = 5
serial_port_1_blackbox_baudrate set to 5

# set serial_port_1_blackbox_baudrate = 6
Value assignment out of range

# set serial_port_1_blackbox_baudrate = 7
Value assignment out of range

hydra, 2015-05-28:
This is by design. Is there some reason you want to use or need a higher baudrate?

thenickdude, 2015-05-28:
Ah @hydra I think you mean it's by design that MSP doesn't run faster than 115,200. Blackbox should be able to but currently has the wrong maximum set in serial_cli.

closed 2015-05-28 by thenickdude

udvlp, 2015-05-28:
I've noticed this problem when I tried to restore a CLI dump with Blackbox configured at 250kBd. Thanks for the fix, @thenickdude. I'm new to Git, next time I'll try to create a pull request myself.

hydra, 2015-05-28:
@thenickdude yes. I missed the blackbox reference because the problem description wasnt clear.  Thanks for the fix.

commit 3ddea2a8 (2015-05-28): Fix blackbox baudrate upper bound on CLI

Closes #960