cleanflight/cleanflight #966: 1.9  (27db9ad) 2015-05-30 generates bogus serial port

stronnag, 2015-05-30:
Using the build "Cleanflight/NAZE 1.9.0 May 30 2015 / 07:29:07 (27db9ad)" and enabling soft serial results in a bogus serial port appearing in CLI dumps. The new serial_3 appears to be bogus, as nothing works via it, whilst soft serial 1 is on the new serial_4 (and in limited testing, appears unreliable).  Also noted on rcgroups by other users. Dunp fragments below:

**Previous working dump:**
```
# mwptools / cf-cli dump 2015-05-27T18:53:24+0100
# cf-cli is a tool to manage Cleanflight CLI dump backup and restore
# <https://github.com/stronnag/mwptools>
# Windows binary <http://www.daria.co.uk/cf-cli/>
#

# version
# # Cleanflight/NAZE 1.9.0 May 22 2015 / 19:57:08 (f2013ab)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY

# map
map TAER1234

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = 0
set input_filtering_mode = 0
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 0
set auto_disarm_delay = 5
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_functions = 1
set serial_port_1_msp_baudrate = 5
set serial_port_1_telemetry_baudrate = 5
set serial_port_1_blackbox_baudrate = 5
set serial_port_1_gps_baudrate = 4
set serial_port_2_functions = 2
set serial_port_2_msp_baudrate = 5
set serial_port_2_telemetry_baudrate = 0
set serial_port_2_blackbox_baudrate = 5
set serial_port_2_gps_baudrate = 4
set serial_port_3_functions = 257
set serial_port_3_msp_baudrate = 2
set serial_port_3_telemetry_baudrate = 2
set serial_port_3_blackbox_baudrate = 5
set serial_port_3_gps_baudrate = 2
set serial_port_4_functions = 0
set serial_port_4_msp_baudrate = 2
set serial_port_4_telemetry_baudrate = 2
set serial_port_4_blackbox_baudrate = 5
set serial_port_4_gps_baudrate = 4
set reboot_character = 82
set gps_provider = 1
```
**CLI dump from 2015-05-30**
```
# mwptools / cf-cli dump 2015-05-30T07:34:52+0100
# cf-cli is a tool to manage Cleanflight CLI dump backup and restore
# <https://github.com/stronnag/mwptools>
# Windows binary <http://www.daria.co.uk/cf-cli/>
#

# version
# # Cleanflight/NAZE 1.9.0 May 30 2015 / 07:29:07 (27db9ad)
# dump master

# mixer
mixer TRI

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature TELEMETRY

# map
map TAER1234

# led
led 0 0,0:::0
led 1 0,0:::0
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = 0
set input_filtering_mode = 0
set min_throttle = 1064
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 160
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 0
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_functions = 1
set serial_port_1_msp_baudrate = 5
set serial_port_1_telemetry_baudrate = 5
set serial_port_1_blackbox_baudrate = 5
set serial_port_1_gps_baudrate = 4
set serial_port_2_functions = 2
set serial_port_2_msp_baudrate = 5
set serial_port_2_telemetry_baudrate = 0
set serial_port_2_blackbox_baudrate = 5
set serial_port_2_gps_baudrate = 4
set serial_port_3_functions = 1
set serial_port_3_msp_baudrate = 2
set serial_port_3_telemetry_baudrate = 2
set serial_port_3_blackbox_baudrate = 5
set serial_port_3_gps_baudrate = 2
set serial_port_4_functions = 0
set serial_port_4_msp_baudrate = 2
set serial_port_4_telemetry_baudrate = 2
set serial_port_4_blackbox_baudrate = 5
set serial_port_4_gps_baudrate = 4
set serial_port_5_functions = 0
set serial_port_5_msp_baudrate = 5
set serial_port_5_telemetry_baudrate = 0
set serial_port_5_blackbox_baudrate = 5
set serial_port_5_gps_baudrate = 4
set reboot_character = 82
set gps_provider = 1
```

stronnag, 2015-05-30:
See also http://www.rcgroups.com/forums/showpost.php?p=31729356&postcount=8201

hydra, 2015-05-30:
Ahh, it's not bogus, it's uart3, but yes i can see how it might be confusing.  UART3 support was added for Paris Air Hero 32 / Mini Acro Naze32 SP

stronnag, 2015-05-30:
It is bogus on a naze32!

hydra, 2015-05-30:
I agree there needs to be some config migration step for this for the configurator.  Not sure what to do about the CLI.  Gargh @ CLI.

hydra, 2015-05-30:
Updating the CLI to only dump/show active serial ports would cause problems when restoring a CLI dump with soft serial ports to a board running defaults.

hydra, 2015-05-30:
Added a comment to the release notes for now.

hydra, 2015-05-30:
I guess, the hardware revision can be used in the cli code... Not an ideal situation.  CLI shouldn't have a dependency on the hardware.

stronnag, 2015-05-30:
I'll test this with the magnificent PR 806 (mavlink) on top, a great soft serial stress tester. Then I'll update the PR and close it if there are no other issues beyond the inconvenience.

stronnag, 2015-05-30:
No issues using soft serial 1 + PR 806 on serial 4. Just needs a big fat warning in the release notes.

hydra, 2015-05-31:
So, I wrote a bit of code for a 'serial' cli command.  It uses the fixed port identifiers.

e.g. for a naze32 with softserial you get this:

```
# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 30 4 19200 57600 0 115200
serial 31 2 115200 19200 0 115200
```

I think it's more user-friendly than the current way and didn't need any hardware revision specific code.  I'll poke it a little and see if I like it.

hydra, 2015-05-31:
It turns out that WITH then code the build is smaller:

OLD:
```
   text	   data	    bss	    dec	    hex	filename
 117212	    188	  13756	 131156	  20054	./obj/main/cleanflight_NAZE.elf
```

new:
```
    text	   data	    bss	    dec	    hex	filename
 116396	    188	  13764	 130348	  1fd2c	./obj/main/cleanflight_NAZE.elf
```
The main reason for this is the removal of the strings like 'serial_port_1_telemetry_baudrate'.

It's also a more flexible design.

I think I'll merge it.

closed 2015-05-31 by hydra

JohnOCFII, 2015-05-31:
I like it and think it makes sense.  It is also a clean break from some of the older-style "scenario" commands that are still being used and causing folks confusion.  Lots of people (at least based on RCGroups thread) are running into issues as they are trying to do all their configuration from the CLI and are not correctly using the Ports tab.

stronnag, 2015-05-31:
Just posted a big fat warning on rcg about these changes :)

commit ca22fb0e (2015-05-31): Replace CLI serial port configuration code with a new `serial` command.

This is done to make it clearer in configuration dumps which ports are
enabled and configured.

There is a nice code size reduction due to less strings being needed.

Serial documentation updated.

Closes #966.