cleanflight/cleanflight #970: Barometer - FLIP32 - quad goes fastly up when switch baro mode on

misakm, 2015-06-01:
Hello,
I am begginer in cleanflight. Just have it on my quad 250 few weeks. I am facing an issue with barometer. This flight controllerhttp://www.banggood.com/Naze32-Flight-Controller-With-32-bit-STM32-for-Multicopter-10Dof-p-953849.html.
When I enable baro mode quad goes fastly up till I again again switch it off. Tried version 1.8.0, 1.8.1 and also pre-release 1.9.0. 
Coud it be possible bug or incorrect configuration. Thank you for check this. Should I add here config dump?

BR, Michal

JohnOCFII, 2015-06-01:
Just FYI - there have been barometer problems on some recent boards.  I had a barometer that lost 300 feet in 15 minutes while the FC was sitting on a table indoors.

Arakon, 2015-06-01:
Did you actually cover the baro in foam to protect it from light and air? Otherwise, it will read complete garbage.

misakm, 2015-06-01:
I have my flip32 in plastic case. I found out that baro is very sensitive to air flow. So I filled holes by tape to test and it is much more better. What kind of foam do you mean? Do you have an example? Can anybody help me where is baro module on flip32 board?

Arakon, 2015-06-01:
Any sort of dark foam, i.e. the stuff used to protect electronics. Sealing the case won't do the trick.
Baro is at the bottom of the board.

misakm, 2015-06-01:
@Arakon Thank you a lot, I will try.

JohnOCFII, 2015-06-01:
@Arakon @misakm The barameter is on the top of the latest Flip 32+.  In the photo it is the white component with two holes.  Some older boards had the barometer on the bottom.  Sometimes the barometer is metal with offset holes.  http://www.readytoflyquads.com/the-flip32-187

Arakon, 2015-06-02:
The original, yes. But he has a banggood clone, and those are the older version with the baro at the bottom.

hydra, 2015-06-02:
Don't think there's a software issue here.  Alt-hold code is being improved and there are other issues that deal with it.

closed 2015-06-02 by hydra

misakm, 2015-06-03:
@hydra - Yes this is hardware issue. Anyway thank you all for your advices :)

nitro20, 2016-05-29:
i have a 6dof naze full version and the baro does the same thing..but i can make it stop by lowering pitch/roll PID..mainly P..but its so low that the quad is not right. So with the PIDs pitch and roll and yaw tuned, i turned Air Mode, Baro and Mag off because even in mag and air mode modes it seemed to get jittery, but with baro it will just take off like its on crack right now..not that i use mag or baro in arco mode to race around and stunt, and i dont even know what air mode does, but its disturbing when i think about how fast some one could of got hurt when i switched to horizon mode and the quad went up like a bat out hell. plus i payed for a full version. 
  i havent mest with the baro or mags pids because i just want to fly and dont need to, it flys great in arco, i think, (no basis of comparison, im a loner FPV'R) and its so dangerous to arm a "lose cannon" Quad copter that could just go go..???

nipunnawanjana1, 2018-02-20:
i have work this follow this video https://youtu.be/9Es1NzxLyWE