cleanflight/cleanflight #971: [SONAR + 1.9.X] Issue with Hex-X configuration

alex-savin, 2015-06-01:
Hi,
I have tried pre-release and nightly build of 1.9.X branch and got strange issue while sonar enabled, my Hex-X multicopter don't spin 5 and 6 motors. But in configuration everything seems okay.
With Great Regards

hydra, 2015-06-01:
Thanks for the report. There was some code cleanup with regards to pin allocation for sonar when cc3d sonar support was added so maybe theres an issue with it.

hydra, 2015-06-02:
Sonar tested with Naze32 in Parallel PWM and PPM mode. No issue found.  Please see documentation regarding which pins to use for sonar.  PWM: M5/M6, PPM: RC7/8

Sonar + Parallel PWM + HEX = Impossible on Naze32, not enough IO pins are available.

closed 2015-06-02 by hydra

alex-savin, 2015-06-03:
Hi,
Once again about this issue. I guess have found a cause - enabling Current Meter and Sonar simultaneously. After these action I've lost motors number 5 and 6. But when I disabled Current Meter, but Sonar left enabled, motors start spinning again. 
With Great Regards