cleanflight/cleanflight #976: Lia & Lisa flight controller support

xLORDxSIDIUSx, 2015-06-01:
Would it be possible to add support for Lisa/M and Lia (Quadshot mocha) flight controllers? they are both STM32 based.  this would give cleanflight VTOL capability.

hydra, 2015-06-02:
Please provide more details, links etc.  Don't see how supporting a different FC gives you VTOL support?  that's more on the software side of things.

xLORDxSIDIUSx, 2015-06-03:
The paparazzi based flight controller in question can be found here:
http://wiki.paparazziuav.org/wiki/Lisa#Lia
http://wiki.paparazziuav.org/wiki/Lia

The craft i have is here:
https://www.kickstarter.com/projects/jkg/quadshot-an-aerobatic-blend-of-rc-helis-and-planes/description
http://transition-robotics.com/products/the-mocha

after reading your comment, i realized that yes, the software must be the key to realizing the transition mechanic, because i don't see anything on the flight controller that is substantially different from a naze32. It does have multiple UART and I2C connections.

closed 2017-02-25 by hydra