cleanflight/cleanflight #977: Vbat reset using bluetooth module

Life1024, 2015-06-02:
I'm using a Naze 32 board, I'm using C.F. v1.8.1, frsky d4r2 with telem ok, i have a buzzer too.
My bluetooth module is a crius - "MWC Multiwii Bluetooth Parameter Debugging Programmer Module"
I had issues using the module with standard pins on the naze32 when battery and USB were both plugged.
So i decided to go for USART2 TX and RX on RC3 & RC4.
Everything seem to work then, but when i set up my PIDs while my RIG is powered up, my PID are set correctly but my VBAT is reset to 0v.
Then i'm unable to arm the quad again because of that.
So the naze is reseting and then just beeping because of the VBAT reset.
I have to connect it to CF to set it to the good value again, PIDs are OK.

hydra, 2015-06-02:
There is some now code coming which will deal with this.

hydra, 2015-06-02:
See #955

closed 2015-06-02 by hydra