cleanflight/cleanflight #978: GPS - Remove satellite information to free up memory.

digitalentity, 2015-06-02:
It occured to me that actual information on satellites in view is useless for flight controller  (https://github.com/cleanflight/cleanflight/blob/master/src/main/io/gps.c#L84):
```
uint8_t GPS_numCh;                          // Number of channels
uint8_t GPS_svinfo_chn[GPS_SV_MAXSATS];     // Channel number
uint8_t GPS_svinfo_svid[GPS_SV_MAXSATS];    // Satellite ID
uint8_t GPS_svinfo_quality[GPS_SV_MAXSATS]; // Bitfield Qualtity
uint8_t GPS_svinfo_cno[GPS_SV_MAXSATS];     // Carrier to Noise Ratio (Signal Strength)
```

For pilot the only needed information on GPS quality is number of satellites in view (`GPS_numSat`) and fix status. Navigation code also does not use the detailed satellite information. I think that detailed satellite information is useless for flight controller and can be safely removed without affecting flight controller capabilities.
@hydra what do you think?

JohnieBraaf, 2015-06-02:
Are you concerned about the processing speed of the GPS messages or about the code size? Compared to the actual information used, indeed its a lot of processing that can be prevented. Although I like the visual feedback in the configurator.

JohnOCFII, 2015-06-02:
Do external OSD's rely on that data coming from the flight controller?  MW_OSD, for example?

hydra, 2015-06-02:
Well, actually the OLED display used it to draw nice little graphs on the screen too, not just used for MSP.  I know of no OSD that uses it.  Stronnag's MWP tools might use it.

ublox m8 support causes further issues as the protocol allows for even more space vehicle tracking.

Not considering removing now, but if things get tight we could look to improve it.  A better candidate would be the led strip DMA buffer.

closed 2015-06-02 by hydra

digitalentity, 2015-06-02:
Got it, thanks.