cleanflight/cleanflight #988: Vtail4 Prop Direction and Mixer Incorrect

maymayday90, 2015-06-04:
Hi all,

I believe the prop direction and mixer for Vtail4 in Cleanflight are incorrect. All prop directions need to be opposite of those seen in the Configurator picture (and therefore flipping mixer values). 
[image]

This has to do with the induced yaw by the rear motors. For example, if you wanted to yaw left, you would speed up M3 (given simplistic mixing). However, with the current prop directions, the torque applied to the frame because of motor acceleration (same way a quad yaws) applies a right yaw. And therefore fights the left yaw that the Vtail is trying to achieve.

This usually is evident by yaw waggle in flight, and is difficult if not impossible to tune, since the yaw is fighting itself.

This is not an issue with the Atail as the motor axis is flipped.

As an example, here is the custom mix for the Armattan Morphling Vtail, which worked great in baseflight:

cmix 1 1 -0.580 0.580 1.000
cmix 2 1 -0.460 -0.390 -0.500
cmix 3 1 0.580 0.580 -1.000
cmix 4 1 0.460 -0.390 0.500

Thanks,
Andrew

hydra, 2015-06-04:
yes, i too noticed the default v-tail mixer settings didn't work on that frame.  I'm curious as to which frames and configurations the default v-tail mix actually works on.

maymayday90, 2015-06-04:
My guess is none, unfortunately, so I'm curious as to where this original mixer came from. My guess is that is was just ported from the Atail mixer.

If you want the Vtail to fly well, you can't have yaw fighting itself. And as far as I can tell, it will be doing so for every Vtail because of the motor axis and induced yaw.

add1ct3dd, 2015-06-04:
In that sense, if there is no use for it, it may as well be updated to support the Armattan frame above.

hydra, 2015-06-04:
I agree.

maymayday90, 2015-06-04:
I don't know if it's best to put my custom mixer as the default one, as it won't transfer as well to frames with different dimensions. I need to take a look at it again anyways, as there is a NG error on one axis (though it flies fine). 

The numbers just need to be inverted from the Atail mixer, and the picture updated. If I new where the mixers were located in the code, I could branch it for testing (though I don't know how to compile).

hydra, 2015-06-11:
See

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L208-L213

and

https://github.com/cleanflight/cleanflight-configurator/blob/master/support/svg_model_motors.rb

https://github.com/cleanflight/cleanflight/tree/master/docs/development

There, all the info you need :smile:

Please change, test and create PR for motor mixer and images, thanks!

stronnag, 2015-06-13:
The vtail motor directions were researched (i.e. copied, only call it research) directly from Baseflight If they are wrong in BF, then they are wrong also in CF.  Changing this in the script that generates the SVG is trivial; changing the perhaps mixer less so.

cmwwebfx, 2015-06-17:
Rather than fixing this one, please consider to just rename it as A-Tail. Then reconfigure for a New vtail.

cmwwebfx, 2015-06-17:
As you can see here the current setup will work with what I am currently developing. Rear view of tail setup.

[image]

dkisselev, 2015-06-17:
@cmwwebfx A-Tail already exists, the issue here is that the current v-tail mix is an exact copy of it. So v-tail just needs to be tweaked so that the motors point the right way,

tedbyers, 2015-06-25:
What I have found in checking this out is that the mix predates baseflight and the same numbers for the v-tail mix in MultiWii are the same as those in cleanflight.

I have never heard of an A-tail before looking at this code.  Can someone point me to a reference that compares the A-tail with the V-tail and regular quad: advantages and disadvantages of each?

And, can someone point me to a resource that explains what those numbers really mean, how changing them impinges on how the drone flies, and if there is a relation between the numbers used and specific properties of the airframe (such as arm length or the angle in the V-tail?  The motivation for this last question is that I was made aware of MMD's cmix numbers for the two V-tails available at armattanquads.com (posted about half a year ago in RCG), and I am curious about two things: 1) why are they different? and 2) how would I create a default mix for v-tails?  I am in the middle of designing prototypes for v-tail hexacopters and octocopters, among others, so if I can understand this, I can develop appropriate mixes for these new airframes too.  Changing the code is trivially easy for me, having been coding in C/C++ for decades, but drones are new to me and while I have a decent background in math, I have not found the resources most suitable to understand how the math and physics of these marvelous machines relates to this code.

Thanks

Ted

joshuabardwell, 2015-06-25:
I used to fly an A-tail. My main reason for choosing an A-tail over a V-tail was that the default MultiWii quad motor directions allowed torque and thrust to push the same direction, as has been pointed out in this thread. My theory as to the history of the mix is that someone originally built a V-tail using the default mix, and since thrust is stronger than torque-induced yaw, everything seemed to work okay, and nobody put any more thought into it.

There may be some aerodynamic differences between A-tail and V-tail. Whether the air is mixing on the top or bottom of the prop disc. My gut feeling is that these differences are really minimal at the degree of tilt we typically see on V-tail and A-tail copters. If you had something past, say, 45 degrees of tilt, it feels like it would start to matter more. I don't have any data to support this, though.

Default mix for V-tail should reverse all motor directions relative to QUADX so that torque and thrust work together, not against each other. This is the bare minimum.

It's my opinion that V-tail and A-tail mixes should also reduce or eliminate roll authority from the rear motors. The reason is that rear motors have much stronger yaw than front motors due to their tilt, and as a result, you will get yaw coupling on roll inputs. I flew my A-tail with zero roll mixed into the rear motors.

Likewise, I found that reducing yaw authority in the front motors to 25% or below produced the smoothest yaw, with the least coupling of pitch/roll. At the time, however, I was flying a 450 class copter with 10" props. I think this is probably much less of an issue on smaller copters, where there is less prop momentum. And also, now we have active braking as a much more common feature of ESCs. However, because of the better yaw authority that comes from smaller props, IMO you're much more likely to see A-tail and V-tails being larger frames using larger props. So maybe that should be the target of the default mix.

The default throttle mix for the rear motors depends on how many degrees of tilt they have. Trigonometry should be used to determine the relative strength of the vertical vector, and the rear motors should have that percentage added. Likewise, on pitch, since the rear motors will have less authority on that axis than the front ones. Some research into common motor angles on popular frames might be helpful.

tedbyers, 2015-06-25:
Thanks Joshua.  That is very helpful in understanding the basics.  But, can I trouble you for two more things?  First, can you explain the relationship between roll, pitch, yaw and thrust and the numbers I see in MMD's cmix earlier in this discussion?  Are there maximum and minimum acceptable values for those numbers?  Second, I may be a glutton for punishment, but can you point me to a webpage or website that describes the math you touched on in detail?  Seriously, my background is in mathematical ecology, part of which involved mathematical chaos, so a little algebra, calculus and geometry is not a problem for me.

I ask because in addition to wanting to play with A-tail and V-tail quads, I want to take a look at a pentacopter and hexacopter in which the basic airframe is a V-tail or A-tail quad, but to which I add a motor at the back which is oriented orthogonally to the other motors (so all the thrust it produces is devoted to flying forward) for the pentacopter.  And for the hexacopter (with an H-frame)m the front and back pairs of motors correspond to the usual motors on the A-tail or V-tail, but the middle pair of motors are oriented orthogonally to the rest of the motors, again so that all the thrust they produce is used for flying forward.  And I have other ideas I want to experiment with, but first I must come to understand the math behind these mixes, so I can get that right from the beginning.

Thanks

Ted

joshuabardwell, 2015-06-25:
@tedbyers I will do my best. My understanding of the cmix command is not 100% complete, but I think I can at least point you in the right direction.

First, please read up on the cmix command if you haven't already.

http://blog.oscarliang.net/custom-motor-output-mix-quadcopter/

joshuabardwell, 2015-06-25:
@tedbyers I will do my best. My understanding of the cmix command is not 100% complete, but I think I can at least point you in the right direction.

First, please read up on the cmix command if you haven't already.

http://blog.oscarliang.net/custom-motor-output-mix-quadcopter/
https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md

I think of the cmix command as a multiplier that is applied to the output of the PID loop. The nominal way of configuring cmix is to have the numbers be from 0 to 1, with each number representing the percentage that should be applied to the output. In reality, the system works with numbers above 1, as long as the ratio between the numbers is the same. Somewhere, there is a limit where you overflow a variable if the cmix numbers are too big, but I don't know what it is off hand.

The pages I linked to above should help answer your questions about how pitch, yaw, etc... fit into the cmix command. So for example, to configure the back-left motor (number 3) with zero roll authority, you would enter:

    Default: cmix 3 1.0 1.0 1.0 1.0
    V-tail:  cmix 3 1.0 0.0 1.0 1.0

To set the front-left motor to 25% yaw authority:

    Default: cmix 4 1.0 1.0 -1.0 -1.0 
    V-tail   cmix 4 1.0 1.0 -1.0 -0.25

For the rear motors, you want to increase pitch authority relative to the tilt of the motor. Although my trig is pretty weak, I'm pretty sure you calculate this by taking the cosine of the tilt angle. So a V-tail with 20 degrees tilt would have 6% less vertical thrust in the rear motors. You could address this by setting the rear motors' pitch mix to 1.06, or by setting the front motors' pitch mix to 0.94. As you can see, the loss of thrust is pretty small for even moderate tilt angles, but fixing the mix like this helps the PID loop have less work to do.

In order to have "lifting-only motors", I believe the correct approach is to set the motor's throttle parameter to 1, but set the pitch, roll, and yaw parameters to zero. The direction parameter doesn't matter since the motor is not being used for torque yaw.

    cmix 5 1 0 0 0 0

I don't know the approach for having "pushing motors". The reason for this is that the throttle will be used to control all the lifting motors, so you will need a second input, like a pot on your transmitter, to control the pushing motors. At that point, maybe the right approach is to use gimbal_flags=4 to break out the channel from your transmitter and feed it directly to the pusher-motors' ESCs.

tedbyers, 2015-06-26:
Thanks. I have read the first reference, and am starting the second, but this is very helpful.  thanks.

For the pushing motors, would not they have 100% input from the elevator and have that scaled by the %throttle (so the motor is off when there is not elevator input, but have basically a product of the throttle input and the elevator input, to determine how fast the drone flies forward (can this be done in the FC, or would that have to be programmed in the Tx; or do we even need such a product, that is, will getting 100% input have the pusher turning at the highest RPM it can support?), but have the other motors take a little input from the elevator.  We don't want the other motors having 100% input from the elevators as part of the reason for having the pushers is that we don't want as extreme an angle on the airframe when flying forward fast, but we must have some so that a negative elevator value can still slow the drone.  How much pitch should the front and rear motors get so that they can produce just enough pitch to do a backward flip (optional - I can live without this), or use the corner motors rather like an air-brake, but not so much the drone will take on much of an angle when flying forward fast (IMHO, absolutely essential)?  And I wonder if it would be helpful to have the pusher help on yaw, and how that would be handled (optional, but maybe helpful)?  Lots to study.....

joshuabardwell, 2015-06-26:
I think you are assuming you will be flying in an autolevel mode, where you will hold constant stick position to hold attitude. Is that right?

In that case, one approach would be to tie the pusher-motor throttles to the elevator stick, and let the FC autolevel the copter along the pitch axis. You won't have independent pitch control, but maybe that's desirable, especially if you are flying a camera platform with a gimbal. You would need to accomplish this in the transmitter, for example, by setting the elevator channel to mid-point and just leaving it there, and then putting the pusher-motor-throttle on one of the aux channels and passing it through directly to the pusher-motors' ESCs. 

But there is some problem here in that autolevel isn't perfect, and you probably will still need some pitch control to hover. Maybe one approach would be to trim the accelerometer slightly pitch-back and then counteract that backwards drift with the pusher motor.

For help with yaw, look into thrust vectoring.

If you are concerned about fast forward flight and acro-style flight, the answer to your concern about pitch is tilted motor mounts. This allows the copter to fly forward with minimal pitch. It has to pitch back to hover, but for acro-style flight, hovering is not the main focus.

If you are trying to build a camera platform, your best approach will be to use gimbal to stabilize your camera, rather than trying to invent all sorts of new complications to avoid tilting the airframe itself.

tedbyers, 2015-06-26:
Thanks for this

yes, I will primarily be flying in either angle or horizon mode, but I will want it to work in acro mode also.  That is why I'd give a little pitch input to the motors on the corners.  The question then reduces to how much.  So, if I can use only the elevator input to get maximum RPM from the 'pusher' motors, then to aide the FC in leveling, could I give the other motors as little as 10% or 20%, or would it have to be higher?

I wonder, though, if input from the barometer can be used in aiding altitude control.

What I am after is a mix that provides for a rather versatile aircraft, as suitable for, say, FPV racing as it would be for AP or acro.  Would it be fair to expect that I could come up with decent default mixes for A-tails and v-tails as a function of the angle of the rear arms relative to the horizontal (say 4: 30, 35, 40 and 45 degrees) using the trig you mentioned?

Finally, do you have a good reference for thrust vectoring as it applies to multirotors?  Google so far has had an abysmal record, giving only garbage or pages (such as wikipedia's page) on thrust vectoring on jet aircraft.  I get the general concept, and am quite interested in it, but I am after details I can use on calculating mixes for multirotors.

Thanks

joshuabardwell, 2015-06-26:
You would want the motors on the corners to have 100% pitch authority in the mix. You would lock the elevator input to center so that the FC never perceived that you were moving the pitch stick at all. The FC would then autolevel the copter as best it could and you would fly forward by using the elevator stick to throttle up the pusher motor.

You can't achieve what you're thinking of in acro mode--only autolevel mode. In acro mode, you must use the elevator stick to control pitch.

But all of this is missing an important detail: if you are achieving thrust via the rear-facing prop, how do you slow down? You actually need a motor facing forward AND back. Or you need a 3D ESC and a prop so that you can reverse the thrust motor to both push and pull.

The baro can be used to activate altitude hold mode. This is a basic function of cleanflight.

Here is someone who has done something like what you're thinking of.

https://www.youtube.com/watch?v=hiDPbcVLbMo

closed 2015-09-05 by hydra

azempire, 2018-10-24:
I have a Eachine Vtail and made my own calcs based on your advise to change the CUSTOM to adapt to it. 
My question is why having different yaw in the rear motors (halved to 0.5). 
I understood the advantage of having a Vtail was having more authority on Yaw, but reducing the rear motors to half is not changing that? 
Thanks

hydra, 2018-10-25:
no, since the vtail with 0.5 setting uses the rear motors for rotation 50% for vectored thrust and 50% for differential torque.

commit 28aad588 (2015-09-05): Fix vtail mixer defaults.  Closes #988

Previously the vtail mixer was a copy of the atail mixer which was
incorrect.

The new defaults cater for frames like the Armattan Morphling Vtail.