cleanflight/cleanflight #989: Constraining RX values is potentially dangerous

TighMacFanatic, 2015-06-04:
https://github.com/cleanflight/cleanflight/commit/7a01bab50ff467c72adf38f4f385873f7a681d4d

Constraining RX values to sane values is fine, but setting to mid range when out of bounds is a bad idea.  Without knowing that rx_min_usec and rx_max_usec needed to be tweaked for my receiver bounds, I got the behaviour that when I went to minimum on most of my channels, they would suddenly jump to mid range until I eased back.  It also caused my aux channels to seem to stop working, as they were always out of the new bounds.

If the receiver loses signal and the stick channels go below rx_min_usec, the throttle will be set to 50% for failsafe_delay (1 second by default I believe) prior to failsafe triggering.  This could cause a sudden jump.  It would be much better to set these values to the last sane value seen when they go out of bounds, or at the very least for the throttle channel.

hydra, 2015-06-04:
This is not new behavior, just the limits have changed are are tighter and now configurable. Feel free to use the old hard-codes values of 750 and 2250.

See also this comment - https://github.com/cleanflight/cleanflight/pull/982#issuecomment-108633159

See also #918

Since there's already a conversation going for this topic I'll close this issue.  Please search before opening new issues.

closed 2015-06-04 by hydra

hydra, 2015-08-04:
Fixed with #995.