cleanflight/cleanflight #991: Firmware v1.9.0 can't save Yaw Servo settings

phantom-8, 2015-06-04:
After upgrading to fw v1.9.0 on Naze32 via Configurator v0.65.0, the "Yaw Servo" setting page under "Servos" tab doesn't work.  Changes made on this page can't be save to eeprom.  Previous versions fw 1.8.1 worked fine.

hydra, 2015-06-05:
thanks for the report.  Please confirm that you also tried latest configurator with firmware < 1.9.0.

Please note this issue should have been opened in the configurator issue tracker.

DeetAus, 2015-06-06:
I had the same issue, I went back to v1.8.1 and with the latest configurator I can change the servo settings

Also I found I couldn't load my v1.8.1 backup file to the NAZE32 after upgrading to V1.9. I would get the line of script saying it was loading but it would just hang
I was going through and changing settings manually but when I got to the servo page I found nothing would change or save

Configurator v0.65.0,  AFRO NAZE32

ATA250, 2015-06-11:
Same for me, latest configurator works fine with 1.8.1, but will not save servo settings for 1.9.0.

shoreflyer, 2015-06-14:
Yeah, same for me, previous versions of cleanflight (<1.9.0) with latest (0.65.0) configurator worked...

SteveAmor, 2015-06-15:
Fixed in https://github.com/cleanflight/cleanflight/commit/9a8a31676b8612e924964f212da1a5c1e63f5c20 for Cleanflight code.

nsccap, 2015-07-01:
I also ran into this when I updated my cf from a "almost 1.9.0" / 4857050 to official 1.9.0 and restoring my backup file hung. It was of course equally reproducible by trying to modify the servo tab...

hydra, 2015-07-10:
Closing this.  Configurator needs work for latest master code and existing servo configuration code will likely be replaced with new code due to new servo mixing rules and servo angle support.

closed 2015-07-10 by hydra

deejayspinz, 2015-08-21:
Aug 21st just downloaded the latest 1.9.0 and can confirm this is still not fixed.  Can't save any servo settings.

ituri, 2017-04-28:
Is it correct that this is still an issue? v2.0.2 is still showing the same behaviour.