cleanflight/cleanflight #996: BlackBox - Gyro data and quad model is missing

ksal0, 2015-06-05:
Gyro data is no longer rendered. The new gyroADC data when rendered are basically strait lines and now useless. 

Also the quad model is missing. 
1.9.0 pre release was good. 
This started with 1.9.0 stable
Image is at 1/1 sample rate

[image]

digitalentity, 2015-06-05:
I also have the similar issue - blackbox_decode with --simulate-imu option shows an error: "Can't simulate the IMU because accelerometer or gyroscope data is missing"

strfn, 2015-06-05:
Same issue here.

SteveAmor, 2015-06-05:
I guess I broke this when refactoring the cleanflight code, replacing gyroData with gyroADC. gyroData is in the Blackbox viewer code. I'll look to fix later.

SteveAmor, 2015-06-05:
Fixed in PR#998

SteveAmor, 2015-06-10:
Officially fixed with updated black box viewer and tools. This issue can be closed.

closed 2015-06-11 by thenickdude