cleanflight/cleanflight #999: Erase Dataflash occassionally locks up the FC

Arakon, 2015-06-06:
- CC3D, 1.9.0 stable, Configurator 0.65
- Larger 64 mbit Dataflash (W25Q64FVSSIG) soldered to the board

When flash is pretty full (60%+), erase will get stuck forever at "Erasing flash". The FC freezes.. i.e. LED will no longer flash when tilted, no LED response to mode switching, USB will not connect again after pressing Disconnect. Previously dumped log plays without problems, so there seems to be no corruption of the actual data.
Disconnecting it from power solves the issue. Flash then shows as less filled.
Erasing again now completes within 20 seconds or so, without another lockup.
Happened 3 out of 4 times, and each time the flash was over 50% full. Under 50%, it seems to erase without issue.

thenickdude, 2015-06-09:
Is this when erasing using the Configurator's dataflash tab, or on the CLI tab with the flash_erase command?

Arakon, 2015-06-09:
Using the dataflash tab. It just stays forever at the "Erasing flash" window. I had it happen again yesterday, Dataflash was 100% filled, clicked erase, left it sitting for half an hour.. still stuck. Unplugged USB, closed CF, started CF, plugged in USB, checked dataflash tab -> flash was empty.

thenickdude, 2015-06-09:
Hm, we erase the entire flash chip and as far as I know the erase time doesn't scale with the quantity of filled blocks, so I can't think of a good reason for it to happen more often with full chips. 

Out of interest, do you have a decoupling cap installed in the spot next to the flash chip (the full Naze32 ships with this cap present). If not, it may be a flash chip glitch from voltage sag on its input as the erase operation begins.

Arakon, 2015-06-09:
[image]
This is one of the two cc3d I have with the larger flash. I'm fairly sure one of the two caps is attached to the chip.. at least both of them are connected to the chips' vcc line too.
I just tried with the one in my quad. At 3% full, the flash takes approx. 15-20 seconds to erase and the board keeps working. I also tried 4.3 mb full (just over half) and the board kept working and finished erasing, but during the erase process, the LED that usually keeps flashing since it sees no receiver etc, kept flashing out of its usual rythm, as if the FC was having trouble coping.

hydra, 2015-06-10:
with regards to the flashing, that's caused by looptime jitter - unrelated.

hydra, 2015-06-10:
I'd just like to chime in on the original problem too, I have never had a problem erasing any flash chip on any of the boards I have - CC3D, Naze32, SPRacingF3.  Various flash sizes, brands and board revisions.

Since your flash chip was hand-soldered perhaps there's an issue with the assembly or the component?

@thenickdude do you think there's anything we can do?

Arakon, 2015-06-10:
Then it would happen every time.. also, since this chip doesn't have a dedicated erase line, other issues would also occur if it was the soldering. I did of course confirm a good connection of all pins too.

thenickdude, 2015-06-10:
Yeah I haven't had this issue either, though my chip is just the 16mbit stock one. 

The interesting part is how it actually locked up the board for @Arakon, since the flash code isn't supposed to wait on the chip after the erase begins (the Configurator polls the flash chip to check its ready state, but I thought if the chip didn't reply on the I2C bus the FC would just continue operation). 

That's a lockup that we'd like to avoid. It seems like if we don't know the cause then there's no reason that it couldn't happen in flight when the FC attempts to write to the flash.

That said, unless it can be reproduced under a debugger I don't think we'll figure it out without more user reports.

digitalentity, 2015-06-10:
I've encountered a similar lockup a few times with stock CC3D at pre 1.8 versions. What I figured out that it is more likely to occur if using VCP to communicate with FC. However I've not encountered this issue since 1.8, not even once. Hope this info helps.

fedorcomander, 2015-06-22:
Have soldered chips of differend sizes, with caps and without. never had a single issue.

hydra, 2015-07-10:
I had an issue a week or so ago, have been unable to reproduce it.

To find out if the crash occurs at the start or end of the ease cycle you can tilt the aircraft while the flash is being erased.  If it crashes the LED will stop blinking.  Count the blinks to see how far into the erase process you get.  I do this everytime i erase the flash and it's not crashed since i started doing this...

thenickdude, 2015-07-10:
@hydra on CC3D?

borisbstyle, 2015-07-31:
Downloading dataflash for weeks multiple times per day and never hit this issue.

Arakon, 2015-07-31:
It still happens for me at times, about once every 10-20 downloads and erases. I can actually entirely get around it by clicking Erase, then Cancel.

hydra, 2015-09-15:
@Arakon thanks for the info, i wonder if this helps @thenickdude ?

thenickdude, 2015-09-15:
Clicking the cancel button means that CF doesn't poll the flash chip for its status while it is erasing. 

So we might be getting into an infinite loop in the SPI code when the chip responds inappropriately to a status request, or there could be a race in the SPI code. I'll have a poke around.

closed 2017-02-25 by hydra