cleanflight/cleanflight #1000: Profile and rate profile switching.

makattack, 2015-06-07:
Under 1.8.1, I was able to configure all three profiles 0-2, along with all three rateprofiles such that I could select any rate profile from an AUX channel no matter which profile I was using. This does not seem to work under 1.9.0.

Under 1.8.1, I had this configuration: 
adjrange 0 0 4 950 2050 12 4

On a flip32+ running SBUS rc input (X4R-SB RX), I used to be able to use a three position switch to select different rateprofiles independent of the profile.

Additionally, the profile switching using stick commands used to result in the status/green LED blinking 1-3 times to indicate the profile selected. This no longer happens under 1.9.0 -- which does blink once for profile 0, but doesn't change the steady blink rate for profile 1 nor 2. That's also new in 1.9.0, for the status light to blink on a regular interval vs. stay steady on.

While searching the code to see if there was a change that looked to have obviously altered this behavior:
https://github.com/cleanflight/cleanflight/search?utf8=%E2%9C%93&q=changeControlRateProfile

I noticed that rc_controls.c, while including config.h which already has a prototype for the changeControlRateProfile() function, duplicates the function prototype just before a function definition for applySelectAdjustment() (line 574).

NOTE: CLI commands to change the rateprofile does work.

Archon03, 2015-06-08:
While we're at it, can we make the "PID Tuning" tab number the profiles 0,1,2 like the CLI does (or vice versa, both use 1,2,3)?

hydra, 2015-06-10:
@Archon03 that's a gui issue.  this issue is for the firmware.  please stay focussed.

@makattack I was using rate profile selection using 2 and 3 positions switches on 3 different transmitters on 4 different quads last week, pretty sure there's no fundamental problem with 1.9.0.

can you provide a config dump and screenshots;
1) screenshot of RX tab with switch in low position
2) screenshot of RX tab with swithc in high position
3) screenshot of Adjustments tab.

i suspect RX channel endpoints are incorrect or some other mis-configuration at this point.

makattack, 2015-06-10:
Thanks for the analysis. Here is my config dump:

<pre>
# version
# Cleanflight/NAZE 1.9.0 May 31 2015 / 13:17:47 (98f7549)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature CURRENT_METER

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
[... 228 lines cut ...]

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500
</pre>

Low position:
[image]

[image]

Middle position:
[image]

High position:
[image]

[image]

makattack, 2015-06-10:
In thinking about this more, I tried also adjusting my endpoints for AUX5 to be between 1000-2000, but that didn't seem to change anything.

Perhaps the problem is how I'm testing this functionality? I'm in the CLI the whole time, starting with low position. I'll type "dump rates" and see rate profile 0. When I switch to the middle position, and repeat the command, it's still in rate profile 0.

I do notice, if I exit the CLI, and come back in, and issue "dump rates" I will see that it has changed, but I assumed it was only because it set the rate profile on reboot.

hydra, 2015-06-10:
Yes, it's how you're testing it!

the adjustment controls from RX are ignored when in the CLI.  This is by design.

hydra, 2015-06-10:
Please submit a new PR that updates the documentation if you find it is unclear.

closed 2015-06-10 by hydra

makattack, 2015-06-10:
Thank you Dominic. I can submit a PR for the documentation, but would also be happy at helping out with the update of it as well. What about the change in behaviour for the status LED on changing profiles? Do you think that warrants a new PR?