cleanflight/cleanflight #1002: Servo min/max/mid/rate update through Cleanflight Configurator Does not work

ldonnay, 2015-06-09:
Using Cleanflight 1.9 with Configurator 0.65, updating of min/max/mid rate does not work. When clicking on save, there is no action: and log window does not display anything. When changing tab and coming back to Servo tab, values come back to previous version.
Servo Tilt is activated and when coming back to Cleanflight 1.8.1, everythingh is working.

supergregg, 2015-06-09:
Sam problem here. Upgraded to 1.9.0 and now i cant get my plane to perform properly. All "compensation" directions are wrong. Had to revert to 1.8.1.

DKDarkness, 2015-06-09:
Same problem. Running Cleanflight 1.9.0 for Naze32 on a Flip32+ board under configurator 0.65.0. Any changes made in the "Servos" tab when running tricopter or flying wing configuration, will not be saved. No log entry will be written either.

commit 9a8a3167 (2015-06-10): Reduce maximum servos from 10 to 8.

1) the 2 extra servos were not used for anything, yet.
2) the MSP packet to set ALL the servo configurations in one go is
larger than the MSP input buffer size.

Likely the MSP_SET_SERVO_CONF should be updated with command that takes
an index of a servo to configure, as per mode ranges/colors/leds/etc.

Fixes #1002
closed 2015-06-10 by hydra