cleanflight/cleanflight #1003: PPM spikes/dead band shown in Dominic video

Buzzliteyear, 2015-06-09:
The spikes shown in Doninics video here ( https://www.youtube.com/watch?v=EViw4lBRa94 )
I had the similar dead band problem with DTFuhf to flip32+ board but could happen with any ppm set-up maybe

as you notice in Doninics video (1:18 min) when he gets a spike that channel is sitting at about 991 

for me if i set the low to just above 1000 and high just under 2000 ( say 1004-1998) it would not happen on the bench test

Brett

hydra, 2015-06-10:
not quite sure what to do with this issue.  are you saying you still have a problem?  'a similar problem' isn't good enough.  you need to say if you have the EXACT same problem, and if so give full details of what firmware you're using, a cli dump and details of your hardware and how it's connected.

I will close this issue, but if there's an issue please give details as above and we can re-open it

hydra, 2015-06-10:
Also, my name is spelt 'Dominic' :smile:

closed 2015-06-10 by hydra

Buzzliteyear, 2015-06-10:
I don't have a problem now after setting the range as above but in your video the range was low and that was my issue also which caused spikes and a dead band and or re center of control output shown on the receiver graph in cleanflight which when control stick was pushed to the limit ,dont think it would matter which firmware just saying it looks like a range fine tuning thing or out a range thing,
sorry Dominic missed the apostrophe or what ever that little mark is in Dominic's video and then spelt it wrong at midnight

using Configurator: 0.65.0
CLFL, version: 1.9.0