cleanflight/cleanflight #1006: servo tilt overrides throttle channels when using the airplane/ fixed wing mix

craigbs, 2015-06-10:
servo tilt overrides throttle channels when using the airplane/ fixed wing mix

hydra, 2015-06-10:
has this recently started happening or has it always done this?

hydra, 2015-06-10:
looking at mixer.c it's always done it.

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L452-L456

servos 0 and 1 are always used.

https://github.com/cleanflight/cleanflight/blob/master/src/main/flight/mixer.c#L497-L500

and here, SERVO_PLANE_INDEX_MIN SERVO_FLAPS == 2 and thus servos 0-4 are used.

hydra, 2015-06-10:
Fix was simple enough, applied.  Please grab a development build that includes this change and test.

[image]

closed 2015-06-10 by hydra

hydra, 2015-06-10:
It should be noted that the GUI also needs updating to support configuration, however the firmware itself does support it.

hydra, 2015-06-10:
The second commit finally allows the computed gimbal servo value to be used to drive the servos.  without it the value was computed but never used except for display purposes.

leothehuman, 2015-07-26:
Will this affect tricopter output layout? Can a tricopter use SERVO_TILT now?

commit 889b14df (2015-06-10): Support pan and tilt on fixed wing.  Fixes #1006

The general idea is that nothing calls pwmWriteServo except for
writeServos() and that writeServos() knows which servos it has used.