cleanflight/cleanflight #1009: L1 is blinking green erratically with firmware 1.9.0

beakdoctor, 2015-06-10:
I have a AfroFlight 32 rev 5 and after upgrading to the latest firmware (1.9.0) my L1 led keeps blinking semi-erratically. Sometimes it blinks steadily, maybe once per second, and then it'll flash faster sometimes. It never stops doing this, but my quad seems to work anyway. It'll arm and fly. What does the flashing light mean?

I've tried recalibrating the accelerometer but it doesn't help.

If I go back to 1.8.1 I get a solid blue and red light after booting has completed.

kestrel253, 2015-06-10:
I experience the same behavior on an Acro Naze board.  Green LED is steady with blinks about once per second when the quad is disarmed.

Arakon, 2015-06-10:
Same on a CC3D.

hydra, 2015-06-10:
normally it's RX loss detection or low battery.

hydra, 2015-06-10:
need you guys to narrow it down, give configs, details of how you've got things wired up, etc.

this would be a good time for someone to start a troubleshooting section for the documentation and submit it to the project so that it can be added to and improved.

beakdoctor, 2015-06-10:
Well, it happens right after I've flashed the board so it should be default config I guess? Or are some things retained? Problem goes away if I flash with 1.8.1, and comes back if I flash with 1.9.0. It doesn't matter if I'm only powering from USB (RX is not on at this point) or via ESC (RX is on now). When I arm the board it stops blinking.

hydra, 2015-06-11:
pretty sure it's going to be flashing because you have no battery hooked up to vbat connector which monitors battery by default

there are some PR's that will work out if no battery is connected coming too which may make your life easier.

see #955 and #952 and #954

hydra, 2015-06-11:
@beakdoctor @Arakon @kestrel253 can one of you volenteer to create a troubleshooting guide and add this?

beakdoctor, 2015-06-11:
I just reflashed with 1.9.0 again and turned off the VBAT feature but it doesn't make a difference. Would that indicate it's something else? 

With 1.8.1 I also have a solid red light on L0 on boot, but now it is completely off. 

I guess I could start a troubleshooting guide but I don't understand what's going on yet.

beakdoctor, 2015-06-11:
Curious thing happened just now after being connected for maybe 10 minutes. The green light stopped flashing and only the blue led is lit (connected via USB only). I then tried to calibrate the accelerometer and the flashing resumed. Left it for another 10 minutes and it stopped.

Chadster, 2015-06-14:
I have the same problem on both my Naze and Mullet boards.  On 1.9, L1 flashes constantly but also inconsistently as described above.  Note this also happens when the only connection to the controller is USB (everything else unplugged).  Everything else works normally and my quads fly great, but this prevents me from using the buzzer feature.  If I flash both boards back to 1.8 everything works fine.

It seems that the buzzer code was overhauled in 1.9 but the only documentation I can find doesn't list what the default values are or how to change, etc.  I will try fooling around with CLI settings but would e great if someone already figured this out and could document.

SteveAmor, 2015-06-14:
I thought I was in for an evening of "git bisect".  However, simply disabling the VBAT monitoring feature stopped the (L1) flashing that I have also observed since upgrading to 1.9.0.

I suggest this particular issue is closed.

beakdoctor, 2015-06-14:
Sadly, I tried that before with no result. Since it worked for you I guess I have to try again to absolutely make sure it's off. I will do that tomorrow.

SteveAmor, 2015-06-18:
@beakdoctor did you test again?

kestrel253, 2015-06-18:
I disabled VBAT and this fixed the problem for me.  I think we can close this issue.

digitaldeli, 2015-06-22:
I have the same issue. The green light will blink at an irregular frequency when not armed. If I pick up the quad and move it around that seems to affect the blinking. My set up:

-Dragonfly32 Pro
-cleanflight 1.9
-quad X
-VBatt is on and wires are connected. I get the voltage in the GUI
-CPPM with FrSky D4RII
-FrSky telemetry using soft serial 1

Anything else that is helpful, I can post it. Not sure how much detail is needed.

beakdoctor, 2015-06-23:
Sorry for taking my sweet time testing this again, but as I thought disabling VBAT does nothing for me. I still get the blinking light.

Lordnibbler, 2015-06-25:
disabling VBAT also did not solve this for me.  i also am unable to arm while this green LED is flashing, naze32 rev 5 1.9.0

skaman82, 2015-06-25:
On all my naze full and acro boards, everything works just fine, its just the strange blinking of the green led. In 1.8.1 it wasn't there. Disabling VBAT doesn't help. So not a real issue for me but still strange.

BobSnorkel, 2015-06-26:
Same here... all of the above. NAZE32 with 1.9.0 and only after firmware upgrade. Tried disabling VBAT, Failsafe, readjusting Min/Max Throttle limits (as this had caused me issues in the past so i thought I'd try), calibrated Acc and all that Jazz. Was flying great still until today where the board arms fine, motors start ok but then everything just dies after upping the throttle for take off. Have to disconnect then reconnect the battery to reset everything but it just happens again at each take-off attempt... Stumped.??

SteveAmor, 2015-06-26:
@BobSnorkel I've had that happen to me and it was a faulty cell (on load) in the battery pack. Tested out fine with no load. Ended up with a battery checker on the balance connector and noticed the drop in voltage on one cell when I spooled up

tejo, 2015-07-09:
Same here with a brand new afromini 32 . Green light flashing. No way to stop it.

tejo, 2015-07-09:
@hydra the commit that introduces the issue is https://github.com/cleanflight/cleanflight/commit/e6733b4dfccf8a32c297f8ccdd214ae3e1619a55 . Founding was easy, spot the reason for me is a total different story :). Hope this can help somehow. Thanks

hydra, 2015-07-09:
@tejo That's expected, warning led was tied to beeper in e6733b4

The main reason for blinky led are:
a) something is causingthe beeper to beep. (e.g. low battery alarm, mode range for beeper, etc.)
b) low battery with VBAT enabled
c) rx loss detected.
d) aircraft tilted to much.

BobSnorkel, 2015-07-09:
@SteveAmor Thanks, totally was the faulty cell that was causing the drop on spool. 

@ Hydra As for the blinking Green LED. Your a - d aren't the causes in my case as i have eliminated all of those previously and yet still have the 'Blink'. Goes solid on Arm though so at this stage I'm really not concerned as it really doesn't seem to affect anything other than the retinas! B-)

tejo, 2015-07-10:
@hydra that make sense, thanks :+1:

Kamikid, 2015-07-26:
I am having the same issue on latest firmware. Never had this problem on older firmware versions. LED just keeps flashing green and beeper beeps for no apparent reason. I have tried disabling VBAT, kept quad leveled, battery charged, rx connected.... Hopefully we find a fix soon!

tejo, 2015-07-27:
@Kamikid have you tried to recalibrate? After recalibration I have no issues.

varesa, 2015-07-27:
I flashed a quad with 1.9 and made a few flights. At some point I got the green flashing led and couldn't arm. The issue went away after several power cycles. The quad was level and was allowed to complete calibration undisturbed. 

I have VBAT disabled and am using a D4R-II in PPM mode. The rx had a green light indicating okay signal and the PPM wire seemed to be properly connected.

Kamikid, 2015-07-28:
@tejo Yes, I have tried re-calibrating the ESC's and Acceleromter

Josrau, 2015-11-07:
I have the same issue but with a twist. I have a Rev5 Naze32 and originally when I set up my quad with Clean Flight 1.8, everything went well, had it up and flying in no time. Did a bit of PID tuning and 3-4 weeks ago started noticing some problems. At first, no beeps out of the FC so a week ago decided to update the firmware to 1.9.0, plugged it in to the PC via USB port and had the green light constantly flickering, went through and upgraded the firmware and rebooted, then the next issue started, just sat there and would not complete the reboot cycle. So unplugged the USB cable and re-plugged the USB, now the Blue, Green and Red led are on. Connect to the board and it looks like it has loaded the 1.9.0 firmware. Go to the settings tab, go through the accelerometer calibration on flat surface, again hung up in reboot cycle. Disconnect, re-connect, Blue, Green and Red Led's on, Green flickering still, go to the settings tab with the image of the quad and notice that the image of the quad is jumping on the yaw axis even though I am not touching or moving the quad.

Do I have a faulty FC? Can anyone offer some suggestions on getting this FC back working?

Thank you for your help...Joseph.

Josrau, 2015-11-09:
No comment from any one?

I would think someone may have some knowledge about this problem.

Please help, see above. Thank you.

rolfz, 2015-12-06:
I had the same problem, and my NAZE32 rev5 just did not work with Cleanflight. Big issue is that Rev5 schematic is not available, so it is impossible to check if it is really compatible ! I guess not. Does someone have a rev5 schematic available that would allow to check the firmware source and modifiy it if required ???

Chadster, 2015-12-07:
This is one of many reasons why I upgraded my Naze to Betaflight.  Not only
does the buzzer work correctly, but the flight performance is like night
and day over Cleanflight.

brandonhill, 2016-05-02:
I just switched to beta flight and no longer experience the flashing green LED issue.

closed 2017-02-25 by hydra