cleanflight/cleanflight #1013: IO - I2C receiver implementation

iforce2d, 2015-06-14:
Firstly, thanks for all the work on this project! This is not an issue with Cleanflight, but the IRC chat which seems to be the recommended contact method is not very convenient... aren't there some forums somewhere?

I am adding "FEATURE_RX_I2C" in my fork, and I'm having some problems with the I2C interface. The goal is to use a nRF24L01 radio module with an arduino acting as an I2C slave to give RX inputs to the flight controller. The nRF24 transceiver modules are quite cheap and allow for digital data to be exchanged in both directions, some uses for which could include automated flight (eg. follow-me mode, coordinating multiple aircraft, or simply changing waypoints on the fly), digital telemetry, ground-based blackbox etc.

This seems quite similar to FEATURE_RX_MSP so the additions I've made are pretty much a copy of that, with the exception that an i2cRead is done from during updateRx in rx.c to get the values from the arduino. The problem I'm having is that the I2C bus gets messed up, specifically:
- the i2cRead for the RX values fails about 50% of the time
- values for gyro, mag and baro are all zero (or very occasionally, twitching with garbage values)
When I revert the RX type to something else (eg. PPM) so that the arduino is silent on the I2C bus, there are no problems. Strangely, the accelerometer values are unaffected and are read correctly in all situations.

I'm pretty sure all wiring is correct because when the i2cRead succeeds the values are correct. Counting I2C interactions on the arduino side shows that the arduino sees about 5% less reads than writes. In other words, for every time the FC writes a byte to the arduino to request content, the follow-up read of the content does not always arrive. 

I have tried a bunch of things, all to no avail:
- pullup resistors (4.7k)
- increasing I2C_DEFAULT_TIMEOUT
- different slave addresses
- changed I2C bus speed in arduino lib to 400kHz
- disabled USE_MPU_DATA_READY_SIGNAL and USE_MAG_DATA_READY_SIGNAL
- wait DELAY_50_HZ between uses of i2cRead
- enable SOFT_I2C (this caused the configurator to be unable to connect at all)

What else am I missing, that might cause the I2C bus to be corrupted?
Is there a better place to ask a question like this? (not IRC!!)

The hardware is a Flip32+.
My changes here: https://github.com/iforce2d/cleanflight/commit/35d3b49c751f32cab349bb6a7ed9e8c95a64f091

hydra, 2015-06-17:
how do you know the problem isn't with the AVR code?

have you tried a simple AVR sketch that implements I2C slave that just returns an incrementing number or something simple?

have you tried using two AVRs to talk to each other?

have you verified that the I2C bus signals are clean using a scope?

have you tried using a logic analyzer?

iforce2d, 2015-06-17:
After some more investigating it looks like the problem is indeed on the arduino side. I found that the minimum arduino sketch needed to cause the problem is simply:

void setup() {Wire.begin(0x42);} void loop() {}

... where 0x42 is the address the FC is using for i2cRead(). This alone is enough to cause garbage values to be read by the gyro, mag and baro. The acc is not affected. If the AVR does not do Wire.begin(), or the FC does not do i2cRead() for that address, the I2C bus remains clean.

Yes, the arduino sketch is transferring just a single byte.

Yes, I've had two AVR's talking to each other through I2C no problem, and I used the slave sketch from a previous project to start this modification. As I mentioned the i2cRead() call does succeed about 50% of the time, and when it does the values it receives are correct.

I don't own a scope or logic analyzer, but it's probably about time I got one. I'm a software developer by trade who just got into arduino and multirotors quite recently. To 'debug' this so far I'm setting values in rcData[] and looking at the receiver bars in the GUI :)

iforce2d, 2015-07-06:
I tried three arduinos and two Flip32s and every kind of troubleshooting tip I could for about two weeks, and eventually gave up on doing this with I2C. I then tried some tricky soldering and using SPI1 as mentioned here https://github.com/cleanflight/cleanflight/issues/532 with similar results - the data would arrive correctly about 20% of the time.

After all that, I ended up doing it with software SPI on receiver pins 5-8 and it worked immediately! What the hell.... anyway, this is a nice solution because it does not need any arduino go-between. I was worried that it might be slow but so far all seems good.

Is software SPI worth doing a pull request for? I could also add my code for FEATURE_RX_NRF24 but it's probably quite a niche feature given that you also need to make your own transmitter, and will need to build your own cleanflight if you want to modify the packet structure.

closed 2015-07-06 by iforce2d

ledvinap, 2015-07-06:
I had partially ported Cleanflight to Crazyflie. But I stopped when I realized that it won't lift any of my receivers ...
If onboard NRF24 is supported, the port can be probably finished ...

iforce2d, 2015-08-31:
@ledvinap Sorry I forgot to reply here, but a few days after your last comment I committed the nRF24 stuff to a branch on my clone: https://github.com/iforce2d/cleanflight/tree/nrf24

I made a video explaining the connections etc: https://www.youtube.com/watch?v=K10qpQ47Q6Y

I've been using it for a couple of months now without any major issues. For some reason it does not work when I power the nRF24 from a separate regulator, it only works when I power it from the 3.3v pin of my Flip32. For use as a non-acking receiver this is fine, but I don't think that FC pin would be able to supply enough power for a PA-type nRF24 to transmit, so I have not tried any telemetry with it yet.