cleanflight/cleanflight #1014: Unable to set MPU6050 gyro_lpf to 256

borisbstyle, 2015-06-15:
accgyro_mpu6050.c line360
bool mpu6050GyroDetect(const mpu6050Config_t *configToUse, gyro_t *gyro, uint16_t lpf)
...
gyro->init = mpu6050GyroInit;
    gyro->read = mpu6050GyroRead;

    // 16.4 dps/lsb scalefactor
    gyro->scale = 1.0f / 16.4f;

    if (lpf >= 188)
        mpuLowPassFilter = INV_FILTER_188HZ;
    else if (lpf >= 98)
......

ledvinap, 2015-06-15:
This is quite reasonable - setting 256Hz changes data sampling frequency (to 8kHz), it may cause problems elsewhere ...
Al least set sample rate divider to /8 if you want to use 256Hz ... 

https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/accgyro_mpu6050.c#L340
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/accgyro_mpu6050.c#L394-L398

borisbstyle, 2015-06-15:
Ok thanks for the explenation. Should this go into the master?

ledvinap, 2015-06-15:
My personal opinion is that disabling LPF2 without reading data at 8kHz makes little sense. MPU6050 uses first order internal filter (according to delay from datasheet), so using high cutoff frequency will cause aliasing problems. With 2ms looptime Nyquist frequency is 250Hz, all higher frequencies will be aliased into passband (and noise at 500Hz is attenuated only by 50%).

borisbstyle, 2015-06-16:
@ledvinap 
Why is it different for MPU6000? CC3D users are able to get 256.

ledvinap, 2015-06-16:
My guess is that it is copied from different source than 6050 ...
It may be evidence that 256Hz works, so implementing it for 6050 will need only basic testing

borisbstyle, 2015-06-16:
Well I flew a few quick packs on 256 on my mini h. It seemed fine, but gave more noise as expected.
I personally don't mind it as it is now, but I do mind the difference between 6000 and 6050, what can give different results on different boards.
To my opinion both should do simillar. What one would you suggest? Keep it safe 188 or both to 256?

BTW is that sampling rate modification really needed from your code?

borisbstyle, 2015-06-16:
Look at the different behaviour. What happens when you set it to 189 on CC3D? Does it stay the same or what happens as there is no case for that?

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_spi_mpu6000.c#L277-L304

vs

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/accgyro_mpu6050.c#L375-L389

ledvinap, 2015-06-16:
I don't know ... 256Hz with note about sample rate is probably OK now ... 

In my case i use 6050 FIFO to get samples at 1kHz and than filter them onboard (looptime is integer multiple of gyro frequency). With I2C 1kHz is about maximum rate to read data out and I want it to be consistent. Whit spi connected gyro I would consider reading at 8kHz and doing most filtering myself ...

ledvinap, 2015-06-16:
as for mpu6000  - default gets used obviously. 6050 approach with ranges is better in this respect ...

borisbstyle, 2015-06-16:
Yes. Different behaviour on different boards should be minimized. I agree about the MPU6050 code where even I sometimes forget those values in the field still made it possible to change.

borisbstyle, 2015-06-16:
https://github.com/cleanflight/cleanflight/issues/1014

ctzsnooze, 2015-06-17:
You're probably well aware of the code from multiwii 2.3:

in config.h....

      //#define MPU6050_LPF_256HZ     // This is the default setting, no need to uncomment, just for reference
      //#define MPU6050_LPF_188HZ
      //#define MPU6050_LPF_98HZ
      //#define MPU6050_LPF_42HZ
      //#define MPU6050_LPF_20HZ
      //#define MPU6050_LPF_10HZ
      //#define MPU6050_LPF_5HZ       // Use this only in extreme cases, rather change motors and/or props

in sensors.cpp:

    //MPU6050 Gyro LPF setting
    #if defined(MPU6050_LPF_256HZ) || defined(MPU6050_LPF_188HZ) || defined(MPU6050_LPF_98HZ) || defined(MPU6050_LPF_42HZ) || defined(MPU6050_LPF_20HZ) || defined(MPU6050_LPF_10HZ) || defined(MPU6050_LPF_5HZ)
    #if defined(MPU6050_LPF_256HZ)
        #define MPU6050_DLPF_CFG   0
    #endif
    #if defined(MPU6050_LPF_188HZ)
        #define MPU6050_DLPF_CFG   1
    #endif
...
    #else  //Default settings LPF 256Hz/8000Hz sample
        #define MPU6050_DLPF_CFG   0
    #endif

then

  i2c_writeReg(MPU6050_ADDRESS, 0x1A, MPU6050_DLPF_CFG); //CONFIG        -- EXT_SYNC_SET 0 (disable input pin for data sync) ; default DLPF_CFG = 0 => ACC bandwidth = 260Hz  GYRO bandwidth = 256Hz)

So in multiwii the default is 8kHz sample rate with 256Hz dlpf in the 6050.   When selecting any dlpf rate value below 256 in the 6050, the sample rate is always 1kHz.

I and many others have used 256Hz dlpf setting in multiwii without any problem whatsoever at loop times slow as 3500us. It was obviously crisper than 42Kz.

It doesn't make sense to me why why CleanFlight is not making use of 8KHz sampling / 256 Hz lpf in hardware.  

It would be easy to change the code appropriately... I don't care much about compatibility with other things but if we have the choice of 256 we should get 256, unless there is very good reason to know this would be bad.

Cheers - C

ctzsnooze, 2015-06-17:
Even in CleanFlight where it writes to the 0x1A register, it says this:

    i2cWrite(MPU6050_ADDRESS, MPU_RA_CONFIG, mpuLowPassFilter); //CONFIG        -- EXT_SYNC_SET 0 (disable input pin for data sync) ; default DLPF_CFG = 0 => ACC bandwidth = 260Hz  GYRO bandwidth = 256Hz)

The note indicates the default should be 0 (256), but in CleanFlight it's actually 1:

    enum lpf_e {
    INV_FILTER_256HZ_NOLPF2 = 0,
    INV_FILTER_188HZ,
    INV_FILTER_98HZ,
    INV_FILTER_42HZ,
    INV_FILTER_20HZ,
    INV_FILTER_10HZ,
    INV_FILTER_5HZ,
    INV_FILTER_2100HZ_NOLPF,
    NUM_FILTER
    };

In cleanflight option 0 is described as INV_FILTER_256HZ_NOLPF2 but I don't know what that means, there is an LPF in the 6050 and there is no other LPF in software anyway.

This should definitely be fixed imho, I'll make the change now in my version here around 375:

    if (lpf >= 256)
        mpuLowPassFilter = INV_FILTER_256HZ_NOLPF2;
   else  if (lpf >= 188)
        mpuLowPassFilter = INV_FILTER_188HZ;

simple enough to do, and to test.

C

ledvinap, 2015-06-17:
The problem with 256Hz is signal aliasing. Nyquist frequency of register readout is important (`1/looptime`).
With default 3.5ms looptime, the readout frequency is 285Hz (actually there will be switching between 333 and 250Hz if MPU6050 registers are updated at 1kHz) and Nyquist frequency is 143Hz. So any signal over 143Hz will be reflected into 143Hz band (and signal at 285Hz will be seen as DC offset! 333/250Hz switching will complicate it a bit, maybe with even worse results). This is both for GYRO and ACC readings (ACC 1kHz internal sampling will create aliasing on its own, only mechanical filter can help).
6x3 propellers on my 250 size quad run at 160 Hz at hover (9600rpm) and go to about 280Hz at (limited) full throttle (if my spectrogram is correct), so vibrations are in 'problematic' band. On larger quads rpm will be lover, but there will be harmonic frequencies present ... 

Also keep in mind that internal LPF filter is only first order - signal at double the cutoff frequency will be attenuated only by ~ 50%

One big advantage of disabling LPF2 (INV_FILTER_256HZ_NOLPF2) is that internal registers are updated at 8Khz, so there is less jitter when sampling at looptime, this may improve things considerably (or not ... )

ctzsnooze, 2015-06-17:
I get a bit confused when you say "disabling LPF2 (INV_FILTER_256HZ_NOLPF2)".  My reading of the 6050 data sheet is that the LPF in the 6050 cannot be disabled, when the 0x1a register is zero, the LPF is 256 and the sampling rate is 8kHz.  What is the 'LPF2' ?

If the user has a loop time of say 1250, that is 800Hz, Nyqvist is 400 - data filtered to 256 should be not so bad?  

I do know for absolute sure that my multiwii craft flew really well, and without issues, when 6050's were set to 8kHz/256Hz.  This has not been my experience in CleanFlight.

Using blackbox I do not see significant P noise with gyro_lpf at 256 and the motors seem smooth.  Only if I enable D is there a problem - and a big one.  I do see obvious exaggeration of higher frequency noise in the D component.  Just set D to zero in the PID page and the exaggeration disappears.  So the D term calculation requires an LPF to stop the delta D value getting higher and higher as frequency of noise goes up and up (even if the amplitude does not change).  Boris has a solution that I think is very effective and necessary with his PT1 code.  With this code I think proper gyro_lpf at 256 should be OK.  If not something else is wrong.

ledvinap, 2015-06-17:
From constants is seems there are two filters in MPU6050 - one with 256Hz cutoff (and 8kHz output) and then second LPF with selected frequency(with 1kHz output). But maybe it is inherited from older device ...

undocumented DLPF == 7 may disable the first filter ... 

It would be interesting comparing blackbox logs with different settings ... Honestly I don't know what to expect. On my quad motor vibration is noticeable, so I try to fight it ...

As for D-term, LPF is way to go (I would personally test second order Butterworth, but that is just my guess).

borisbstyle, 2015-06-17:
Yes I tried the real 256 and different variations to it, but on my quads things were only made worse. 188 was really sweet spot on my quads.
There must be a reason why it's disabled here. If anyone wants to spend some time testing. Be my guest :)

I tested butterworth and it works really well even better than PT1. Only thing I had is that the graphs looked really nice, but I had difficulties to retune the bumps in hard flips and rolls. Not sure why, but couldn't get it flying as well as with pt1.
I have some logs from butterworth testing.
I also tested without moving average, while on pt1 element I was still using it.
Will sure spend some more time on it once I get to it.

ctzsnooze, 2015-06-17:
I understood that if you select 256 for lpf the 6050 sample rate is 8000.  For all other lpf settings the sample rate is 1000.  As far as i could tell there was only one filter.

I tested adding both a first order software LPF on incoming gyro data and/or a Dterm LPF as per Boris's code.  

For several gyro_lpf values I tested a range of different FC filter settings.  In each case I first tried to find a P value that after sudden stick input took say 2 cycles to damp down naturally while lightly hand held.  This was easy to control with not so much D.  I'd then tune it as best I could, and assess the extent to which the motors jumped up and down as I throttled up.  If they stayed much the same then power delivery was good, but if they jumped all over the place, of course, it wasn't.

I only thoroughly tested LuxFloat.

At gyro_lpf of 256, with default 3-point D filtering only, (no LPF filters) it was impossible to tune, far too much vibration.  Required 42 to be able to be tuned even then not good.

Adding my LPF's, I started with gyro_lpf of 256.  At 200Hz for gyros 50Hz for Dterm, it was better than stock code, but motor PPM output fluctuated more than acceptable, worse with more P.  Most of this was due to P; even if I set D to 0 it persisted.  After stick movements, it was very responsive, the frame oscillation was high frequency, and although possible to control stick oscillation with D (didnt need much), the motor drive jiggled up and down too much.

With more filtering - my filter set at 100 for gyros and 20 on D term, the quad was much better, and at 75-17 better still.  At lower LPF's I couldn't quite get so much P, and I needed a little more D (it was being filtered out).  Not much different though.  Any of these settings I think would fly very well.

I tried lower gyro_lpf as well.  With gyro_lpf at 98, 75-17 or 100-20 FC filters were really good.  Motor PPM variation across the throttle band was minimal (not much D or P noise) and it was relatively easy to tune.

With gyro_lpf of 42 it was a bit difficult to get crisp responses, but was sort of but OK at 75/17 or 100/20, only the Dterm filtering doing anything like this.

The upshot of it all was that for this particular quad, there was too much stray noise to benefit from 256.  It didn't matter where the gyro data was filtered, in the 6050 or in software on the FC, it needed filtering somewhere to stop the higher frequency incoming motor/prop noise from being sent back out to the motors and wasting energy.  

I don't think it matters much where the main gyro LPF is done, this frame just has to have it done at around 75-100Hz, and then have a Dterm filter around 17-25.  These values seem about best for this frame ( P is 1.2, D 20).

I think that slightly heavier less vibration prone frames might be better with the higher value gyro LPF values, and with the appropriate D term filter, but not small fast racing type frames.  My little 140g quad is very vibration prone and I can't get past about 100Hz really.  Interesting that Boris finds 188 to be good on his better balanced units.  That makes sense.  

What is truly amazing, however, is how absolutely useless the stock code is on my frame.  It absolutely demands a gyro_lpf of 42 and can barely be tuned like that with crazy motor speed fluctuations, all D related.  Add Boris's D term filtering and it's a whole different ball game, suddenly it's far smoother and much more easily tuned even with gyro_lpf of 98.

ledvinap, 2015-06-17:
@borisbstyle: The reason may be totally unrelated to CF ... 

Second order Butterworth has 180deg delay, first order will have only 90deg. It may be complicating things in D-term, but IMO it should not be major problem.

The averaging influence is interesting. Can you test what happens when you only insert 1 cycle delay instead of averaging? (use only older deltas)

ledvinap, 2015-06-17:
@ctzsnooze : thanks for testing description, it is very interesting.
D-term is probably much less sensitive to delay, so there is more room for filter. It may be interesting to use second order filter with double bandwidth - delay and high frequency noise attenuation should be about the same, but more bandwidth will be used for control.

As for location of filters - I still think that high frequencies should be attenuated as much as possible before sampling MPU6050 output. That means setting MPU6050 cutoff as low as practical (as long as it does not degrade performance / as long as there is delay budget to spend) - MPU6050 cutoff should not be higher that internal filter (it may be best to set both filters to same value).

ledvinap, 2015-06-17:
@ctzsnooze : You may try to improve motor balancing. I had to use weight on motor bell 90deg to propeller (gemfan carbon nylon 6x3), after that I decreased vibration considerably.

borisbstyle, 2015-06-18:
@ledvinap 
Yeah it seems that low looptime really need the moving average. And I have a feeling that a bit delay in D has positive effect.

Checkout this butterworth filter I was testing. I think I only screwed it up on luxfloat, but other pid controllers seem fine.
https://github.com/cleanflight/cleanflight/pull/1024

ctzsnooze, 2015-06-18:
@ledvinap - I was just about to ask about the sequence of gyro filters, seeing as I have implemented one in software in the FC that I can run independently of that in the mpu6050.  I've been thinking about it all day!  :-)  I wish I was as expert as you!

My thoughts were that if set the 6050 to sample at 8kHz and filter at 256, then the risk of primary sampling aliasing issues should be very small, and the stuff we want to see should be accurately represented.  On the other hand, at any LPF under 256, the sample rate drops to 1000Hz, which could be a simple multiple of 500Hz sometimes used to control ESC's, and perhaps may be more prone than 8Khz to primary sampling aliasing.  So I kind of figured that the best representation of the data would be when using 256 / 8kHz.

The next potential point of aliasing is when the 6050 output is requested each loop through the FC.  With 700-800Hz loop times there is the potential for aliasing particularly at loop time frequencies.  If the 6050 is filtering at 256, there is only 2 octaves attenuation of such frequencies.  Sampling at 188 is hardly different, but comes at the cost of only 1kHz gyro sampling in the first place.

On the other hand, if we ran the mpu6050 lower down at say 188, or perhaps more significantly 98, there should be much less chance of aliasing by the FC since its input should not change much at frequencies close to the loop time.  The risk is that there may be more sampling aliasing in the 6050.

I figured it might be better to run the mpu6050 at 256/8k, then filter gyro in sw at say 100-200 (with a separate lower Dterm filter for delta, which we know works well). 

On my noisy quad I figure I should closely compare these choices (all with a Dterm filter):  

(a) 6050 at 256/8k and FC filter at 100, vs 
(b) 6050 at 88 only vs 
(c) 6050 at 88 and FC filter at 100

I think late last night they seemed similar.

For Boris I wonder if running the 6050 at 256/8k with a second sw filter in the FC at (say) 200 would be better or worse than just the 6050 at 188/1k.  It would be interesting to find out.

I know I can make my quad smoother, I deliberately use old props etc for these tests to challenge the system.  :-)

C

borisbstyle, 2015-06-18:
@ctzsnooze I don't know, but it somehow doesn't feel right to disable onboard gyro at all and than increase the codesize to still do full gyro filtering onboard.
MPU6050 also does 1st order filtering so you will not win delay decrease?

At the end we want to have it flyin really well instead of only generating nice graphs.

Butterworth filter. Nice graph, but untunable and flies really bad. 
https://dl.dropboxusercontent.com/u/31537757/cleanflight/LOG00221_butterworth.TXT

PT1 element Super locked in!
https://dl.dropboxusercontent.com/u/31537757/cleanflight/LOG00222_pt1.TXT

ledvinap, 2015-06-18:
@ctzsnooze : you are mixing sampling rates

Selecting MPU6050 mode with 1kHz output rate inserts filter that runs (hopefully) on 8Khz, but only one result out of eight is stored in output registers. This is lossless operation (ignoring rounding/quantization) as long as there is no frequency content over 500Hz (Nyquist frequency). Selected lowpass should ensure his. 
Gyro sampling is still done at 8kHz (and first 256Hz filter is still present)

The sample rate is still 8Khz (maybe even higher) for gyro, with analog lowpass filter before ADC. 

For ACC there is probably 260Hz analog filter before 1kHz ADC.

Using 8KHz mode will work great if you read out all data available - at 8kHz. That is impossible with I2C. 
But you are reading data at loop frequency (<1kHz), so only advantage of 8Khz mode is that sampling jitter is 8x lower (sample is at most 125us old). With 1kHz mode same samples will be almost 1ms delayed.

ctzsnooze, 2015-06-18:
There is no evidence for analog filtering on the 6050 before its ADC.  They are - unfortunately - all digital low pass filters. See:  http://www.cdiweb.com/datasheets/invensense/PS-MPU-6000A.pdf
[image]
It would be great if the filters were digitally controlled analog before digitisation, but I think not.  If they had digitally-controlled analog LPF's before digitisation, they would say so.

Yes, that's a good point, an extra advantage of reading from samples at 256/8k over say 188/1k is that at 8k there should be a tiny bit less jitter - as you say, any sample is at most 125us old vs up to 1ms old at 1kHz.

The other advantage of 8k is that sampling aliasing will be occurring around 4k, hopefully well above most quad motor frequencies (and hopefully relatively easy to physically damp out), whereas at 1k there is the likelihood of sampling aliasing for frequencies around 500Hz - first and second harmonics of motor rpm are around 500Hz.

A DLPF is always active with current code in the 6050; if we want 8k sample rate, this will always be present and at 256Hz.

So I am really curious - if I then run a second first order LPF in the FC code at (say) set around half that frequency - 128Hz for example - would that be such a bad thing?

As far as I can tell, probably not, as you said earlier.  There would be greater attenuation of frequencies above 256Hz - 12db/octave vs 6db/octave - that's exactly what we want.  :-)

Phase shifts would exist past 100Hz and become greater past 200Hz, but amplitudes past 200Hz would be much lower also, and 200Hz is getting fairly high.

That's why I personally think it better for well-balanced quads that really want the best possible responses to set gyro_lpf to 256.

If motor signals start responding to noise, and if they are mostly D, then lower the Dterm filter; if they are mostly P, add you don't want to lower P from the current value, then activate a software LPF in the FC at a frequency that stops the incoming noise getting re-directed as Pterm noise to the motors. 

This is what I was doing late last night.  It became obvious, after setting a Dterm filter correctly, that increasing functional gyro LPF to higher frequencies increased the quantum of P noise transfer to motors, as expected.  The goal for me was to avoid either P or D noise transfer to the motors (i.e. see the motors smoothly ramping up with throttle, not jittering up and down like mad), while retaining responsiveness and the highest possible filter values.  Quite a set of compromises!  :-)

Boris definitely stick with gyro_lpf at 188 and the dTerm PT1 filter on your quads. But if you have a moment, try setting gyro_lpf to 256 and put a PT1 filter on gyro in the FC set to at 188 also.  See what that flies like - if the two are any different.  I anticipate being essentially indistinguishable, perhaps you may be able to increase P a fraction with two filters in series without oscillation.  If so that would be good.

ledvinap, 2015-06-18:
The analog filter is not configurable. It is simple aliasing ADC filter. But I may be wrong .. 

> The other advantage of 8k is that sampling aliasing will be occurring around 4k

This is not true. Even in 1kHz output modes gyro is sampled at 8kHz and filtered.  Any signal over looptime Nyquist will be wrapped, it does not matter if gyro output is 1kHz or 8kHz (ignoring jitrer here). But lower MPU6050 LPF means that  high-frequency signals are attenuated more, so less noise is left to alias.

Using two first order filters in series results in second order rolloff - this is usually intended.

Delay is intrinsic in filtering. Decreasing bandwidth to half aprox. doubles delay. And second order filter has about double delay of first order one.
It is probably better to use higher order filter (better attenuation on very high frequencies) to lowering bandwidth.
But there is limit where increasing gyro bandwidth won't improve bandwidth of whole system. There is also trade-off between bandwidth and noise. From borisbstyles experiments it seems that this frequency may be lower than 100Hz, but that is just my wild guess.

ctzsnooze, 2015-06-18:
OK cool yes I'd be intending second-order rolloff, and the delay shouldn't be a problem, I think. 

It would be great if there were some better spec sheets on the mpu6050.  None I can find mention *any* analog front-end filtering at all; no mention of frequency response etc.  Very annoying.

I'm confused a bit by what you say about gyro sampling, because this is what Inversense say about sampling in the 6050:

[image]

They say Gyroscope Output Rate = 8kHz (when acc_lpf = 256), and that at all other acc_lpf settings it is 1kHz.  The 6050 'Sample Rate' is different, and independent of the gyro sample rate, it's the rate at which registers are updated.  That's why I interpreted the sample rate when acc_lpf is 256 to have a higher Nyqvist frequency - if I'm wrong that's OK but that's the way their documentation goes...

I think Boris finds his quads go best around 200Hz (with his Dterm filter around 17-20 - but his quads are more solid, higher in inertia and have better balanced motors and props than my light acro crash'n'vibrate test platform.  :-)

ledvinap, 2015-06-19:
I don't know ... it did not make much sense until I considered power consumption ...

I did test flights with 256 and 188 setting, recording all gyro data at 1kHz.
When I plot power spectral density for whole flight, 188 has considerably lower energy at high frequencies, 256Hz has very pronounced second harmonic at Nyquist frequency.
Overall 188 looks better, there is less signal even at low frequencies.

[image]

When i compare only hover section of my data, it is not that clear. High frequency noise is attenuated, but the situation is inverted at sub 100Hz frequencies .. It may indicate aliasing, or it is caused by wind or by noise that is not removed by onboard filter:
[image]

And here is PSD of data filtered by 'throttle is max, other sticks centered'. I did not consider this graph when flying, so results may be biased.
[image]

The peak at half motor frequency is quite strange and may signalize aliasing problems ...

But it is easily possible I am comparing apples to oranges ... I don't know what to expect..
I anyone has idea what to test, please tell me ...

ledvinap, 2015-06-19:
BTW: here are spectrograms of both filter settings (sorry for the size)
188Hz LPF:
[image]
256Hz LPF:
[image]

:

ledvinap, 2015-06-19:
It would be interesting to analyze data recorded at 8kHz, but I don't have SPI connected MPU6050 and faster readout is problematic over I2C (I can probably get to 2-3kHz, but it wont be trivial)...

ctzsnooze, 2015-06-19:
This is really great!  Spectrograms are so incredibly helpful - I use color spectral arrays to track changes in brain electrical activity at work so I'm quite familiar with them. 

May I clarify if the vertical axis on the line-graph spectrograms is dB or linear?  And for the colour array spectrograms, is the colour scaling linear or logarithmic?

If the line spectrograms are logarithmic, and I guess they are, then we need steeper roll-off generally for gyro data than 3dB/octave, I think, and that means a second gyro lpf in software after the main one in the 6050.

That data basically includes raw frame vibrations and also vibrations arising from feedback of noise via the motors.  With constant stick input, we know there is noise amplification/feedback via the motors if the motor output signals are jittery (assuming we fly in still air in a straight line).  But we cannot separate the two in the spectrogram.

I think that the 'real' data is all on the left, below 50Hz.  Everything else I think is essentially unrelated to flight performance.  

Personally I think that we need more than 3db/octave filtering of gyro noise.  I will stick with applying a second gyro LPF in cleanflight, probably set at the same value, or a bit below, the LPF in the 6050, most likely at 98 or 188 Hz.  It would be very interesting to see spectrograms from your system with 6db/octave gyro filtering, and to hear Boris's impression of the flight characteristics.  We should test non-ideal systems, not just super well-balanced systems.

I've reached the conclusion that Boris's Dterm filter should be tuned.  The ideal value depends primarily on the natural un-damped P oscillation frequency, which is, in turn, determined by PID P, frame/motor physics, and gyro filtering values.  For a given frame, it seems to mostly be determined by gyro LPF.

When the undamped P oscillation frequency is 'fast' (say 20Hz or maybe higher, e.g. with higher gyro_lpf settings), you'll tend to get fast responding super 'locked-in' 'rock-solid' feel.  But controlling those oscillations with D gets tricky, and more noise is getting back via the FC to the ESCs.  Boris's Dterm filter must not cut the frequencies that require Dterm, or they may be difficult to control.  If Boris's Dterm filter is too low, it may inhibit the D response, leading the user to turn up PID D to compensate (might be OK), or get phase shift in the correction signal that makes D unusable.  

In contrast, when the undamped P oscillation frequency is 'slow' (i.e. 5Hz or less, typical for gyro_lpf of 42 with not so hard PID P), Boris's Dterm filter can probably be set lower, better controlling D noise.

In simple terms, the higher the natural P oscillation frequency, the higher Boris's Dterm LPF value needs to be - and, vice versa, for lower natural P oscillation frequencies, the Dterm LPF can be lower (and therefore more effective at blocking D term motor noise amplification).

My much crude technique for testing the relationship between filtering and feedback of motor/prop noise to the ESC control signal is relatively simple.  

First, with props on, held in my hand, I do motor testing directly in the motors tab (throttle up with sliders, no Tx/FC involved).  There will be no jitter in the on-screen display of individual ESC/motor inputs because I control them directly.  There will be some frame vibration, I will feel and see it.  I make sure acc_lpf_factor is off (or the accSmooth filtering removed in imu.c) so I can see the magnitude of this 'intrinsic' frame vibration on-screen.

Then I switch off the sliders and use the Tx/FC to control motors.  I look to see if there is more jitter now than before, with a given set of filters.  If the motor control signals now jitter is all over the place, this will add to frame vibration, decrease motor output and flight controllability, and makes the motors hot. It means that frame vibrations are being amplified by the FC and fed back to the motors.   In contrast, if the variability of motor control signals is small (ideally insignificant), despite the normal frame vibration, motors will run smooth, power delivery won't be affected, etc.  

This way I can make changes to filtering and see what the actual impact on the system will be, at my desk.  It is a little dangerous, though.

If the FC induces motor signal jitter from noise, it is easy to see the extent to which this is caused by D.  Just drop D in the PID settings to zero.  If the motor control signal jitter largely disappears, it was largely due to D.  If it does not change, it was due not to D, probably to P.  It is possible to drop PID P temporarily to confirm this.  Also quickly adding extreme TPA helps demonstrate what might happen with greater overall gyro filtering or lower PIDs at higher throttle settings - e.g. testing by comparing TPA of 100 vs 0, starting at 1100.

So it is possible to determine the 'real-world' impact of the filtering options, and whether they are P or D mediated, relatively quickly.  It is also possible to 'tune' PID's (approximately) to suite the different filtering option, because filtering influences tuning quite a bit.  And it is easy at the same time to note stick responsiveness and natural P oscillation frequency in the absence of D.

I find that Boris's D term filter does a really good job of reducing D term noise amplification, essentially to insignificance, assuming that overall gyro noise is reasonably controlled.

[... cut ...]
We could then have a profile that tolerates out of balance props and provides smooth cool motors, accepting maybe has less precise control; another that requires a well balanced setup (and that we change props when dented) but gives faster, more precise control; and a third with extremely tight responses maybe at the limit where some noise may be fed back and we might accept that the motors might get a bit hot.  The PIDs could then be optimised for each profile since the optimal values do depend on gyro_lpf. 

I think we are really getting much closer to a good solution.

Thanks so much!

Cheers - C

ctzsnooze, 2015-06-20:
Made this movie showing how various filtering options affect the amount of noise-induced motor control variability as described above.  Please take a moment to watch it.  Fascinating stuff!

https://dl.dropboxusercontent.com/u/9505117/DLPF.mov

Note one small thing.  Different gyro_lpf settings the 6050 to change its acc filter also, so increasing gyro_lpf will increase the apparent 'displayed noise' in the motors tab of the configurator, even if in reality there is no such change in reality.  Also I had acc_lpf_factor set to 0 to maximise frequency response of the acc display in the motors tab.

In the video, I made, for each value of gyro_lpf from 42 to 256, these recordings of the motors tab screens:

- 'acc noise' at the given gyro_lpf while throttling up directly (without the Tx/FC involved)
- recorded acc noise and motor variability using Tx throttle with FC active, with the stock CleanFlight code (3-point D filter only)
- then repeated the test but using Boris's Dterm LPF and a second sw gyro filter in a 5:1 ratio, at 50/10Hz,75/15Hz, 100/20Hz, and 200/40Hz.

It's amazing how bad the standard CF code is!  Motor control signals go all over the place even with gyro_lpf as low as 42. Thrust is way down even at 98 - it's totally unusable with gyro_lpf of 256. 

Boris's Dterm LPF makes the big difference.  I found previously that a second basic gyro LPF also helps a lot.  

On my frame, a gyro_lpf of 98 and sw filtering at 100/20 seems a good balance between responsiveness and power; at 188 and 256 it's also OK at 100/20.  

I have to repeat the tests at say 10, 15 and 20 for Dterm LPF, without the second software gyro LPF, to check which is doing most of the work.

ctzsnooze, 2015-06-20:
Just had a thought that maybe it was mostly Boris's D-term filtering, or maybe just my additional software gyro filtering.

So repeated a midrange test with gyro_lpf at 98, with three possibilities:

 - gyro 100 with Dterm 20, 
 - Dterm 20 only, and 
 - gyro 100 only.

Here are the results:

https://dl.dropboxusercontent.com/u/9505117/98_G100_D20.mov
https://dl.dropboxusercontent.com/u/9505117/98_D20_no_G.mov
https://dl.dropboxusercontent.com/u/9505117/98_G100_no_D.mov

In order of reducing induced motor noise, best was combination of both filters, next best Dterm only, and last a second gyro filter only.

That's what I'll be doing!   :-)

Cheers, Chris.

pulsar256, 2015-06-20:
@ctzsnooze when you refer to gyro SW filtering, I take it, it is abased a on previous @borisbstyle's PR (gyroADC FIR Filter or something, can't find the PR anymore), right? I am following your findings with much attention as I have to deal with a particularly noisy frame right now and would like to reproduce your experiments.

borisbstyle, 2015-06-20:
@ctzsnooze
I see the guys from OP and taulabs also using soft filtering for the gyro and acc as well.
It might not be a bad idea after all, but I think it should be done differently to keep the P and D as much possible in phase.
So basically

borisbstyle, 2015-06-20:
@pulsar256 
That one was removes as it didn't improve anything during my tests, but that was on a clean setup and I actually never really verified that part of code from robert properly.
We need to determine a good soft filtering for gyro signal. just a 1st degree order filter to minimize delays. Perhaps the same filter algorythm can be used for gyro and acc as well, bit not  sure need some input from an expert
@ledvinap. What do you think of trying gyro/acc filtering on noisy quads? Check the logs from hydro for example in the dterm pr.

ctzsnooze, 2015-06-20:
@pulsar256  - my code is a bit scrambled now.  But it works.  It's only done in LuxFloat but the principles are simple....

Initially I just copied Boris's first order LPF code and applied it to gyroRate:

        gyroRate = gyroData[axis] * gyro.scale; // gyro output scaled to dps
	// CTZSNOOZE ADDED LPF as per Boris applied to gyroRate
	gyroRate = lastGyroRate[axis] + dT / (RC + dT) * (gyroRate - lastGyroRate[axis]); 

I then set  RC * 5 for the smoothing factor when applied to D delta, forcing its LPF  5 times lower than the gyroRate LPF.

To make life easier for testing I took acc_lpf_factor out of accSmooth in imu.c (the accSmooth calculation isn't smoothed anymore by acc_lpf_factor, but that's the only side-effect).  I then re-used it to hold my desired gyroRate LPF value.  That let me store different values for the the gyroRate filtering in different PID profiles, very handy.  I put a conditional so that if acc_lpf_factor was less than 10, it reverted to the default code with only the basic 3-point moving average filter.  So now I have this:

	if (imuRuntimeConfig->acc_lpf_factor > 9) {
	RC = 1.0f / ( 2.0f * (float)M_PI * imuRuntimeConfig->acc_lpf_factor );
	}
	else {
	RC = 1.0f / ( 2.0f * (float)M_PI * F_CUT);
	}

then this:

        gyroRate = gyroData[axis] * gyro.scale; // gyro output scaled to dps
		// CTTZSNOOZE ADDED LPF as per Boris applied to GyroRate  if acc_lpf_factor over 9
        if (imuRuntimeConfig->acc_lpf_factor > 9) {
		gyroRate = lastGyroRate[axis] + dT / (RC + dT) * (gyroRate - lastGyroRate[axis]); 
		}

and finally, to do Boris's delta smoothing code at a time constant 5 times longer than for gyroRate:

		delta /= dT;
	if (imuRuntimeConfig->acc_lpf_factor > 9) {
 		delta = lastDTerm[axis] + dT / ((RC * 5.0f) + dT) * (delta - lastDTerm[axis]);
 		lastDTerm[axis] = delta;
 	} else {
		deltaSum = delta1[axis] + delta2[axis] + delta;
		delta2[axis] = delta1[axis];
		delta1[axis] = delta;
		delta = (deltaSum / 3.0f);
	}
		DTerm = constrainf(delta * pidProfile->D_f[axis] * PIDweight[axis] / 100, -300.0f, 300.0f);

In essence both P and D get equally filtered on the first pass.  

Boris's code is a simple first order filter with a 3db/octave rolloff; by applying it to gyroRate it just smooths the input to both the P and D calculations.

From Levidnap's spectral analyses it's obvious there is a LOT of high frequency noise about.  

If both gyro_lpf and the second software gyroRate filter are set at about the same value, we get 6db/octave filtering rather than 3db/octave for this noise.  Boris's D term filter is still needed to correct the inherent 'amplification' tendency in D term calculations as frequency increases.  That term should, I think, be set just above the undamped natural oscillation frequency for a given frame.  

With the code above a noisy quad can fly quite nicely with bent props on gyro_lpf 98, gyroRate LPF say 75 or 100, and the Dterm LPF at say 17-20.  

Resolving the best values for a given quad would be easiest if we had per-profile values in the CLI to set this software gyro_Rate LPF and separately to set the deltaD LPF.  

As mentioned previously I think it would be a huge advantage to also be able to set gyro_lpf on a per-profile, rather than global, basis.

I've finally done some fine tuning using the BlackBox - I am so grateful for that, it's amazingly helpful.  I made three separate profiles with same gyro_lpf of 100 and 50, 75 and 100Hz secondary filters.  Interestingly, PID's when optimally tuned were about the same.  The 75Hz variant seemed closest to what I want.  All relatively easy to tune.  After 5 days of rain I think the sun is out tomorrow so I can test it out properly.

I had a quick look at using 'error' rather than 'rate' in Lux as the input to the D delta calculation.  Rewrite uses 'error' as the input to D, which is why rewrite subtracts, rather than adds D to total PID.  Briefly I think this is promising, it seems likely to be more responsive but some special filtering on D is needed when error is used as the input and there are step changes in set-point, like we have on CPPM steps.

borisbstyle, 2015-06-20:
@ctzsnooze
I think that isn't bad idea at all. I was initially thinking you were filtering gyro and than adding additional filter to D. But on this way both are pretty much in phase.

borisbstyle, 2015-06-20:
@ctzsnooze
I will test that and compare if it has any negative effects and see to find a nice way for implementing it for all pid controllers.
Maybe we coule just filter the Ptejrm

pulsar256, 2015-06-20:
@ctzsnooze thanks for explaining that in detail!

borisbstyle, 2015-06-20:
@ctzsnooze Your method is indeed pretty good. Check my local branch here.
https://github.com/borisbstyle/cleanflight/tree/local
Here I am reusing the same function to do Pterm filtering as weill with pterm_cut_hz command. I used 50hz for Pterm and 14hz for Dterm. I disabled the MPU6050 filter completaly. This truelly gave me very smooth graphs, sound of the motors and very locked in quad!

ctzsnooze, 2015-06-21:
That's great news!

I have indeed been smoothing the incoming gyro data, which then makes both Pterm and Dterm smoother, but smoothing the Pterm itself, as we have for Dterm, and leaving the incoming gyro data relatively untouched, may indeed be a neater solution.

It's worth noting that it isn't possible to entirely disable filtering in the mpu 6050.  It's always there.  Sending DLPF_CFG = to the 6050 makes it sample gyros at 8kHz gyro and filter at 256Hz.  We cannot escape the 256Hz filtering, as far as I know.  In a sense that's good, anything that high will be useless noise that we want to block anyway (levidnap and blackbox shows this). BTW I see you fixed sending 0 when gyro_lpf is 256 in your code using enum rather than if then else, very nice!  :-)

If you are filtering Pterm way down at 50Hz and 14Hz for Dterm, and it flies well, that's really good news. Using overall gyro filtering that low I found it wobbly and difficult to tune.  So that is very positive.  

Most likely with your current setup you could set a gyro_lpf to say 98Hz without any adverse sideffects, It would be interesting to know.  Test then with known out of balance props!  :-)  as there may be even better noise rejection (your frames might not need it normally).  

If it's OK I'll use your code base as my testing point from now on.  I may add a software overall gyro filter also just for testing on very noisy frames.  Having a vert hard cut on high frequencies that we just aren't interested in can be helpful on very noisy frames, as my previous tests showed.  Being well above your 50Hz Pterm filter should this should cause no adverse effects at all.

PS - your coding skills are better than mine - I can't quite follow how it actually gets the filtering done on both Pterm and Dterm at the same time, but I take your word for it!  :-)  

PSS - I noticed that Dterm needs very heavy filtering when controlling error rather than rate, because each PPM step up has 'infinite' Dterm for an instant.  This opposes P at that moment and needs managing somehow.  I think that's why originally there was the 3 point moving average on D, it would be an effective way of fixing it in rewrite and multiwii, which targeted error.  It's Lux that targets gyro rate, not error, and I think fundamentally that causes the problems with frame noise in luxfloat particularly. It will be easier to resolve this with your profile based P and D term filters.  My gut feeling is that getting Dterm to target error signal in LuxFloat will probably result in even crisper responsiveness and that generally it's the best way to go for all controllers.  I plan to remove PIDC 4 and replace it with an error seeking Luxfloat, just for flight testing purposes.

I think we are really getting somewhere now, and we have the tools to move forward quickly.  I'm amazed and impressed by what we are achieving.  Thanks for all your help!  :-)

borisbstyle, 2015-06-21:
Yeah you should prevent filtering on filtering and again filtering on that.
That's probably why you had wobble. Not because of filtering, but phase shifts. Thats why I didn't agree about that approach
On this way you can separate the things with least amount of filtering and keep things in phase as much as possible.
You don't really need more than 50hz on multirotors. I was using 188 and higher to prevent delay.
Yeah I know about 256hz, but for multirotor terms that is no filtering at all actually :)

Yeah trying to keep less same returning code is the way. I think it is even better to make a struct, which can dynamically be allocated in the memory with pointers each time so you can introduce as much as possible filters when needed.
But in here we only need p and d filtering.

If you look my posted log my acc data is very noisy, but my level was still excellent as the pterm is the same :).
The more and more I look into this I think this is the way to go.

ctzsnooze, 2015-06-22:
THIS IS INSANELY GREAT!!!!    Definitely, definitely the way to go.  

Using your local branch code, my previously outrageously wobbly little quad is now - cliche aside - as 'locked in' as anything could be.  

Loop times are a little higher I noticed but not by heaps.  Being able to separately filter Pterm and Dterm (and on my mod gyroRate as well) makes it really easy to figure out what works and what doesn't.  And it seems to work really, really well.

I agree about 50Hz - its about how fast each ppm pulse comes in, and human beings have trouble reacting faster.  I've noticed that no matter how abrupt I try to make my stick inputs, the rise time of my best 'step' input shows up as being spread over, say, 4 PPM pulses in blackbox, i.e. 12Hz...  :-)

And yes I agree entirely, it doesn't matter to me if the frame is vibrating, what matters most is that the motors aren't amplifying it getting their output reduced.  Jittery signals to motors mean lost power, short flights and hot motors.  I'd trade off a fraction less responsiveness to avoid that kind of stuff.  But now we can have both.  

My 'noisy' frame has obvious vibration when testing using the motors tab sliders - due to motor/prop imbalances, frame resonances, 'beating' of similar speed motors, and probably ESC induced pulse timing variations - but the key thing is that all this vibration has *nothing to do with PIDs*.  It is inevitable.  We just have to stop the FC responding and reacting to it.

These new filtering options allow nearly totally smooth motor control pulses despite frame vibrations.  Stock CF code does not.  It's a blast to now have full power, knowing that frame vibrations are not 'stealing' motor control!

Back to coding matters....

I've taken out the 3 point smoothing don't think it makes any difference.  Do you still think it does?

I agree that it's best to run gyro_lpf at 256.  At 256, there is 1-2 ms less 6050/sampling delay than any other 6050 setting, and cleaner data (less likely to be aliased) into the FC.

I don't agree about 'filters on top of filters' thing, well not entirely. The kind of filters we are using do not cause delays or phase shift below about half the value they are set to.  True, they add a 45 degree phase shift i.e. about 1/4 cycle delay at the nominal frequency, but say if you set one at 100Hz, stuff at 50Hz basically walks through almost entirely unchanged. It's true that stuff at 200Hz would be delayed and have a phase shift, but it will be cut to 1/4 of its normal value and should start becoming irrelevant.

Putting a sw gyroRate filter at say 128Hz will do nothing to the control systems that we want to work nicely at 50Hz and below, but will take out a lot of higher frequency noise, and from Levidnap's spectrograms, there's a lot of it up there that needs to go.

I've tested your Pterm + Dterm code, with my optional sw gyroRate filter. It's really great to set each parameter individually on a per-profile basis, which is absolutely great.  

I'm currently playing with, as my 'performance' profile:

    set dterm_cut_hz = 32
    set pterm_cut_hz = 64
    set gyro_cut_hz = 128

This is very responsive and easy to tune with some D term noise. I am trying out the following settings in the two other profiles, just to see how they go:

    set dterm_cut_hz = 16
    set pterm_cut_hz = 32
    set gyro_cut_hz = 64
    
   set dterm_cut_hz = 25
   set pterm_cut_hz = 50
   set gyro_cut_hz = 100

In all of them motor power delivery is essentially linear with only a little jitter.  I am running high TPA at 0.5.  

Personally I think we need all three filtering options available; a gyro_cut is very helpful.

Cheers = C

PS this is what I did for the gyroRate LPF:

    static float    RCGYRO;
	RCGYRO = 1.0f / ( 2.0f * (float)M_PI * pidProfile->gyro_cut_hz );

and

    // gyroRate low pass CT
        if (pidProfile->gyro_cut_hz) {
        	gyroRate = lastGyroRate[axis] + dT / (RCGYRO + dT) * (gyroRate - lastGyroRate[axis]);
        }

with the appropriate changes in config to support gyro_cut_hz

borisbstyle, 2015-06-22:
Yes the more and more I look into the way the current code works there is definately need for splitting up the filtering.
The jittered output to the motors has to be minimized. It can be a huge loss in speed and efficiency.

I see you are filtering multiplied gyro rate actually instead of gyro? That means that P value has double filtering. Is this really needed? 

This is the part where you should filter gyro to get it through the rest of the code and properly show it in blackbox
https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/sensors/gyro.c#L122

This is the part to separately filter acc. Disabling onboard gyro lpf also means acc filter is disabled. MPU9050 supports for example separate configuration for acc and gyro filtering.
https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/sensors/acceleration.c#L174

The moving average should be in there to not impact the people having these options disabled. So default behaviour should stay the same. But should definately test without when applying filters to gyro.

But anyway. No need to introduce  new RCGYRO. You could simply reuse the filter function again :D. RC is a temporarily value and no need to have multiple of those.

It's too bad your are still only testing luxfloat. In my tests where I went through PID controller 0-5 luxfloat was the winner in noise amplifying. I think that's a fundamental problem in this pid controller as it doesn't use much internal filtering as floating point precision makes it more sensitive to noise than integer. See for example harakiri what also uses floating point math, but has a lof of code around to prevent amplifying noise.

It seems like your are looking your personal best settings for your specific setup. But my main goals in here are:
- Improve Filtering in the current Cleanflight code (or better said add one as there is no filtering at all)
- The filtering should not introduce much delay. Preferably 1st order filter
- Make this code flexible to reuse on all pid controllers on same/simillar way
- The new code should not impact the "default" behaviour as it is now
- Even users with clean setups should make profit and not lose responsitivity

Options so far if you ask me:
- Only Dterm filter (pretty much proven and definately improvement to get into master) see https://github.com/cleanflight/cleanflight/pull/1028
- Additional Pterm filter. The way to have least amount of filters. Seperate filtering for D and P with different cutoff frequencies. (experimental yet, but seems to be an improvement)
- Use software filtering for gyro input. I think once this used there should not be any Pterm filtering, but perhaps only D. The soft filtering should give better results than the MPU6050 internal filter in this one.

Also for other filtering we have to check different filtering methods instead of only this method used now, which by the way seems to do a great job.

borisbstyle, 2015-06-22:
BTW I will reconsider the option RateError filtering. Just because it is being used for delta as well, but will first try to filter gyro input and see how it goes.

borisbstyle, 2015-06-22:
@ctzsnooze 
Did you by the way also try gyro_lpf of 188 with pterm and dterm filter only? I think that will remove the need of your rateError filter. I was looking through my logs and goint from 256 to 188 makes a lot of difference. Also some others reported that as well on noisy frames.
On that way you still keep the same lower delay on D and you add some additional filtering on P.

ctzsnooze, 2015-06-22:
Hi Boris - much to discuss...  many options now and maybe we need time to think about them.

First, what I mean by D term 'targeting error'.  P and I always target error - the Pterm is, as you know, directly derived from the difference between the gyro rate and the rate requested by the sticks.  But D can either target the rate of change of the error (which is how it's done in rewrite) or the rate of change of the actual controlled value (which is how it is done in Luxfloat).  

    delta = gyroRate - lastGyroRate[axis];  //lux

vs

    delta = RateError - lastError[axis];    //rewrite

The two approaches are fundamentally different, which is why the PIDsum Dterm must be negative in Lux and positive in rewrite.  The rewrite approach seems on my initial testing to result in a faster return and less wobble, but I need to do more work on this.  I've just replaced PIDC4 with an error-targeting Lux variant so I can more easily compare them in the air just by changing profiles.

Today I tested code with Pterm, Dterm and gyroRate LPF's.  I flew many batteries with the settings I mentioned in my previous post, by far the most 'solid' were the higher filter values with a higher P value.  Even so, the motors were solid, strong and smooth.  The tuning wasn't perfect, bit of wobble in prop wash, but I had a heap of fun and it flew really well, very crisp - and noise wasn't an issue - pilot skills were!  :-)

I agree that the outcome of this should be a general solution for all controllers.  I am working mostly on Lux because, being the worst, it means that I may identify why it is so, and learn from that. 

I agree also that separate control over Pterm and Dterm filtering is the biggest step forward.  

One advantage of smoothing Pterm is that because Pterm targets error, there is always a sharp step up in Pterm whenever the control PPM pulse goes up a step.  At that instant, the Pterm shoots up in a straight line, and that goes direct to the motors.  It probably gets aliased by the ESC PPM rates as well.  Smoothing Pterm makes those sharp steps rounder - it's easy to see in blackbox.  This can only happen when you smooth Pterm at this point.  So I conceptually really like Pterm smoothing and it is very easy to see what it does in blackbox.  This is great.

The same feature would be of great benefit for PIDCs with Dterm calculations that target error, because Dterm in particular gets huge spikes when it targets error.  Heavy Dterm filtering for error-targeting controllers will smooth that out - needs to be down around 15-17Hz though and to be honest I've not been that low in testing to date.

Now as to the question of whether, in addition to Dterm and Pterm LPFs... should there be a software gyro LPF with 6050/gyro_LPF fixed at 256, or should we only use the 6050 LPF alone for gyro filtering.

For a number of reasons I think a separate gyro LPF in sw is useful.  I run through them here:

First, changing 6050/gyro_rate LPF's change what we see in the blackbox and the motors tab.  Whenever we drop the gyro_rate value, the incoming data is itself more heavily filtered, not just gyro but also acc, making the 'appearance' of less noise even if the same amount is actually present.  That's why gyro_lpf of 256 is always going to 'appear' worse than 188, and 188 appears worse than 98, and 98 appears way worse than 42 - even if the underlying noise in the frame is exactly the same.  

The only valid way to compare outcome when changing 6050 filtering is to look at the actual wobble in the PID responses, or the amount of noise-derived jitter in the motor signal.  Looking at noise in gyro traces or acc traces is no help in deciding if a change in gyro_lpf is positive or not.

In contrast, if we do the gyro filtering in sw, and keep the data acquisition filtering constant at 256, the noisein the incoming gyro data will truly reflect what is happening in the frame, and be correctly shown in the blackbox or that acc display.  So we get to see the 'true' benefit (or otherwise) of the top end gyro LPF, if we enable one.  This makes it much easier to see the true effect of overall gyro LPF filtering. 

Maybe you might consider putting an LPF on gyroRate, as I have done, again making it so that if 0 it is disabled.  So it's easy to add or remove it.  Then for testing, compare:

-  6050 gyro_rate 256 (i.e. 'off'), sw gyroRate filter to 188 or 98, vs
-  6050 gyro_rate 188 or 98, with sw gyroRate filter off.

Of the two, my experience is that it is better to have the first setup; smoother motors and less high frequency PID noise, and no negative issues that I could detect.  For reasons outlined previously the recordings of the gyro or acc noise levels in blackbox cannot be meaningfully compared between the two, one must compare the amount of noise in the total PID or the amount of noise induced jitter signal in the motor output.

The other advantage of a profile-adjustable software overall gyro filter is that we can set any cut frequency we like, not just 98 or 188, and do that per-profile, which greatly simplifies testing.

And finally sticking with 256 in the 6050 lets us get data on average 1.5-2ms earlier than 188, that helps a bit with latency.

I guess I'm now more clearly than ever of the view that gyro_lpf should always be 256 and that if there is ever a need for overall gyro filtering, it should be done in software on a per-profile basis.

[... cut ...] or testing, and for adjusting to suit a given frame.  Best not hard coded imho and better I think not directed at gyro_ADC.

Finally, the 3point moving average filter could always be retained for default functionality as a conditional... e.g.:

	delta /= dT;
        if (pidProfile->dterm_cut_hz) {
        	delta = lastDTerm[axis] + dT / (RCDTERM + dT) * (delta - lastDTerm[axis]);
    		lastDTerm[axis] = delta;
       else {
        	// default three point moving average code
       }

Cheers - C

borisbstyle, 2015-06-22:
Your indication of having wobble and prop wash is exactly the indication that there is too much delay involved in the total path. Exactly what I was experiencing in my previous filter tests with different kind of approaches. Your final verdict should really be done based on final tuning point.
My experience is that once you get this elastic kind of wobble there is no way to tune it out properly.

That's why I also think your apprach of filtering on filtering isn't the right one. P and D shift much more in phase. But maybe we should do some graphical syntesizing of different appoaches.

The need of moving average is a tough one to understand for me, but without that one I couldn't get rid of the bumps on low looptimes. I guess it's looptime dependant. As I run 1200ms looptime it seems that moving average was necessary. I might do more testing with higher looptimes. Just guessing here.
But removing it would require a lot of testing and comparing. And that takes a lot of time :(

Why didn't you try something like this on your noisy setup without using of your gyrorate filter? 
pterm = 40hz (or even lower)
dterm = 12hz
gyro_lpf = 256 or 188

You can indeed move gyro_lpf to the profile settings indeed. I believe it's not that trivial from what I have seen. 
Note that average user doesn't understand these settings :D

henn1001, 2015-06-22:
Hey all,
i´m still trying to wrap my head around the following thing...
As im trying out differend combinations of these filters too, i figured that we dont want to see these kind of motor jitters: http://static.rcgroups.net/forums/attachments/5/2/4/8/5/4/a7989949-230-4.JPG

But what about these two:
[image]
Slow wobbly kinda smooth lines vs overall constant lines with high frequency jitter on it? What do we consider better for our motors?

ledvinap, 2015-06-22:
I did another test - using small coreless motor with eccentric weight attached and glued to Sparky PCB as signal generator. Voltage is increased every 3 seconds, from 1V to 5V, in 0.1V steps.
Here are spectrograms for 256Hz and 188Hz MPU6050 setting:

gyro_lpf = 256:
[image]

gyro_lpf = 188:
[image]

IMO it is obvious that using gyro_lpf = 256 with data readout at 1kHz makes no sense - signal is only linearly attenuated (peak acceleration from motor rises with frequency), there is lot's of noise in band that affects flight characteristics. These data are obtained with true 1kHz readout (using FIFO), with one read per looptime the aliasing would be even worse (I'll try to simulate that).

It is reasonable to assume that MPU6050 has two internal LPFs - one with 256Hz bandwidth and second (LPF2) with user selected bandwidth. Enabling LPF2 changes MPU6050 response to second order above 256Hz, the cost is 0.9ms delay.

Also there is not much difference between 188 and 256 at input frequency around 1kHz, it is highly unlikely that internal sampling frequency is changed..

ledvinap, 2015-06-22:
@ctzsnooze : 
Filtering is linear operation - filtering Gyro and RC data separately and than subtracting them should be equivalent to subtracting them and then filtering. So you may try filtering RC data separately (and with different cutoff) from gyro data.
It may also make sense to implement rate limiting filter on RC data - limit slope of RC inputs to value that PID can follow. Tuning will be a bit more complicated, but results may be worth it ... 

As for LPF2 filter - data at 256Hz does not appear worse - they are worse (aliasing) and no filtering can fix that (See the spectrograms. You can use LPF and cut everything above some frequency - Y axis. But you can't separate noise and signal in band you need for control).

Gyro filter in CF still makes sense, even with MPU6050 filtering. Using two filters in series will improve rolloff above cufoff frequency, so high attenuation far above cutoff will improve.

And delay is intrinsic for any causal filter, you can't avoid it. You can just use it differently - for example use second order filter with double bandwidth and same delay as first order one.

borisbstyle, 2015-06-22:
That is exactly what I was assuming. What about spracingf3 where we could get faster looptimes up to 2khz? We could read gyro more often.

Anyway I am going to give a FIR filter one more try for gyro input and do some more tuning based on my knowledge now.

What do you suggest for best approach in trying to eliminate as many noise as possible?

My experience so far is the best with pterm and dterm filtering on 188hz gyro

I might add fir uppon that :)

ledvinap, 2015-06-22:
@borisbstyle : Gyro readout is I2C speed limited, so faster readout is feasible only over SPI ...

ledvinap, 2015-06-22:
@borisbstyle : FIR is problematic - delay will be half the number of taps, so you can't use reasonable filter length. IIR filter (and especially Butterworth) is much better with respect to rolloff/delay performance.

borisbstyle, 2015-06-22:
ok I just did some simulations with fir and indeed it is slow.

Maybe current gyro_lpf, pterm, dterm approach is the best actually.
Or do you have a good coefficient suggestions for butterworth?

borisbstyle, 2015-06-22:
I know you were busy with FIFO as well. can you point me out what changes are required to get that going?

ledvinap, 2015-06-22:
As for the 3-sample averaging : here are responses when combined with 1st order Butterworth at 50Hz (1kHz samplerate):
Delta only:
[image]
Averaging:
[image]

It is quite possible that FIR helps to attenuate frequencies around 300Hz - motor noise is in this band ...

ledvinap, 2015-06-22:
@borisbstyle : 

FIFO: it's just quick hack to test it: https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/accgyro_mpu6050.c#L451 (grep/compare/check nearby commits for rest of code)

Butterworth:
https://github.com/cleanflight/cleanflight/pull/1024#issuecomment-113098221 (just change sign for A2 and A3)

[40 more replies]

closed 2015-09-09 by borisbstyle