cleanflight/cleanflight #1015: quadcopter: low-frequency vibration on declining

kaefert, 2015-06-15:
I've built a 250-mini-quadcopter with a naze32-full board running cleanflight controlling it.

On declining, I get a low frequency vibration. I've experienced the same on my previous micro-quads the Hubsan X4 quadcopters. Is this a unavoidable problem with quadcopters, or is this a PID-tuning issue?

If you don't know what I mean, see this video from yesterday at the linked times:
https://youtu.be/6r5tBHGABUE?t=15
https://youtu.be/6r5tBHGABUE?t=74
https://youtu.be/6r5tBHGABUE?t=194

Arakon, 2015-06-15:
You are descending into the propwash you just created with your copter. Either fly into a direction while descending, or try to adjust it with PIDs, but you can't get rid of the turbulences that are there.

kaefert, 2015-06-15:
oh! that makes sense. thanks for the explanation!

closed 2015-06-15 by kaefert