cleanflight/cleanflight #1017: Baro readout seems broken

yGO77, 2015-06-16:
Tested with 1.8.1 all is working fine.
With 1.9.0 the barometer outputs looks wrong to me as it only show variation between -15 cm and 10 cm.

SteveAmor, 2015-06-16:
I too see this issue but didn't raise it yet as it's on a dev board. In V1.9.0 the baro graph appears to show rate of change of pressure. Move the baro quickly and it changes height an the graph but quickly converges back to zero. 

V1.9.0 now has some filtering on the baro which would explain narrower variation at rest. 

@yGO77 do you see stable changes in height?

yGO77, 2015-06-16:
No, if I put my system 2 m from the initial startup altitude it's showing
-1 cm.

Move it back to initial pos, it goes 0 and then back to -1 cm...

SteveAmor, 2015-06-16:
@yGO77 what board do you have?  As I say, mine is a dev board (Naze acro with added I2C baro) so I wasn't sure if it was just me.   But I see exactly what you see in 1.8.1 and 1.9.0.
Have you flown 1.9.0 with the baro?  is it just the graph or do you think the altitude code is also broken?

yGO77, 2015-06-16:
It's on the ImmersionRC Vortex.  I'm pretty sure the altitude code is
broken.

digitalentity, 2015-06-16:
That's strange, but I can't confirm the issue, baro works flawlessly on my dev builds. In fact I am using baro to develop new features, and I would have noticed that readings are wrong. Will try to confirm the issue tomorrow on my dev board using official 1.9.0 build.

SteveAmor, 2015-06-16:
I'm seeing this V1.9.0 graph issue (and no impact of baro/altitude mode in flight) because I have connected an I2C baro (BMP180 to a Naze Acro) without the EOC interrupt connection (also without XCLR).  I disabled the following in target.h and now the graph displays as expected.   

//#define BARO_XCLR_GPIO   GPIOC
//#define BARO_XCLR_PIN    Pin_13
//#define BARO_EOC_GPIO    GPIOC
//#define BARO_EOC_PIN     Pin_14
//#define BARO_APB2_PERIPHERALS RCC_APB2Periph_GPIOC

Perhaps the ImmersionRC Vortex fc only has I2C also and no EOC interrupt connection?

yGO77, 2015-06-16:
The Vortex simply queries cleanflight for altitude using MSP.

borisbstyle, 2015-06-16:
Something to do with this one?
https://github.com/cleanflight/cleanflight/pull/718

SteveAmor, 2015-06-16:
My problem was that the CF firmware for a Naze target assumes you have End of Conversion (EOC) connected up.  I just connected a cheap baro with the two I2C connections (plus power) to an Acro Naze and as a result, the FC waits for an EOC interrupt but never gets one.  I removed the use of baro interrups in the code and it now works as expected (i.e. the flight control loop now waits a given amount of time for a conversion to take place and correctly reports differences in pressure). 
However, tuning the PIDs is an entirely different ball game...... :-)

SteveAmor, 2015-06-16:
@yGO77 I tried to find your FC but couldn't.  What CF firnware do you use (Naze)?

yGO77, 2015-06-17:
@SteveAmor, yes we use Naze.

skaman82, 2015-06-18:
Same issue here. I have a cheap BMP180 board connected to the SDL SDA pads underneath the naze32 acro board (and 3v-pad + ground). The baro is found automatically but I'm seeing no changes to the baro graph unless I move the board very fast up and down. Then the graph goes down to almost zero automatically. Same as described by "SteveAmor"

SteveAmor, 2015-06-18:
@skaman82 the default code won't work with an I2C baro connected to an acro boasrd. You need to modify the code to disable EOC interrupt. 

@ledvinap - I know you're the interrupt expert. I was thinking I could test for EOC but if it's not connected, it will probably float and I'll get a false positive?

skaman82, 2015-06-18:
@SteveAmor but in 1.8.1 it seems to work

ledvinap, 2015-06-18:
@SteveAmor : I just can read the datasheet ;-)

You can use GPIO pullup/pulldown to get floating pin in defined state. Then you can start timer on conversion start and handle situation when EOC is not asserted in expected time. This should be implemented even when EOC is connected - if some unexpected error occur, conversion should be restarted.

ledvinap, 2015-06-18:
@skaman82: EOC handling is new feature ...

SteveAmor, 2015-06-18:
@ledvinap lol :). Good point. Would you recommend GPIO pullup(/down) is disabled if there is an on board baro? Or always on?

skaman82, 2015-06-18:
@SteveAmor @hydra ok, thanks. My preferred setup on almost all my quads is an external compass (combined with gps) for precise readings. Can't realize that on the naze32 full board unless I cut the SDL and SDA traces to the on board mag :( So whats left is the acro version. But GPS hold and RTH makes no sense unless you have some height measurement (IMO). I'm not a developer an have no clue how to modify the code and I do not want so cut traces on my expensive naze32 full board. All I can do, ist to give you guys some suggestions how to make users like me very happy (again). :) It would be great to make such setup like mine work again! Maybe in the next release the is a CLI command that could make the external baro work again an deactivate EOC? (no clue what that is). Thanks

SteveAmor, 2015-06-19:
I've worked out the code change required to make detection automatic and will do a Pull Request in the next few days.

yGO77, 2015-06-19:
@SteveAmor Many thanks for the fix!

skaman82, 2015-06-19:
@SteveAmor Thanks!

SteveAmor, 2015-06-20:
I believe this is now fixed in PR #1033 
I have tested on an acro Naze but can only simulate the operation on a Full Naze to make sure Full Naze users aren't affected by this code change.

stronnag, 2015-06-20:
I don't think this is fixed for BMP180 in #1033, see comment therein

SteveAmor, 2015-06-20:
I have an Acro Naze with a bmp180 connected - and working. Will check comments in PR #1033

skaman82, 2015-06-25:
Is this already merged? Just tried the following Build: 
http://cleanflight.memoryleaks.org/builds/cleanflight_cleanflight/2015-06-25/ 
I have the acro naze (Board: AFNA, version: 2) and the GY-68-BMP180 connected (SDA/SDL + 3v). 

Status says: 
System Uptime: 43 seconds, Voltage: 0 * 0.1V (1S battery)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP085, MAG=HMC5883
Cycle Time: 3548, I2C Errors: 0, config size: 1724

I still can not see any changes for altitude in the sensor tab of the GUI unless I move the board very fast.

SteveAmor, 2015-06-25:
@skaman82 PR #1033 has not been merged yet. That will fix your issue. There is a link in PR #1033 for a build that includes the code you need to fix this.

skaman82, 2015-06-25:
@SteveAmor Thanks! Just tested this. It works now for me :)

SteveAmor, 2015-06-25:
@skaman82 can you comment on PR #1033 please :-)

adtinnis, 2015-08-19:
I am not sure if this is related to this problem.  I have connected a bmp180 barometer over I2C.  When I have this connected to the Cleanflight Configurator, it seems to work fine.  When I lift up the quad, the barometer/altitude reading on the 'Sensors' tab goes up, by what seems the right amount.  However, when I attached the props and take off, the altitude goes negative.  As a test, I covered the barometer with electrical tape completely to prevent any prop wash from getting in there, but this seems to have little effect.

I have made the changes to the code base shown above and built a version of Cleanflight that seems to work in all other ways.

Any ideas?

closed 2017-02-25 by hydra