cleanflight/cleanflight #1019: small YAW movement erratic when compass on

arachnida, 2015-06-16:
When using Harakiri PID and MAG is on small movements of YAW are not held, like there is some deadband active. If the YAW stick is pushed a bit further the YAW movement jumps and this new heading is stored in magHold. YAW_Deadband=0 in my test. When the jump occurs I can see in the logs that rcCommand(YAW) jumps from a low value like 30 to exact 70. (see pics) This value was found by Boris in line 437 of mw.c acting as some kind of filter.

436     if (ABS(rcCommand[YAW]) < 70 && FLIGHT_MODE(MAG_MODE)) { 
437         int16_t dif = heading - magHold; 
438         if (dif <= -180) 
439             dif += 360; 
440         if (dif >= +180) 
441             dif -= 360; 
442         dif *= -masterConfig.yaw_control_direction; 
443         if (STATE(SMALL_ANGLE)) 
444             rcCommand[YAW] -= dif * currentProfile->pidProfile.P8[PIDMAG] / 30;    // 18 deg 
445     } else 
446         magHold = heading; 

This erratic behaviour can best be seen when you have a high YAW rate. When wanting to do a small YAW movement the original heading is returned when I release the YAW stick.

[image]
[image]
Also see https://www.youtube.com/watch?v=hn6pF2bxMMg at 0:04 and 0:24, at 0:17 I didn't push my stick so far and the copter turns it's nose back to the original heading when I center my stick.

arachnida, 2015-06-16:
Boris compiled me version where he substituted the value 70 in line 436 by 15, this seems to be a good solution as the erratic jumping is gone. But why is this line in the code as we also have YAW_deadband?

borisbstyle, 2015-06-16:
Yes this is the version you have now. I couldn't think of reason to keep the deadband that wide till rccommand 70, but perhaps there is a good reason for that?
https://github.com/borisbstyle/cleanflight/tree/mag_hold_dead

hydra, 2015-06-17:
Gotta love the blackbox logging by @thenickdude :smile:

borisbstyle, 2015-06-18:
@hydra
yes it makes it so much easier to analyze these things :)
I immediatelly spotted the part of code doing this.
But what do you think about this mag hold? To me it totally doesn't sound right to modify the actual rcCommand. And why is it using rcCommand for the in this anyway? So many different options.

arachnida, 2015-06-26:
Did anyone notice this issue with other PID controllers?

borisbstyle, 2015-06-29:
@arachnida 
Can you confirm that my fix gave a good solution for this problem? 
I don't think that there are many people using this. Perhaps you could test it with other PID controllers as well to make it a good PR suggestion? I expect the behaviour to be the same on all pid controllers.

arachnida, 2015-06-29:
@borisbstyle :
I've only made two flights with your version but it seemed that the problem was solved. I wanted to see this confirmed in the log too but I forgot to enable the blackbox feature, stupid me. For the moment I'm grounded as I am having lots of problems suffering from hay fever and can't fly. Still remains the question why these program lines are there, maybe a solution to solve a pre historic bug in the old baseflight?

arachnida, 2015-07-01:
@borisbstyle :
Yesterday I was able to make a more serious test including BB, there were no problems at all during slow YAW with compass enabled. In the BB graps I can also not see any jumps in the lines, not even a small one. Did you really change the value from 70 to 15 or did you set it at 0?

So for me this looks like a good PR suggestion, although I didn't test any other PID controllers than Harakiri.

borisbstyle, 2015-07-01:
@arachnida 
Yes test a few other pid controllers as well and I will make a PR. Yes I changed it to 15. 15 is not very much at all.
See this branch you are using now:
https://github.com/borisbstyle/cleanflight/commit/f03c3dcb94db9cea9362fe8308acea8c6c08846f

I only don't know why the 70 was there. It might be the part of how this function is designed. Maybe it was done on purpose when using mag hold to be able to make small movements, but always get back into the original heading, but on the other hand it doesn't make sense at all.

borisbstyle, 2015-07-01:
I made a PR anyway to see if it catches more attention from others

Mechatricity, 2015-07-01:
This definitely occurs on PID3.  I can apply a small amount of yaw, but it will drift back to original heading.  If i break the deadband it will induce a snap yaw movement.

arachnida, 2015-07-01:
Thanks Boris. Strange that other pilots didn't notice this behaviour, but then I guess most devs have acro boards.

arachnida, 2015-07-01:
Oops, Mechatricity jumped inbetween. At least one more pilot having this issue, I'm not dreaming.

borisbstyle, 2015-07-01:
@Mechatricity 
Can you compile the PR and retest if it's acceptable to you as well? I can also compile it for you if you want?

Mechatricity, 2015-07-01:
If you could compile that would be stellar.  I may not have a chance to test until Fri but will report back ASAP.

borisbstyle, 2015-07-01:
@Mechatricity 
I assume you use naze board
https://dl.dropboxusercontent.com/u/31537757/cleanflight/cleanflight_NAZE_mag_deadband15.hex

Mechatricity, 2015-07-01:
I do indeed, thanks Boris.  Will retest.

arachnida, 2015-07-01:
Please mind that flashing this version will erase your settings, even if you already were on 1.9.0.

hydra, 2015-08-12:
It makes me wonder if this should be configurable or not.

arachnida, 2015-08-12:
Hard to tell as I don't understand why these lines are there in the first place. The new value by Boris works for my setup, maybe not for others. We should need more testers...

closed 2017-02-25 by hydra