cleanflight/cleanflight #1022: PPM failure to AUX channels in Orange LRS

eslanek, 2015-06-17:
He! I have a problem with the latest version of CF. I'm using a Naze32 Acro a receiver OrangeLRS by PPM whit openlrsng ppm on ch5 . The problem is that with the latest version CF 1.9 the Aux channels 1-2-3-4 go crazy and meaningless move . However this does not happen with 1.8.1 . Can anybody help me.

henn1001, 2015-06-17:
IMPORTANT: ** Verify your RX and failsafe is configured after upgrading, check your channel endpoints when your rx is operating normally and when in failsafe mode, see rx_min_usec and rx_max_usec **

Check your endpoints...

eslanek, 2015-06-17:
I'm going to see and comment on the result later , thanks

eslanek, 2015-06-18:
Exactly, that was the problem . Mio error, create a new profile for a new model in the transmitter yy default channels were 950-1500-2000 , some even more . Ajustadoa 1050-1500-1950 everything works right , thanks

closed 2017-02-25 by hydra