cleanflight/cleanflight #1025: GPS enable causes OSD overlay to disapear

Buzzliteyear, 2015-06-18:
With 1.8.1 AirPlane mode and GPS enable the new (15x15mm sq) Micro MinimOSD overlay randomly disappears from say 30-90 seconds in flight  (cant use 1.9 as cant save servo direction tabs) flip32+ FC

OSD works fine for weeks until gps added

# dump

# version
Cleanflight/NAZE 1.8.1 Apr  2 2015 / 17:06:11 (988ae2d)
# dump master

# mixer
mixer AIRPLANE

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS

# map
map AETR1234

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set input_filtering_mode = 0
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 5
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_functions = 1
set serial_port_1_msp_baudrate = 5
set serial_port_1_telemetry_baudrate = 0
set serial_port_1_blackbox_baudrate = 5
set serial_port_1_gps_baudrate = 4
set serial_port_2_functions = 1
set serial_port_2_msp_baudrate = 5
set serial_port_2_telemetry_baudrate = 0
set serial_port_2_blackbox_baudrate = 5
set serial_port_2_gps_baudrate = 4
set serial_port_3_functions = 0
set serial_port_3_msp_baudrate = 5
set serial_port_3_telemetry_baudrate = 0
set serial_port_3_blackbox_baudrate = 5
set serial_port_3_gps_baudrate = 4
set serial_port_4_functions = 2
set serial_port_4_msp_baudrate = 5
set serial_port_4_telemetry_baudrate = 0
set serial_port_4_blackbox_baudrate = 5
set serial_port_4_gps_baudrate = 4
set reboot_character = 82
set gps_provider = 1
set gps_sbas_mode = 0
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 0
set spektrum_sat_bind = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set frsky_vfas_precision = 0
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 34
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set current_meter_type = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = -90
set align_board_pitch = 90
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set pid_at_min_throttle = 1
set acc_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 0

# dump profile

# profile
profile 0

# aux
aux 0 2 0 1175 2100
aux 1 0 0 900 900
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
aux 20 0 0 900 900
aux 21 0 0 900 900
aux 22 0 0 900 900
aux 23 0 0 900 900
aux 24 0 0 900 900
aux 25 0 0 900 900
aux 26 0 0 900 900
aux 27 0 0 900 900
aux 28 0 0 900 900
aux 29 0 0 900 900
aux 30 0 0 900 900
aux 31 0 0 900 900
aux 32 0 0 900 900
aux 33 0 0 900 900
aux 34 0 0 900 900
aux 35 0 0 900 900
aux 36 0 0 900 900
aux 37 0 0 900 900
aux 38 0 0 900 900
aux 39 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# servo
servo 0 1020 2000 1500 30 -1
servo 1 1020 2000 1500 30 -1
servo 2 1020 2000 1500 100 -1
servo 3 1020 2000 1500 100 -1
servo 4 1020 2000 1500 100 -1
servo 5 1020 2000 1500 -100 -1
servo 6 1020 2000 1500 -100 -1
servo 7 1020 2000 1500 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_direction = 1
set tri_unarmed_servo = 1
set servo_lowpass_freq = 400
set servo_lowpass_enable = 0
set default_rate_profile = 0
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_min_usec = 985
set failsafe_max_usec = 2115
set gimbal_flags = 1
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 1900
set pid_controller = 0
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  2.500
set i_pitchf =  0.600
set d_pitchf =  0.060
set p_rollf =  2.500
set i_rollf =  0.600
set d_rollf =  0.060
set p_yawf =  8.000
set i_yawf =  0.500
set d_yawf =  0.050
set level_horizon =  3.000
set level_angle =  5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

thenickdude, 2015-06-22:
To save people some time, decoding those serial port settings, you have:

```
UART1: MSP at 115200
UART2: MSP at 115200
SOFTSERIAL1: nothing
SOFTSERIAL2: GPS
```

hydra, 2015-07-08:
Likely an incompatibility with the firmware on your microminim osd.  Use scarab-osd firmware.

https://github.com/ShikOfTheRa/scarab-osd

Please test and report back if it still doesn't work.

closed 2017-02-25 by hydra