cleanflight/cleanflight #1027: Reporting two issues

rsappia, 2015-06-18:
Hi,

I am using a CC3D board. I was using with no problem version 1.8.1 and wanted to upgrade to 1.9.0. It wont work directly from Cleanflight configurator. The connection is lost as soon as I click on flash firmware.I had to download the .bin file and flash it through Openpilot.

The second issue that I found is that I was not able to restore my configurations from 1.8.1. I see that when I select the restore file, a message is displayed "Migrating configuration file generated by configurator: 0.65.0" and the packet error counter will start going high. Any suggestion or idea ? Am I missing something here?

I will be going back to the 1.8.1 until I can find a solution for the restoring issue...

Thanks in advance!!

skaman82, 2015-06-18:
I also had the second issue (restore from 1.8.1 backup file) on the naze32 board.

borisbstyle, 2015-06-19:
@rsappia 
You can alsu use cli dump as well, which worked for me. Just add the flight modes after that manually.

thenickdude, 2015-06-22:
Please don't report two issues in the same ticket, it makes it impossible to track their status independently.

The second of your issues sounds like it would fit better on the Configurator project:

https://github.com/cleanflight/cleanflight-configurator/issues

Redshifft, 2015-06-22:
@rsappia  Are you saying 1.8.1 would allow normal usb connections to the CC3D board ?

Flashed an Atom with 1.9 and constant "battery connect first then USB" when changing anything is a pain, otherwise board is non responsive......Unfamiliar with CC3D boards so thought it was the norm.

rsappia, 2015-06-22:
Hi @Redshifft , I am really not sure... what I finally did was to flash it directly from the Open pilot application.

@borisbstyle Thanks for the tip ;) It worked partially for me since it looks like there are some command line name that changed between 1.8.1 and 1.9.0. I copy paste the dump from my 1.8.1 config but I noticed that there were some commands that were rejected. I was lucky it didnt happen with most of the command. :)

@thenickdude My mistake, first time posting here. Is it possible to relocate my second question to the right thread?

Zero9r, 2015-06-22:
@Redshifft I have three CC3D's now and yes they all require to be booted with battery before USB.  Don't know why but they also need it with Openpilot as well after you got it programmed up.  its weird that it works without battery right out of box, but not after install.

What I do is leave a battery pluged in and if it needs to reboot. you just have to hit the disconnect button and unplug replug in your USB to the computer.  don't need to remove battery every time.  wonder if this is a bootloader issue or what.  but its a requirement for most CC3d users.

CopterFail, 2015-06-24:
If you want to get rid of this replug and battery using - flash the HEX file directly from the configurator using an usb-serial adapter. This will overwrite the openpilot bootloader, but there is no easy way back to openpilot.

closed 2017-02-25 by hydra