cleanflight/cleanflight #1030: CJMCU can't save modes settings

vorosj, 2015-06-19:
Hi,
board: CJMCU (Goodluckbuy)
receiver: Lemon Rc DSMX satellite on port2 (PA3)
Description: The mode settings are working while the board is powered, but not saved. Of course I pushed the save button. On the next power on the mode settins are empty. The modes settings even cleared if I choose save and reboot on another page. 
All functions (IMU, receiver) are working, and the other settings are properly saved.

vorosj, 2015-06-19:
I figured out that it depends on the receiver channel mapping. 
First I set TAER1234 and the modes are not saved, after I set it to AETR1234 the modes save is working. (My transmitter has free channel mapping.)
Hope that helps.

SteveAmor, 2015-06-20:
@vorosj please advise the version of firmware and Cleanflight configurator.

vorosj, 2015-06-20:
Running - OS: Windows, Chrome: 43.0.2357.124, Configurator: 0.65.0
Flight controller info, identifier: CLFL, version: 1.9.0
Running firmware released on: May 31 2015 13:17:03
Board: CJM1, version: 0
Unique device ID received - 0x66fff555254707067235945

Arakon, 2015-06-20:
I'm running the exact same version and configurator. DSM2 satellite, order is TEAR1234, settings save fine including modes.

vorosj, 2015-06-21:
Not easy then. 
I use a DSMX satellite, and the order was TAER1234 when the problem occured. All settings saved fine but the modes. Of course I am not sure if the order is related to the problem, I just told the story as accurate as possible. But I am 100% sure that the problem was there, and it occoured at least 10 times. It was there after I went back to the previous firmware version and back to the current version again. 
The problem gone when I modified the order to AETR1234.

Trussroads, 2015-06-21:
Just to clarify my own confusion, what board is being referenced when saying CJMCU? As this is basically a branding name and do make repro's of many common FC's as well as some unique hardware too.. Is it the brushed micro, or maybe the cc3d repro? Thanks

vorosj, 2015-06-21:
The brushed micro:
http://blog.oscarliang.net/build-fpv-micro-quadcopter-smallest-quad/

vorosj, 2015-06-21:
[image]

JohnieBraaf, 2015-07-07:
I'm having the same issue. Modes won't save in both TAER1234 or AETR1234 modes. 
CF1.9 Configurator 0.65

thenickdude, 2015-07-08:
Can you post a CLI dump? I'm curious if this problem is specific to profile >= 2.

JohnieBraaf, 2015-07-08:
CLI dump pasted here: https://github.com/cleanflight/cleanflight/issues/1094

JohnieBraaf, 2015-07-09:
The modes appear to be saving just fine, except when you reboot the board it loses the settings. It behaves as if it is being stored in RAM.

JohnieBraaf, 2015-07-09:
Somehow I don't have this issue any longer...

hydra, 2015-07-10:
I can't reproduce this. Tested using latest master with both default RX map and with TAER1234.  Tested on profiles 0,1 and 2.  Saved, unplugged, plugged, connecteded - new mode i just added was present.

Closing.  Please open a new PR and reference this one if the problem resurfaces or exact steps to repeat are found.

closed 2015-07-10 by hydra