cleanflight/cleanflight #1031: motors spin up on save and reboot

pkdevel, 2015-06-19:
whenever i click on 'save and reboot' or save via cli the motors spin up very shortly
im using 1.9.0 and oneshot. i see that there is only one initialization of oneshot now, maybe this is a regression.
if i can help providing more detailed information, please give me a hint

bssstudio, 2015-06-27:
Maybe it will be useful to know what kind of ESCs (ESC model and firmware) are you using.

I had similar issues with oneshot. The motors would twitch randomly and I ended up lowering the min command to about 900 or so.

hydra, 2015-06-27:
@pkdevel loads of work was put into this over the last couple of releases so im suprised youre having issues.  The double init was fixed too.

Need config dump and EXACT details of your escs and their settings before anything can be done.

borisbstyle, 2015-06-30:
Yeah very curious to how you can still have this issue

pkdevel, 2015-07-06:
hey guys,

sorry for the late response.
i'm using the ultra escs 20a with default settings, except that oneshot and active fw is turned on.
cant provide a dump right now, maybe tonight.

i have to say that i merged boris' "pt1 element" pr into the 1.9.0 version.
but i'm using the exact same version on multiple systems (naze and cc3d) without any issues,
except for two setups: one naze with those ultras and a default emax 250 setup with a cc3d.

pkdevel, 2015-07-06:
just found a dump on this machine, i think was the backup at this time, before i change my naze full to a naze acro:

# dump

# version
# Cleanflight/NAZE 1.9.0 Jun 15 2015 / 16:48:08 (98f7549)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature FAILSAFE
feature RSSI_ADC
feature LED_STRIP
feature ONESHOT125

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 3,15:S:AC:2
led 1 4,15:S:C:2
led 2 5,15:S:C:2
led 3 6,15:S:C:2
led 4 7,15:S:C:2
led 5 8,15:S:C:2
led 6 9,15:S:C:2
led 7 10,15:S:C:2
led 8 11,15:S:C:2
led 9 12,15:S:AC:2
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
[... 220 lines cut ...]
set i_level = 15
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500
set main_cut_hz = 17

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 130
set rc_expo = 65
set rc_yaw_expo = 35
set thr_mid = 40
set thr_expo = 35
set roll_rate = 35
set pitch_rate = 35
set yaw_rate = 30
set tpa_rate = 50
set tpa_breakpoint = 1350

#

borisbstyle, 2015-07-06:
that version is quite old. You need to try the latest master.

pkdevel, 2015-07-06:
@borisbstyle just saw your pr #1028. i guess you're talking about this one.
gonna merge this tonight and give it a try.

will report asap.

cheers

pkdevel, 2015-07-08:
hey guys,

so i just tried out the plain 1.9.0 tag version with only #1028 merged into.
i still have the same issue:
https://youtu.be/TogyE-QSzQ0
https://youtu.be/fQSzll4NiLg

funny thing is, i tried with my dump and also with default settings. as soon as i enable oneshot and reboot a second time, the motors spin up.

i guess its sensless to post another cli dump, since there is only oneshot enabled.

here are my ultraesc settings:
[image]

can i provide anything else to help?
any ideas how i could debug more?

cheers

borisbstyle, 2015-07-08:
Is this the latest master or the official 1.9?
I cant remember.......but I though the final fix was in 1.9.
So its just ultra esc having this issue?

pkdevel, 2015-07-08:
i'm using the offical/final 1.9.0 tag version.
i don't think this is an issue regarding those ultras … a friend using some cheap blhelis 12a has the same issue

borisbstyle, 2015-07-08:
than it sounds like you are not using the version with the fix. There are
many people with blheli including me which dont have this issue anymore.
Please try the latest master or this one
https://www.dropbox.com/s/9690qnccqguhjre/cleanflight_NAZE.hex?dl=0

i'm using the offical/final 1.9.0 tag version.
i don't think this is an issue regarding those ultras … a friend using some
cheap blhelis 12a has the same issue

hydra, 2015-10-19:
no response, no further reports. closing.  please re-open a new issue and reference this if there is still a problem.

closed 2015-10-19 by hydra

paalmiso, 2015-11-09:
I think I've just experienced this with my IMRC Vortex. I'm on Cleanflight 1.9.1 downloaded from here http://www.immersionrc.com/fpv-products/vortex-racing-quad/#specifications My Vortex Firmware version is v1.0.0.15

How can I help you debug this?

I was editing configuration - checked checkbox to prevent motors spinning after arming. Then I've hit save and reboot, and my motors went full throttle for few seconds shredding skin on my little finger. I've ended on ER, fortunately nothing serious.