cleanflight/cleanflight #1032: 3D setup documentation

Hwurzburg, 2015-06-19:
having just gone thru this and researched the code for the various params, which are currently undocumented, I wrote this WIKI file: 3D setup.md    Unfortunately, I dont know how to ADD a file and its pull request, so I am adding this issue for one of the gurus to review and possibly add to the WIKI:

"When FEATURE 3D is enabled,either by the configuration GUI or CLI command, it allows operation with bi-directional escs to provide inverted as well as normal,upright operation. 

Setup:
-Enable FEATURE 3d in the firmware
-Setup ESC to BIDIRECTIONAL (norm/reverse/birectional slider in BLHeli GUI) and set max pwm in the ESC GUI to the maximum your transmitter outputs on the throttle channel (normally 2000us) and min pwm to the minimum (normally 1000us) and the midpoint to halfway between (normally 1500us)
-Setup the following parameters in the CLI:
 3d_deadband_high: lowest value for upright throttle output when armed, maximum is max_throttle (typically 1600)
 3d_deadband_low: lowest value for inverted throttle output when armed, minimum is mincommand (typically 1400)
                 (Note: switchover point for upright or inverted is midrc on throttle stick)
 3d_neutral: same effect as mincommand, but for 3d operation...ie disarmed pulsewidth to escs (typically 1500)
 3d_deadband_throttle: throttle stick range around midrc for allowing arming to occur (typically 50)
 maxthrottle= max tx output (typically 2000us)
 mincommand = min tx output (typically 1000us)
-Set a TX switch to control arming and set this up in the Mode tab of the configuration GUI

motor_stop and stick arming do NOT work in 3D mode for obvious safety issues. Arming will only occur when the throttle stick is centered (=/- 3d_deadband_throttle) and the ARMing switch is active. The motors will immediately spin at 3d_deadband_low or _high, depending on throttle position above or below midrc. Full power reverse will be at low stick and full power normal will be 
at full high stick. De-activating the ARMing switch will stop motors immediately independent of throttle stick position.

You will need to set the proper rotation for each motor by wiring order for the motor.
3D props are highly recommended to get reasonable and equal power inverted or upright."

SteveAmor, 2015-06-20:
@Hwurzburg this is great information and I would really recommend that you walk through the process to create a pull request.  In fact, you have completed the first step.  I see that you have already forked the cleanflight code.  The next step is for you to get Git running on your computer and following the instructions here:
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md#using-git-and-github
Step 2 requires you to clone your fork to your computer.  In your case, the command would be:
`git clone https://github.com/Hwurzburg/cleanflight.git`
For Step 6, simply copy your `3D setup.md` file to directory `/cleanflight/docs`
You will be able to get near real-time support on the IRC channel if you are having problems.

Hwurzburg, 2015-06-20:
took a while to setup and learn...but I think I have created one

   

https://github.com/cleanflight/cleanflight/blob/master/docs/development/Development.md#using-git-and-github

SteveAmor, 2015-06-20:
@Hwurzburg Fantastic.  You have indeed created a Pull Request.

hydra, 2015-06-28:
See #1034

hydra, 2015-06-28:
Lets close this issue and discuss on the PR itself now.

closed 2015-06-28 by hydra

Janked, 2015-09-29:
@Hwurzburg Anyway I can contact you to discuss setting this up?  I have it all setup properly as far as I know, but it won't arm for me from the radio(I use an arming switch with a normal setup so the switch is right).  In Cleanflight itself if I turn motor testing on, put the slider at 1500 it arms, then moving sliders up it they turn the proper direction and pulling the slider below 1500 they reverse.  So all that is working, it just doesn't seem to work from my radio.  I get a constant beep and green light flashing on the FC when trying to arm.  You can email me at jankomatic AT gmail.com if you might help me out.  :)

Hwurzburg, 2015-09-29:
Look in RX tab...make sure center throttle stick is indeed putting out 1500 before you try to arm...if it works in the Motors tab, then it should arm no problem with RX....but  have to have rc channel centered at 1500us (assuming 3d_neutral is set to 1500)

Janked, 2015-09-29:
I think I have it fixed, will test in a bit.  I actually had failsafe turned on in CF and that seemed to keep it from arming.